openwrt/packages #18878: `gcc`: Download specifically of `gcc` fails with `Cannot find appropriate hash command, ensure the provided hash is either a MD5 or SHA256 checksum.`.

felixbecker2, 2022-07-06:
**Maintainer:** OpenWrt.org (according to `Makefile`)
**Environment:**
* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWRT version: Git branch `openwrt-22.03` `from git://git.openwrt.org/openwrt/openwrt.git`, checked out on 2022-07-05, latest commit: Hash `ebfe66e494e57f4b421f1190d6bff1d361db1b3d` at 2022-07-05T10:16:08+02:00:  
  `git log -1 origin --format='%H | %aI | %an'`:  
  `ebfe66e494e57f4b421f1190d6bff1d361db1b3d | 2022-07-05T10:16:08+02:00 | Sander Vanheule`
* Building on GNU/Linux on x86_64-architecture (Distribution: Artix Linux)

**Description:**

When I select the `gcc` package to be build, a `make -j1 V=s download` fails specifically on `gcc` with:  
```
[...]
make[2]: Entering directory '/home/[...]/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/feeds/packages/devel/gcc'
mkdir -p /home/[...]/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/dl
SHELL= flock /home/[...]/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/tmp/.gcc-11.2.0.tar.xz.flock -c '  	/home/[...]/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/scripts/download.pl "/home/[...]/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/dl" "gcc-11.2.0.tar.xz" "x" "" "@GNU/gcc/gcc-11.2.0"    '
Cannot find appropriate hash command, ensure the provided hash is either a MD5 or SHA256 checksum.
make[2]: *** [Makefile:244: /home/[...]/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/dl/gcc-11.2.0.tar.xz] Error 255
make[2]: Leaving directory '/home/[...]/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/feeds/packages/devel/gcc'
time: package/feeds/packages/gcc/download#1.20#0.17#1.34
    ERROR: package/feeds/packages/gcc failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/gcc/download] Error 1
[...]
```

Configuration details regarding `gcc`:  
`grep -i gcc .config`:  
```
CONFIG_DEFAULT_libgcc=y
CONFIG_EXTRA_GCC_CONFIG_OPTIONS=""
# CONFIG_GCC_DEFAULT_PIE is not set
# CONFIG_GCC_DEFAULT_SSP is not set
CONFIG_GCC_VERSION="11.2.0"
# CONFIG_BUSYBOX_DEFAULT_STATIC_LIBGCC is not set
# CONFIG_BUSYBOX_CONFIG_STATIC_LIBGCC is not set
CONFIG_PACKAGE_libgcc=y
CONFIG_PACKAGE_gcc=m
```

I have tried a `make oldconfig`, this did report to not have made any changes.

Other packages download fine.

Full `.config` is attached:  
[`.config.txt`](https://github.com/openwrt/packages/files/9055654/default.config.txt)

neheb, 2022-07-08:
Is this fixed?

felixbecker2, 2022-07-09:
> Is this fixed?

It seems to be fixed:

I just did a `git pull` and `./scripts/feeds update -a` and `./scripts/feeds install -a`.  
In `make menuconfig`, selected `` again.  
Then, `make -j1 V=sc download`.

Result: Download runs through without error.

felixbecker2, 2022-07-09:
> Is this fixed?

It seems to be fixed:

I just did a `git pull` and `./scripts/feeds update -a` and `./scripts/feeds install -a`.  
In `make menuconfig`, selected `Development` -> `gcc` =m again.  
Then, `make -j1 V=sc download`.

Result: Download runs through without error.

closed 2022-07-09 by felixbecker2