openwrt/packages #18884: `glib2`: Fails to build by `-Werror=format-nonliteral` under 22.03 master.

felixbecker2, 2022-07-07:
**Maintainer:** Peter Wagner <tripolar(at)gmx.at>  
**Environment:**
* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWRT version: Git branch `openwrt-22.03` `from git://git.openwrt.org/openwrt/openwrt.git`, checked out on 2022-07-05, latest commit: Hash `ebfe66e494e57f4b421f1190d6bff1d361db1b3d` at 2022-07-05T10:16:08+02:00:  
  `git log -1 origin --format='%H | %aI | %an'`:  
  `ebfe66e494e57f4b421f1190d6bff1d361db1b3d | 2022-07-05T10:16:08+02:00 | Sander Vanheule`
* Building on GNU/Linux on x86_64-architecture (Distribution: Artix Linux)

**Description:**

Building `package/feeds/packages/glib2` fails due to a warning treated as error in `glib/libcharset/localcharset.c`:

`make -j1 V=s`:  
```
[...]
make[3]: Entering directory '/home/[...]/openwrt-22.03/feeds/packages/libs/glib2'
rm -f /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/glib-2.70.5/.built
touch /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/glib-2.70.5/.built_check
MAKEFLAGS="" /home/[...]/openwrt-22.03/staging_dir/host/bin/ninja  -j1 -C /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/glib-2.70.5/openwrt-build 
ninja: Entering directory `/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/glib-2.70.5/openwrt-build'
[1/562] Compiling C object glib/libcharset/libcharset.a.p/localcharset.c.o
FAILED: glib/libcharset/libcharset.a.p/localcharset.c.o 
ccache_cc -Iglib/libcharset/libcharset.a.p -Iglib/libcharset -I../glib/libcharset -I. -I.. -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/include -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include -flto=auto -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -D_GNU_SOURCE -fno-strict-aliasing -DG_DISABLE_ASSERT -Wduplicated-branches -Wimplicit-fallthrough -Wmisleading-indentation -Wstrict-prototypes -Wunused -Wno-unused-parameter -Wno-bad-function-cast -Wno-cast-function-type -Wno-pedantic -Wno-format-zero-length -Werror=declaration-after-statement -Werror=format=2 -Werror=implicit-function-declaration -Werror=init-self -Werror=missing-include-dirs -Werror=missing-prototypes -Werror=pointer-arith -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/glib-2.70.5=glib-2.70.5 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -fPIC '-DGLIB_CHARSETALIAS_DIR="/usr/lib"' -fvisibility=hidden -MD -MQ glib/libcharset/libcharset.a.p/localcharset.c.o -MF glib/libcharset/libcharset.a.p/localcharset.c.o.d -o glib/libcharset/libcharset.a.p/localcharset.c.o -c ../glib/libcharset/localcharset.c
In file included from ../glib/libcharset/localcharset.c:28:
../../../../staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify/stdio.h: In function 'snprintf':
../../../../staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify/stdio.h:101:9: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  101 |         return __orig_snprintf(__s, __n, __f, __builtin_va_arg_pack());
      |         ^~~~~~
../../../../staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify/stdio.h: In function 'sprintf':
../../../../staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify/stdio.h:110:17: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  110 |                 __r = __orig_snprintf(__s, __b, __f, __builtin_va_arg_pack());
      |                 ^~~
../../../../staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify/stdio.h:114:17: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  114 |                 __r = __orig_sprintf(__s, __f, __builtin_va_arg_pack());
      |                 ^~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:125: /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/glib-2.70.5/.built] Error 1
make[3]: Leaving directory '/home/[...]/openwrt-22.03/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.58#0.21#0.74
    ERROR: package/feeds/packages/glib2 failed to build.
[...]
```

Configuration:
* `grep -i glib2 .config`:  
  ```
  CONFIG_PACKAGE_glib2=y
  ```
* Full `.config`: [&rarr; here](http://ix.io/41PY).

felixbecker2, 2022-07-07:
*<small>Mentioning @tripoloar, since I did not know in the first place that a mention is necessary and I don't know if an edit of the initial post where @tripolar is added triggers a notitication.</small>*

neheb, 2022-07-09:
related to https://github.com/openwrt/packages/pull/17542

I don't know under which conditions this happens.

felixbecker2, 2022-07-09:
Could somewhere an explicit `-Wno-error=format-nonliteral` be added to build options related to `glib2`? I did not manage to have it honoured by the steps I tried.

felixbecker2, 2022-07-09:
*@neheb wrote:*  
> *related to #17542*

Inspired by this I globally disabled "fortify source" in the config (changed  
```
# CONFIG_PKG_FORTIFY_SOURCE_NONE=y is not set
CONFIG_PKG_FORTIFY_SOURCE_1=y
# CONFIG_PKG_FORTIFY_SOURCE_2 is not set
```  
to  
```
CONFIG_PKG_FORTIFY_SOURCE_NONE=y
# CONFIG_PKG_FORTIFY_SOURCE_1 is not set
# CONFIG_PKG_FORTIFY_SOURCE_2 is not set

```  
), did run a `make dirclean`, and then with `make -j1 V=sc package/feeds/packages/glib2/compile` `glib2` compiled fine.

I think better would be to disable the `-Werror=format-nonliteral` which is set somewhere, if not possible properly disable fortify source for `glib2` in a way that it still can be enabled for the other packages.

neheb, 2022-07-20:
Fixed and backported.

closed 2022-07-20 by neheb