openwrt/packages #18896: `nsutils`: Fails to build with `nssearch.c:108:27: error: 'PATH_MAX' undeclared (first use in this function)` in 22.03.

felixbecker2, 2022-07-09:
## Maintainer:

Oskari Rauta <oskari.rauta(at)gmail.com> (According to `package/feeds/packages/nsutils/Makefile`), @oskarirauta  

## Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` `from git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-09, 18:40 UTC,
  - latest commit in main OpenWrt repository: `git log -1 origin --format='%H | %aI | %an'`:  
    `c29b13cfa09ebb4e81d4c6f65a4d1190316126bc | 2022-07-05T15:27:52+01:00 | Kevin Darbyshire-Bryant`,
  - latest commit in `packages` feed:  
    `598bf7936a0213254dd1fb1493f418099c28d60f | 2022-06-09T15:35:40-07:00 | Rosen Penev`.
* Building on GNU/Linux on x86_64-architecture (Distribution: Artix Linux)
* OpenWrt configuration: [`config.txt`](https://github.com/openwrt/packages/files/9077712/config.txt).
* `grep -i nsutils .config`:  
  `CONFIG_PACKAGE_nsutils=m`

## Description:

Compilation of of `nsutils` fails for me with `nssearch.c:108:27: error: 'PATH_MAX' undeclared (first use in this function)` in the above setup.

When I issue `make -j1 V=sc package/feeds/packages/nsutils/compile`, I get:  
```
make[2]: Entering directory '/home/[...]/openwrt-22.03/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/[...]/openwrt-22.03/scripts/config'
make[1]: Entering directory '/home/[...]/openwrt-22.03'
make[2]: Entering directory '/home/[...]/openwrt-22.03/package/libs/toolchain'
echo "libc" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libgcc" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libatomic" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libpthread" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "librt" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/[...]/openwrt-22.03/package/libs/toolchain'
time: package/libs/toolchain/compile#0.36#0.13#0.45
make[2]: Entering directory '/home/[...]/openwrt-22.03/package/libs/libbsd'
make[2]: Leaving directory '/home/[...]/openwrt-22.03/package/libs/libbsd'
time: package/libs/libbsd/compile#0.51#0.15#0.62
make[2]: Entering directory '/home/[...]/openwrt-22.03/package/libs/libcap'
make[2]: Leaving directory '/home/[...]/openwrt-22.03/package/libs/libcap'
time: package/libs/libcap/compile#0.50#0.15#0.60
make[2]: Entering directory '/home/[...]/openwrt-22.03/feeds/packages/utils/nsutils'
rm -f /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2/.built
touch /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2/.built_check
CFLAGS="-O3 -pipe -g0 -Wno-error -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2=nsutils-0.2 -DPIC -fpic -fstack-protector -Wl,-z,now -Wl,-z,relro  -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include " CXXFLAGS="-O3 -pipe -g0 -Wno-error -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2=nsutils-0.2 -DPIC -fpic -fstack-protector -Wl,-z,now -Wl,-z,relro  -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include " LDFLAGS="-L/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -DPIC -fpic -specs=/home/[...]/openwrt-22.03/include/hardened-ld-pie.specs -znow -zrelro " make -j1 -C /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="ccache_cc -c -O3 -pipe -g0 -Wno-error -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2=nsutils-0.2 -DPIC -fpic -fstack-protector -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" ;
make[3]: Entering directory '/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2'
make  all-am
make[4]: Entering directory '/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2'
ccache_cc -DHAVE_CONFIG_H -I.   -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include   -O3 -pipe -g0 -Wno-error -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2=nsutils-0.2 -DPIC -fpic -fstack-protector -Wl,-z,now -Wl,-z,relro  -Wall -MT nssearch.o -MD -MP -MF .deps/nssearch.Tpo -c -o nssearch.o nssearch.c
nssearch.c: In function 'ckns':
nssearch.c:108:27: error: 'PATH_MAX' undeclared (first use in this function)
  108 |                 char nsid[PATH_MAX+1];
      |                           ^~~~~~~~
nssearch.c:108:27: note: each undeclared identifier is reported only once for each function it appears in
nssearch.c:109:21: warning: variable 'len' set but not used [-Wunused-but-set-variable]
  109 |                 int len;
      |                     ^~~
nssearch.c:108:22: warning: unused variable 'nsid' [-Wunused-variable]
  108 |                 char nsid[PATH_MAX+1];
      |                      ^~~~
make[4]: *** [Makefile:454: nssearch.o] Error 1
make[4]: Leaving directory '/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2'
make[3]: *** [Makefile:326: all] Error 2
make[3]: Leaving directory '/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2'
make[2]: *** [Makefile:67: /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/nsutils-0.2/.built] Error 2
make[2]: Leaving directory '/home/[...]/openwrt-22.03/feeds/packages/utils/nsutils'
time: package/feeds/packages/nsutils/compile#1.24#0.28#1.47
    ERROR: package/feeds/packages/nsutils failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/nsutils/compile] Error 1
make[1]: Leaving directory '/home/[...]/openwrt-22.03'
make: *** [/home/[...]/openwrt-22.03/include/toplevel.mk:230: package/feeds/packages/nsutils/compile] Error 2
```

neheb, 2022-07-10:
update feeds.

felixbecker2, 2022-07-10:
*@neheb wrote:*  
> *update feeds.*

Done, now it builds without error.

Closing this issue.

closed 2022-07-10 by felixbecker2