openwrt/packages #18897: `libxml2`: Fails to build with `SAX.c:22: error: expected identifier or '(' at end of input` in 22.03.

felixbecker2, 2022-07-09:
## Maintainer:

Michael Heimpold <mhei(at)heimpold.de> (According to `package/feeds/packages/libxml2/Makefile`), @mhei

## Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-09, 18:40 UTC,
  - latest commit in main OpenWrt repository: `git log -1 origin --format='%H | %aI | %an'`:  
    `c29b13cfa09ebb4e81d4c6f65a4d1190316126bc | 2022-07-05T15:27:52+01:00 | Kevin Darbyshire-Bryant`,
  - latest commit in `packages` feed:  
    `598bf7936a0213254dd1fb1493f418099c28d60f | 2022-06-09T15:35:40-07:00 | Rosen Penev`.
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [`config.txt`](https://github.com/openwrt/packages/files/9077779/config.txt).
* `grep -i libxml2 .config`:  
  ```
  CONFIG_PACKAGE_libxml2-dev=m
  # CONFIG_libstrophe-libxml2 is not set
  CONFIG_PACKAGE_libxml2=y
  # CONFIG_ARIA2_LIBXML2 is not set
  # CONFIG_PACKAGE_libxml2-utils is not set
  ```

## Description:

Compilation of of `libxml2` fails for me with `SAX.c:22: error: expected identifier or '(' at end of input` in the above setup.

When I issue `make -j1 V=sc package/feeds/packages/libxml2/compile`, I get:  
```
make[2]: Entering directory '/home/[...]/openwrt-22.03/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/[...]/openwrt-22.03/scripts/config'
make[1]: Entering directory '/home/[...]/openwrt-22.03'
make[2]: Entering directory '/home/[...]/openwrt-22.03/package/libs/toolchain'
echo "libc" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libgcc" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libatomic" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libpthread" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "librt" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/[...]/openwrt-22.03/package/libs/toolchain'
time: package/libs/toolchain/compile#0.39#0.12#0.50
make[2]: Entering directory '/home/[...]/openwrt-22.03/package/libs/zlib'
echo "zlib" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/zlib.default.install
make[2]: Leaving directory '/home/[...]/openwrt-22.03/package/libs/zlib'
time: package/libs/zlib/compile#0.36#0.14#0.46
make[2]: Entering directory '/home/[...]/openwrt-22.03/package/libs/libiconv-full'
echo "libiconv-full" >> /home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/libiconv-full.default.install
make[2]: Leaving directory '/home/[...]/openwrt-22.03/package/libs/libiconv-full'
time: package/libs/libiconv-full/compile#0.55#0.15#0.67
make[2]: Entering directory '/home/[...]/openwrt-22.03/feeds/packages/libs/libxml2'
rm -f /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14/.built
touch /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14/.built_check
CFLAGS="-O3 -pipe -g0 -Wno-error -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14=libxml2-2.9.14 -DPIC -fpic -fstack-protector -Wl,-z,now -Wl,-z,relro -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/include -DPIC -fpic  -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/include " CXXFLAGS="-O3 -pipe -g0 -Wno-error -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14=libxml2-2.9.14 -DPIC -fpic -fstack-protector -Wl,-z,now -Wl,-z,relro -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/include -DPIC -fpic  -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/include " LDFLAGS="-L/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/home/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -DPIC -fpic -specs=/home/[...]/openwrt-22.03/include/hardened-ld-pie.specs -znow -zrelro -L/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/lib -Wl,-rpath-link=/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/lib -L/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/lib -Wl,-rpath-link=/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/lib " make -j1 -C /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="ccache_cc -c -O3 -pipe -g0 -Wno-error -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14=libxml2-2.9.14 -DPIC -fpic -fstack-protector -Wl,-z,now -Wl,-z,relro -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-full/include -I/home/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-full/include -DPIC -fpic" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" ;
make[3]: Entering directory '/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14'
make  all-recursive
[... 7 lines cut ...]
make[6]: Entering directory '/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14/include'
make[6]: Nothing to be done for 'all-am'.
[... cut ...]
make[2]: *** [Makefile:203: /home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libxml2-2.9.14/.built] Error 2
make[2]: Leaving directory '/home/[...]/openwrt-22.03/feeds/packages/libs/libxml2'
time: package/feeds/packages/libxml2/compile#0.76#0.17#0.89
    ERROR: package/feeds/packages/libxml2 failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/libxml2/compile] Error 1
make[1]: Leaving directory '/home/[...]/openwrt-22.03'
make: *** [/home/[...]/openwrt-22.03/include/toplevel.mk:230: package/feeds/packages/libxml2/compile] Error 2
```

mhei, 2022-07-10:
Can reproduce it, will investigate.

felixbecker2, 2022-07-14:
To be able to get a useful build, I deselected all packages that depend in `libxml2` and then also deselected `libxml2`.

I did make a `make dirclean`.

However, `libxml` _still_ gets build (and fails)!

Any idea why? And any idea how I coud get it compiling without `libxml2` beeing tried to compile?

Here my current `.config`: [`config.txt`](https://github.com/openwrt/packages/files/9113588/config.txt),  
`grep -i libxml .config`:  
```
# CONFIG_PACKAGE_libxml2-dev is not set
# CONFIG_PHP8_LIBXML is not set
# CONFIG_libstrophe-libxml2 is not set
# CONFIG_PACKAGE_libxml2 is not set
# CONFIG_ARIA2_LIBXML2 is not set
# CONFIG_PACKAGE_libxml2-utils is not set
```

mhei, 2022-07-14:
My build succeeds when I deselect "Compile with full language support", i.e.
`# CONFIG_BUILD_NLS is not set`
Can you confirm?

mhei, 2022-07-14:
Your report is really odd, I cannot explain why libxml2 still gets build when not selected at all. You could try so save your .config and then run a `make distclean` - but in theory such a rough method should not be necessary...

felixbecker2, 2022-07-14:
*@mhei wrote:*  
> *My build succeeds when I deselect "Compile with full language support", i.e.  
> `# CONFIG_BUILD_NLS is not set`  
> Can you confirm?*

I can confirm that with unsetting `CONFIG_BUILD_NLS` `libxml` build fine.

So I change the title of this bug report to be more precise and edit my initial post.

*@mhei wrote:*  
> *Your report is really odd, I cannot explain why libxml2 still gets build when not selected at all. You could try so save your .config and then run a `make distclean` - but in theory such a rough method should not be necessary...*

I only did try `make dirclean`. When I did try to compile wayland-library (from the "video" feed) explicitly, I observed that it directly wanted to compile libxml, although `libxml` is not selected as dependency. I unselected all wayland-stuff but `libxml2` _still_ wants to compile. I don't know, could it be that some package depends on it but the `make menuconfig`-logic is somehow broken?

mhei, 2022-09-30:
Can you still reproduce this with latest 22.03 branch? I checked out a fresh copy and cannot reproduce anymore. Maybe this is fixed with last libxml2 update to 2.10.2?

mhei, 2022-10-09:
No reply, closing.

closed 2022-10-09 by mhei

dreirund, 2022-10-10:
> No reply, closing.

Please, not so quick, I am currently not often online and at a powerful
computer to check compilation. I would need 2 weeks more to be able to
test.

Regards!

mhei, 2022-10-15:
Ok, no problem - I just wanted to prevent dangling issues, so your feedback is welcome. Take your time and report here again - you can always reopen.

dreirund, 2023-09-06:
> *No reply, closing.*

Closing, I lost my build infrastructure and have no capacity right now to re-create one.