openwrt/packages #18899: python package build failed: 'setup.py: No such file or directory'

sgpublic, 2022-07-10:
Hello! I‘m a beginner in python, and I'm trying to create a Makefile for `python3-fastapi`:

```Makefile
include $(TOPDIR)/rules.mk

PKG_NAME:=fastapi
PKG_VERSION:=0.78.0
PKG_RELEASE:=1

PYPI_NAME:=$(PKG_NAME)
PKG_HASH:=3233d4a789ba018578658e2af1a4bb5e38bdd122ff722b313666a9b2c6786a83

PKG_MAINTAINER:=tiangolo
PKG_LICENSE:=MIT
PKG_LICENSE_FILES:=LICENSE

HOST_PYTHON3_PACKAGE_BUILD_DEPENDS:=setuptools-scm toml  # setuptools_scm[toml] >= 3.4.1

include $(TOPDIR)/feeds/packages/lang/python/pypi.mk
include $(INCLUDE_DIR)/package.mk
include $(TOPDIR)/feeds/packages/lang/python/python3-package.mk

define Package/python3-$(PKG_NAME)
  SECTION:=lang
  CATEGORY:=Languages
  SUBMENU:=Python
  TITLE:=python3-$(PKG_NAME)
  URL:=https://github.com/tiangolo/fastapi
  DEPENDS:=+python3 +python3-pydantic +python3-starlette
endef

define Package/python3-$(PKG_NAME)/description
FastAPI is a modern, fast (high-performance), web framework for building APIs
with Python 3.6+ based on standard Python type hints.
endef

$(eval $(call Py3Package,python3-$(PKG_NAME)))
$(eval $(call BuildPackage,python3-$(PKG_NAME)))
$(eval $(call BuildPackage,python3-$(PKG_NAME)-src))
```

But `setup.py` is not included in the source code of fastapi:

```
cd "/tmp/openwrt/build_dir/target-x86_64_musl/pypi/fastapi-0.78.0/" && CC="ccache_cc" CCSHARED="ccache_cc -DPIC -fpic" CXX="ccache_cxx" LD="ccache_cc" LDSHARED="ccache_cc -shared" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/tmp/openwrt/build_dir/target-x86_64_musl/pypi/fastapi-0.78.0=fastapi-0.78.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/tmp/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/include -I/tmp/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include/fortify -I/tmp/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/include -I/tmp/openwrt/staging_dir/target-x86_64_musl/usr/include/python3.10" LDFLAGS="-L/tmp/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/usr/lib -L/tmp/openwrt/staging_dir/toolchain-x86_64_gcc-11.2.0_musl/lib -znow -zrelro -lpython3.10" _PYTHON_HOST_PLATFORM="linux-x86_64" __PYVENV_LAUNCHER__="/usr/bin/python3.10" PYTHONPATH="/tmp/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.10:/tmp/openwrt/staging_dir/target-x86_64_musl//usr/lib/python3.10/site-packages:/tmp/openwrt/build_dir/target-x86_64_musl/pypi/fastapi-0.78.0/ipkg-install//usr/lib/python3.10/site-packages" PYTHONDONTWRITEBYTECODE=1 PYTHONOPTIMIZE="" _python_sysroot="/tmp/openwrt/staging_dir/target-x86_64_musl" _python_prefix="/usr" _python_exec_prefix="/usr"     /tmp/openwrt/staging_dir/hostpkg/bin/python3.10  setup.py   install --prefix="/usr" --root="/tmp/openwrt/build_dir/target-x86_64_musl/pypi/fastapi-0.78.0/ipkg-install" --single-version-externally-managed
/tmp/openwrt/staging_dir/hostpkg/bin/python3.10: can't open file '/tmp/openwrt/build_dir/target-x86_64_musl/pypi/fastapi-0.78.0/setup.py': [Errno 2] No such file or directory
make[2]: *** [Makefile:47: /tmp/openwrt/build_dir/target-x86_64_musl/pypi/fastapi-0.78.0/.built] Error 2
make[2]: Leaving directory '/home/a/openwrt/package/little-paimon/python3-fastapi'
time: package/little-paimon/python3-fastapi/compile#3.78#0.58#4.38
    ERROR: package/little-paimon/python3-fastapi failed to build.
make[1]: *** [package/Makefile:116: package/little-paimon/python3-fastapi/compile] Error 1
make[1]: Leaving directory '/home/a/openwrt'
make: *** [/home/a/openwrt/include/toplevel.mk:230: package/python3-fastapi/compile] Error 2
```

So what do I need to do? How to build this package? Thanks!

neheb, 2022-07-11:
ping @jefferyto

jefferyto, 2022-07-11:
The Python build system doesn't yet support building packages without setup.py. You can try [patching in a setup.py file](https://github.com/openwrt/packages/blob/df335b13912d85edd6134d2d5bdec898d808af30/lang/python/python-jsonschema/patches/001-setup.patch) like with python-jsonschema. You can also try using flit to [generate the setup.py file](https://github.com/tiangolo/fastapi/issues/4595#issuecomment-1043815586) (then patching it into the package).

sgpublic, 2022-07-11:
Thank you very much for your reply, it helped me to solve the problem smoothly!

closed 2022-07-11 by sgpublic