openwrt/packages #18900: watchcat: Feature request - Taking down Wireguard interface

cmonty14, 2022-07-10:
Maintainer: @\Roger D
Environment: OpenWrt 21.02.3

Feature request:
Add function _Taking down WireGuard interface_ in case VPN tunnel has failure.

Use case:
WireGuard VPN tunnel depends on endpoint, port and other technical objects.
Above all it depends on an active account with the VPN provider.
If this account is deactivated, WireGuard VPN tunnel won't work anymore.
However, as long as WireGuard interface is up, the router will route packages to this tunnel.
And this results in a severe failure because packages will get lost.

Package _watchcat_ provides [these](https://openwrt.org/docs/guide-user/advanced/watchcat) 3 functions:

- Ping Reboot: reboot the OpenWrt device if a ping to a specific host fails
- Restart Interface: restart a network interface if a ping to a host over that interface fails
- Periodic Reboot: reboot at a set period of time, such as every 24h.

Non of these 3 functions will resolve this issue described above.

Workaround:
This script will ping DNS servers of VPN provider [IVPN](https://www.ivpn.net/); these DNS servers are only accessible through VPN tunnel. If none of the DNS servers reponds, there's a failure with the tunnel and failover to WAN must be triggered by taking down WireGuard interface.
Prerequisite for failover to WAN is WireGuard - [Dynamic Connection](https://openwrt.org/docs/guide-user/services/vpn/wireguard/extras#dynamic_connection).

```
root@eddie:~# cat /usr/local/bin/wg-watchdog.sh 
#!/bin/sh

## Ping IVPN DNS that can only be reached via the VPN tunnel.
## If no contact, put wg interface down.

wgInterface=wg0
repeat=5
dns1=172.16.0.1
dns2=10.0.254.2
dns3=10.0.254.3

while [[ 0 -lt $repeat ]]; do
	if /bin/ping -c 1 $dns1; then
		logger -t "wg0" IVPNdns1 is reachable
		exit 0
        elif /bin/ping -c 1 $dns2; then
		logger -t "wg0" IVPNdns2 is reachable
		exit 0
	elif /bin/ping -c 1 $dns3; then
		logger -t "wg0" IVPNdns3 is reachable
		exit 0
	fi
	logger -t "wg0" WireGuard failure
	tries=$((tries+1))
done

logger -t "wg0" WireGuard failed, putting wg interface down
ifdown $wgInterface
```

felixbecker2, 2022-07-10:
Why does

> * *Ping Reboot: reboot the OpenWrt device if a ping to a specific host fails*
> * *Restart Interface: restart a network interface if a ping to a host over that interface fails*

not work? As I understand your description, if the wireguard VPN tunnel is set up but not functioning, there is no internet access, and ping should fail. Why does it not?

And: If the router is rebooted/ the WireGuard interface is taken down and then up again, but the WireGuard connection could not be established, the WireGuard interface should not succeed to an "up" state. Does it anyway report beeing "up" even if it cannot negociate with the other side?

I think the scenario

> *there's a failure with the tunnel and failover to WAN must be triggered*

is use-case dependent: Do you want guaranteed security (i.e. not allow any internet if the VPN is not working) or guaranteed connectivity (also allow non-VPN internet if the VPN is not working)? The user should be able to decide.

Thanks for your feature request!, this are only my comments, I am not a developer or maintainer at all.

cmonty14, 2022-07-10:
Configuring WireGuard client documented [here](https://openwrt.org/docs/guide-user/services/vpn/wireguard/client?s[]=wireguard#wireguard_client) will setup default route using WireGuard interface.
The default route exists as long as WireGuard interface is up.

felixbecker2, 2022-07-10:
> *The default route exists as long as WireGuard interface is up.*

And wireguard interface will succeed to go to "up" state even if wireguard cannot negotiate with the other side and does not establish a tunnel?

cmonty14, 2022-07-10:
> And wireguard interface will succeed to go to "up" state even if wireguard cannot negotiate with the other side and does not establish a tunnel?

Yes, because endpoint is available, means I can ping that IP.