openwrt/packages #18901: `gnutls`: Build fails with `cp: cannot stat '/[...]/build_dir/target-arm[...]/gnutls-3.7.1/ipkg-install/usr/bin/tpmtool': No such file or directory` on 22.03.

felixbecker2, 2022-07-10:
## Maintainer:

Nikos Mavrogiannopoulos <nmav (at) gnutls.org> according to `package/feeds/packages/gnutls/Makefile`, @nmav

## Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-10, ca. 09:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: .
* `grep -i gnutls .config`:  
  ```
  # CONFIG_ZABBIX_GNUTLS is not set
  # CONFIG_RSYSLOG_gnutls is not set
  CONFIG_PACKAGE_libzip-gnutls=m
  # CONFIG_PACKAGE_libulfius-gnutls is not set
  CONFIG_PACKAGE_libgnutls=m
  CONFIG_GNUTLS_DTLS_SRTP=y
  CONFIG_GNUTLS_ALPN=y
  CONFIG_GNUTLS_OCSP=y
  CONFIG_GNUTLS_CRYPTODEV=y
  CONFIG_GNUTLS_HEARTBEAT=y
  CONFIG_GNUTLS_SRP=y
  CONFIG_GNUTLS_PSK=y
  CONFIG_GNUTLS_ANON=y
  CONFIG_GNUTLS_TPM=y
  CONFIG_GNUTLS_PKCS11=y
  CONFIG_GNUTLS_EXT_LIBTASN1=y
  # CONFIG_PACKAGE_libgnutls-dane is not set
  # CONFIG_LIBCURL_GNUTLS is not set
  # CONFIG_PACKAGE_exim-gnutls is not set
  # CONFIG_MUTT_GNUTLS is not set
  # CONFIG_ARIA2_GNUTLS is not set
  # CONFIG_DNSDIST_GNUTLS is not set
  # CONFIG_OPENCONNECT_GNUTLS is not set
  # CONFIG_VPNC_GNUTLS is not set
  CONFIG_PACKAGE_gnutls-utils=m
  ```

## Description:

Compilation of of `gnutls` fails for me with `cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/tpmtool': No such file or directory` in the above setup.

When I issue `make -j1 V=sc package/feeds/packages/gnutls/compile`, I get:  
```
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'libX11', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'libXext', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'tslib', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'directfb', which does not exist
make[2]: Entering directory '/[...]/openwrt-22.03/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/[...]/openwrt-22.03/scripts/config'
make[1]: Entering directory '/[...]/openwrt-22.03'
make[2]: Entering directory '/[...]/openwrt-22.03/package/libs/toolchain'
echo "libc" >> /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
[... 428 lines cut ...]
echo "libtasn1" >> /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/libtasn1.default.install
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/libs/libtasn1'
time: package/feeds/packages/libtasn1/compile#0.43#0.13#0.49
make[2]: Entering directory '/[...]/openwrt-22.03/feeds/packages/libs/p11-kit'
echo "p11-kit" >> /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo/p11-kit.default.install
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/libs/p11-kit'
time: package/feeds/packages/p11-kit/compile#0.42#0.11#0.48
make[2]: Entering directory '/[...]/openwrt-22.03/feeds/packages/libs/gnutls'
mkdir -p /[...]/openwrt-22.03/bin/targets/sunxi/cortexa7/packages /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/CONTROL /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo
install -d -m0755 /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/usr/bin
cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/gnutls-{cli,serv} /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/usr/bin/
cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/ocsptool /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/usr/bin/
cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/srptool /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/usr/bin/
cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/psktool /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/usr/bin/
cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/p11tool /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/usr/bin/
cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/tpmtool /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-arm_cortex-a7_neon-vfpv4/gnutls-utils/usr/bin/
cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/gnutls-3.7.1/ipkg-install/usr/bin/tpmtool': No such file or directory
make[2]: *** [Makefile:271: /[...]/openwrt-22.03/bin/packages/arm_cortex-a7_neon-vfpv4/packages/gnutls-utils_3.7.1-2_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#0.56#0.23#0.66
    ERROR: package/feeds/packages/gnutls failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1
make[1]: Leaving directory '/[...]/openwrt-22.03'
make: *** [/[...]/openwrt-22.03/include/toplevel.mk:230: package/feeds/packages/gnutls/compile] Error 2
```

neheb, 2022-07-11:
You need to look at config.log to see why it's failing.

neheb, 2022-07-11:
Relevant: https://github.com/openwrt/packages/issues/11676#issuecomment-605059529