openwrt/packages #18902: `socat`: Build for `arm_cortex-a7_neon-vfpv4` on x86_64 fails with cross compilation issue `configure: error: cannot run test program while cross compiling` in 22.03.

felixbecker2, 2022-07-10:
### Maintainer:

Ted Hess <thess(at)kitschensync.net> according to `package/feeds/packages/socat/Makefile`, @thess

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-10, ca. 16:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [`config.txt`](https://github.com/openwrt/packages/files/9079847/config.txt).
* `grep -i socat .config`:  
  ```
  CONFIG_PACKAGE_socat=m
  CONFIG_SOCAT_SSL=y
  ```

### Description:

Compilation of `socat` fails in `./configure` with a cross compilation issue:

`make -j1 V=sc package/feeds/packages/socat/compile`:  
```
[...]
checking for equivalent simple type of uint16_t... configure: error: in `/home/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/socat-1.7.4.1':
configure: error: cannot run test program while cross compiling
[...]
```

Full `make` output compilation output is too long for a github comment, it is attached here: [`socat-compile.log` (68&thinsp;KiB)](https://github.com/openwrt/packages/files/9079851/socat-compile.log).

neheb, 2022-07-11:
You need to look at config.log to see why it's failing.

neheb, 2022-07-11:
```
git grep TRY_RUN | wc -l
19
```

ffs

felixbecker2, 2022-07-11:
Where I can find the `config.log` for an individual package build?

In `package/feeds/packages/socat/` I find none.

felixbecker2, 2022-07-11:
Ahoj @Rosen&nbsp;Penev,

> ```
> git grep TRY_RUN | wc -l
> 19
> ```
> 
> ffs

I don't understand your comment.

Can you tell me in which directory that command to be executed? And
what "ffs" means?

Regards!

neheb, 2022-07-11:
It’s in build_dir/target_xxx/strace-yyy/config.log

The comment was that the build system is testing stuff by running them (not possible when cross compiling). Not sure if relevant here as this builds fine for me locally.

thess, 2022-07-11:
Chances are this is a toolchain / environment problem in your build environment.

The output from 'configure' is located in the package build directory - In your case:
.../build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/socat-1.7.4.1/config.log

In there you can find the command or test that failed.  From the looks of the error, something is failing to launch. Sometimes adding `PKG_FIXUP=autoreconf` in the Makefile (before `include $(INCLUDE_DIR)/package.mk` can resolve obscure platform configuration issues.

neheb, 2022-07-12:
```
configure:10067: error: cannot run test program while cross compiling
```

So what referenced earlier was AC_TRY_RUN, which does not work when cross compiling. Looking at the checks again, they're totally broken and can probably be replaced with static_assert, which happens at compile time.

neheb, 2022-07-12:
K I found something:

```
   2 dnl Does the test code compile without -Werror when types do not fit?
   1 if test "$CHANCE_TO_TYPECHECK" -ne 0; then
1661  AC_TRY_COMPILE([#include <stdlib.h>],[int u; unsigned int v; exit(&u==&v);],,CHANCE_TO_TYPECHECK=0)
   1 fi
```

This throws an error for me and a warning for you even though -Werror is passed. That is so strange...

felixbecker2, 2022-07-13:
> This throws an error for me and a warning for you even though -Werror is passed. That is so strange...
> 
> edit: wait a minute...you're also passing `-O3` for some reason...

Yes, I added it to the compile options globally. I don't need to optimise for size, so I want to optimise for speed. Is this prone to break things?

> edit2: and `-Wno-error`... This compile check must absolutely fail so that it can avoid having to use AC_TRY_RUN.

Yes, I added it to the compile options globally in a hope to circumvent #18884 ("[*`glib2`: Fails to build by `-Werror=format-nonliteral` under 22.03.*](/openwrt/packages/issues/18884)")

felixbecker2, 2022-07-13:
> That should not be global.

* Where should per-package compile options properly be set? (Only by
  tampering with the `Makefile` directly?)

* Where should global optimisation flags be set, when not globally?
  Should we not globally decide to optimise for size or for speed for
  packages that not have own configuration options for that (= most
  packahes)?

Regards!

neheb, 2022-07-13:
I was talking about -Wno-error. As you saw, that broke socat.

felixbecker2, 2022-07-14:
I have removed `-Wno-error` from `CONFIG_TARGET_OPTIMIZATION` and `socat` now builds fine.

I am closing this issue as invalid.

Still interesting why such an option causes `./configure` to run some programme &hellip;

closed 2022-07-14 by felixbecker2