openwrt/packages #18929: utils/watchcat: allow different interface for monitoring to restart feature

sages, 2022-07-14:
Maintainer: <rogerdammit\@gmail.com>
Environment: any

Description:
Feature request.
My use case is monitoring a vxlan interface that needs a different monitoring interface (layer 3) to the underlying (layer 2) interface that needs to be restarted if the link fails. ie ping monitor over the bridge sitting ontop of the vxlan link.
ie ping -I br-lan 1.2.3.4 and if fail, restart lnk1(vxlan interface layer 2)

felixbecker2, 2022-07-14:
_@roger-_

kofec, 2022-07-14:
Probably this functionality will solve your issue:
https://github.com/openwrt/packages/pull/18593

sages, 2022-07-14:
> Probably this functionality will solve your issue: #18593

yep, looks exactly like what I am after

closed 2022-07-21 by sages