openwrt/packages #18941: `lttng-ust`: Fails to build with `libtool:   error: error: relink 'liblttng-ust.la' with the above command before installing it` in 22.03.

felixbecker2, 2022-07-15:
### Maintainer:

**No maintainer specified in `package/feeds/packages/lttng-ust/Makefile`!**

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-14, ca. 12:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`config.txt`](https://github.com/openwrt/packages/files/9120198/config.lttng-ust-build-failure.txt).
* `grep -i lttng .config`:  
  ```
  CONFIG_PACKAGE_lttng-tools=m
  CONFIG_PACKAGE_lttng-ust=m
  ```

### Description:

After a `make dirclean`, a subsequent `make -j6` fails in above configuration on `package/feeds/packages/lttng-ust`. After a `make package/feeds/packages/lttng-ust/clean`, an explicit run of  
`make -j1 V=sc package/feeds/packages/lttng-ust/compile`
fails with  
```
libtool: relink: ccache_cc -shared  -fPIC -DPIC  -Wl,--whole-archive ../snprintf/.libs/libustsnprintf.a ../liblttng-ust-comm/.libs/liblttng-ust-comm.a ./.libs/liblttng-ust-runtime.a ./.libs/liblttng-ust-support.a -Wl,--no-whole-archive  -L/home/felics/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/home/felics/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -L/home/felics/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lttng-ust-2.12.0/ipkg-install/usr/lib -L/usr/lib -llttng-ust-tracepoint -lrt -ldl -lurcu-bp -lurcu-cds -lpthread  -O3 -g0 -mfloat-abi=hard -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro -specs=/home/felics/download/Router-OS/OpenWRT/OpenWRT/openwrt-22.03/include/hardened-ld-pie.specs   -Wl,-soname -Wl,liblttng-ust.so.0 -o .libs/liblttng-ust.so.0.0.0
/usr/lib/liburcu-bp.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
libtool:   error: error: relink 'liblttng-ust.la' with the above command before installing it
make[5]: *** [Makefile:570: install-libLTLIBRARIES] Error 1
```

Full output of above `make` command is too long to be pasted here, so it is attached: [&rarr;&nbsp;`lttng-ust.compile.log`](https://github.com/openwrt/packages/files/9120254/lttng-ust.compile.log).

felixbecker2, 2022-07-18:
\**ping*\* @ja-pa, @neheb as advised by \[[&rarr;&nbsp;this comment](/openwrt/packages/issues/18940#issuecomment-1185541448)\].

dreirund, 2022-07-22:
It succeeds to compile when to the package's `Makefile` (`package/feeds/packages/lttng-ust/Makefile`) the following line is added:  
```
PKG_FIXUP:=autoreconf
```

[&rarr;&nbsp;Patch](https://github.com/openwrt/packages/files/9167436/lttng-ust.autoreconf-fixup.Makefile.patch.txt):  
```
--- Makefile.org	2022-07-22 12:16:00.559239425 +0200
+++ Makefile	2022-07-22 12:16:43.049238096 +0200
@@ -22,2 +22,3 @@
 
+PKG_FIXUP:=autoreconf
 PKG_USE_MIPS16:=0
```

Feel free to use this patch or your own way to incorporate the change.

Regards!

dreirund, 2022-08-23:
The problem is _still_ present in 22.03 with update to latest commits on 2022-08-23, ca. 08:15 UTC.

neheb, 2022-08-23:
Nobody is making a PR, that's what.

dreirund, 2022-08-23:
[this](https://github.com/openwrt/packages/pull/19244) should fix the issue. Please check.

dreirund, 2022-08-23:
> *Nobody is making a PR, that's what*

It's a bit cumbersome and messy for me to work with branches and make pull requests and so on (I don't know much of git). So I provided a [patch file](https://github.com/openwrt/packages/files/9167436/lttng-ust.autoreconf-fixup.Makefile.patch.txt), and the maintainers/ developers with write access here can directly make the changes in that way or another way that suits better your style and conventions. I thought providing information how things can be fixed is enough to do for a reporter of an issue?

I now [tried with a pull request](/openwrt/packages/pull/19244), but chances are high I made things not the way you would like to have it here. Feel free to just make the changes yourself.