openwrt/packages #18974: `libextractor`: Fails to build while relinking with `/usr/lib/libz.so: file not recognized: file format not recognized` in 22.03.

felixbecker2, 2022-07-18:
### Maintainer:

@dangowrt

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-14, ca. 12:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`config.txt`](https://github.com/openwrt/packages/files/9133807/config.libextractor-builderror.txt).
* `grep -i libextractor .config`:  
  ```
  CONFIG_PACKAGE_libextractor=m
  CONFIG_PACKAGE_libextractor-plugin-archive=m
  CONFIG_PACKAGE_libextractor-plugin-deb=m
  CONFIG_PACKAGE_libextractor-plugin-dvi=m
  CONFIG_PACKAGE_libextractor-plugin-flac=m
  CONFIG_PACKAGE_libextractor-plugin-gif=m
  CONFIG_PACKAGE_libextractor-plugin-gstreamer=m
  CONFIG_PACKAGE_libextractor-plugin-it=m
  CONFIG_PACKAGE_libextractor-plugin-jpeg=m
  CONFIG_PACKAGE_libextractor-plugin-man=m
  CONFIG_PACKAGE_libextractor-plugin-mime=m
  CONFIG_PACKAGE_libextractor-plugin-mpeg=m
  CONFIG_PACKAGE_libextractor-plugin-nsf=m
  CONFIG_PACKAGE_libextractor-plugin-nsfe=m
  CONFIG_PACKAGE_libextractor-plugin-odf=m
  CONFIG_PACKAGE_libextractor-plugin-ogg=m
  CONFIG_PACKAGE_libextractor-plugin-png=m
  CONFIG_PACKAGE_libextractor-plugin-ps=m
  CONFIG_PACKAGE_libextractor-plugin-riff=m
  CONFIG_PACKAGE_libextractor-plugin-s3m=m
  CONFIG_PACKAGE_libextractor-plugin-sid=m
  CONFIG_PACKAGE_libextractor-plugin-tiff=m
  CONFIG_PACKAGE_libextractor-plugin-wav=m
  CONFIG_PACKAGE_libextractor-plugin-xm=m
  CONFIG_PACKAGE_libextractor-plugin-zip=m
  ```

### Description:

A `make -j6` fails in above configuration on `package/feeds/packages/libextractor`. After a `make package/feeds/packages/libextractor/clean`, an explicit run of  
`make -j1 V=sc package/feeds/packages/libextractor/compile`  
fails while relinking with `/usr/lib/libz.so: file not recognized: file format not recognized`:  
```
[...]
libtool: relink: /[...]/openwrt-22.03/staging_dir/host/bin/grep -E -e "(EXTRACTOR|pl)_[a-zA-Z0-9_]*_......." ".libs/libextractor_png.exp" > ".libs/libextractor_png.expT"
libtool: relink: mv -f ".libs/libextractor_png.expT" ".libs/libextractor_png.exp"
libtool: relink: echo "{ global:" > .libs/libextractor_png.ver
libtool: relink:  cat .libs/libextractor_png.exp | sed -e "s/\(.*\)/\1;/" >> .libs/libextractor_png.ver
libtool: relink:  echo "local: *; };" >> .libs/libextractor_png.ver
libtool: relink:  ccache_cc -shared  -fPIC -DPIC  .libs/png_extractor.o   -L/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -L/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/lib -L/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/lib -liconv -L/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libextractor-1.11/ipkg-install/usr/lib -L/usr/lib -lextractor_common -lz  -O3 -g0 -mfloat-abi=hard -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wl,-Bsymbolic -specs=/[...]/openwrt-22.03/include/hardened-ld-pie.specs -Wl,-rpath-link=/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/lib   -Wl,-soname -Wl,libextractor_png.so -Wl,-version-script -Wl,.libs/libextractor_png.ver -o .libs/libextractor_png.so
/usr/lib/libz.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
libtool:   error: error: relink 'libextractor_png.la' with the above command before installing it
[...]
```

Full output of the `make` run: [&rarr;&nbsp;`libextractor-compile.log`](https://github.com/openwrt/packages/files/9133805/libextractor-compile.log)

felixbecker2, 2022-07-19:
> Broken libtool, package probably needs `PKG_FIXUP:=autoreconf` or `PKG_FIXUP:=patch-libtool`

I can confirm that adding either `PKG_FIXUP:=autoreconf` or `PKG_FIXUP:=patch-libtool` to the top of `package/feeds/packages/libextractor/Makefile` solves this build issue for me.  
Here is a patch to the `Makefile`: [`Makefile.libtool-workaround.patch.txt`](https://github.com/openwrt/packages/files/9140291/Makefile.libtool-workaround.patch.txt).

So I ask that for this change (or a similar change which fits more in the way `Makefile`s are written for OpenWrt) to be applied.

---

And I also want to add that the build issue also appears when I have unselected all the `CONFIG_PACKAGE_libextractor-plugin-*`-plugins, including `CONFIG_PACKAGE_libextractor-plugin-png`.

neheb, 2022-07-20:
```
rg libdir=/usr/lib
taglib_c.pc
3:libdir=/usr/lib

openblas.pc
1:libdir=/usr/lib

xtables.pc
5:xtlibdir=/usr/lib/iptables
```

No way it's taglib_c.

neheb, 2022-07-20:
Backported fix.

closed 2022-07-20 by neheb

felixbecker2, 2022-07-20:
> ```
> rg libdir=/usr/lib
> taglib_c.pc
> 3:libdir=/usr/lib
> 
> openblas.pc
> 1:libdir=/usr/lib
> 
> xtables.pc
> 5:xtlibdir=/usr/lib/iptables
> ```
> 
> 
>     
>   
> 
> No way it's taglib_c.

Was this comment meant for this issue? Or accidentally posted in the wrong place?

I cannot see any connection of your comment with this issue.

felixbecker2, 2022-07-20:
> ```
> rg libdir=/usr/lib
> taglib_c.pc
> 3:libdir=/usr/lib
> 
> openblas.pc
> 1:libdir=/usr/lib
> 
> xtables.pc
> 5:xtlibdir=/usr/lib/iptables
> ```
> 
> No way it's taglib_c.

I do not see how your comment relates with this issue.

Could it be that you have accidentally postet your comment at the wrong place?