openwrt/packages #18975: `lcdproc`: Fails to build due to missing driver-`.so`-files in 22.03.

felixbecker2, 2022-07-18:
### Maintainer:

@haraldg

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-14, ca. 12:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`config.txt`](https://github.com/openwrt/packages/files/9133845/config.lcdproc-builderror.txt).
* `grep -i lcdproc .config`:  
  ```
  CONFIG_PACKAGE_lcdproc-clients=y
  CONFIG_PACKAGE_lcdproc-drivers=y
  CONFIG_PACKAGE_lcdproc-server=y
  ```

### Description:

A `make -j6` fails in above configuration on `package/feeds/packages/lcdproc`. After a `make package/feeds/packages/lcdproc/clean`, an explicit run of  
`make -j1 V=sc package/feeds/packages/lcdproc/compile`  
fails with missing driver `.so` files: `cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/sdeclcd.so': No such file or directory`:  
```
[...]
cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/curses.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/futaba.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/glcd.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/hd44780.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/IOWarrior.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/i2500vfd.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/lis.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/picolcd.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/shuttleVFD.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/ula200.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/sdeclcd.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/sed1330.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/sed1520.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/stv5730.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/t6963.so /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/ipkg-arm_cortex-a7_neon-vfpv4/lcdproc-drivers/usr/lib/lcdproc/
cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/sdeclcd.so': No such file or directory
cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/sed1330.so': No such file or directory
cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/sed1520.so': No such file or directory
cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/stv5730.so': No such file or directory
cp: cannot stat '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers/t6963.so': No such file or directory
make[2]: *** [Makefile:165: /[...]/openwrt-22.03/bin/packages/arm_cortex-a7_neon-vfpv4/packages/lcdproc-drivers_0.5.9-6_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/utils/lcdproc'
[...]
```

Full output of the `make` run: [&rarr;&nbsp;`lcdproc-compile.log`]
(https://github.com/openwrt/packages/files/9133836/lcdproc-compile.log)

pprindeville, 2022-07-26:
See PR #19055... tester needed.

dreirund, 2022-07-29:
> See PR #19055... tester needed.

I can't test due to lack of parallel port on my device.

-- 
This email has been sent from a computer.

pprindeville, 2022-07-29:
> See PR #19055... tester needed.
> I can't test due to lack of parallel port on my device.
> […](#)

I don't believe this runs on ARM hardware.

haraldg, 2022-07-29:
> I don't believe this runs on ARM hardware.

I use lcdproc only on ARM machines. But of course without an parallel port.

pprindeville, 2022-07-30:
Fixed with PR #19055

closed 2022-07-30 by pprindeville

pprindeville, 2022-07-30:
Will remove x86 restriction when https://github.com/lcdproc/lcdproc/issues/197 gets resolved.

dreirund, 2022-08-03:
> Fixed with PR #19055

I don't see it fixed.

I manually backported the changes mentioned there into the `Makefile` (except for the x86 restriction).

Then, build fails with

```
[...]
ccache_cc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I/[...]T/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/[...]T/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/[...]T/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include  -I/[...]T/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I/[...]T/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libusb-1.0   -fPIC -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/[...]T/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9=lcdproc-0.5.9 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT glcd-glcd-t6963.o -MD -MP -MF .deps/glcd-glcd-t6963.Tpo -c -o glcd-glcd-t6963.o `test -f 'glcd-t6963.c' || echo './'`glcd-t6963.c
mv -f .deps/glcd-glcd-t6963.Tpo .deps/glcd-glcd-t6963.Po
ccache_cc -DHAVE_CONFIG_H -I. -I../..  -I../.. -I/[...]T/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/[...]T/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/[...]T/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include  -fPIC -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/[...]T/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9=lcdproc-0.5.9 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT t6963_low.o -MD -MP -MF .deps/t6963_low.Tpo -c -o t6963_low.o t6963_low.c
In file included from t6963_low.c:33:
t6963_low.c: In function 't6963_low_dsp_ready':
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:344:9: error: impossible constraint in 'asm'
  344 |         __asm__ volatile ("inb %1,%0":"=a" (value)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:344:9: error: impossible constraint in 'asm'
  344 |         __asm__ volatile ("inb %1,%0":"=a" (value)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
port.h:351:9: error: impossible constraint in 'asm'
  351 |         __asm__ volatile ("outb %0,%1\n"::"a" (val), "d"(port)
      |         ^~~~~~~
make[6]: *** [Makefile:1109: t6963_low.o] Error 1
make[6]: Leaving directory '/[...]T/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server/drivers'
make[5]: *** [Makefile:485: all-recursive] Error 1
make[5]: Leaving directory '/[...]T/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/server'
make[4]: *** [Makefile:465: all-recursive] Error 1
make[4]: Leaving directory '/[...]T/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9'
make[3]: *** [Makefile:385: all] Error 2
make[3]: Leaving directory '/[...]T/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9'
make[2]: *** [Makefile:166: /[...]T/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/lcdproc-0.5.9/.built] Error 2
make[2]: Leaving directory '/[...]T/openwrt-22.03/feeds/packages/utils/lcdproc'
time: package/feeds/packages/lcdproc/compile#13.91#5.88#21.49
    ERROR: package/feeds/packages/lcdproc failed to build.
[...]
```

Can you move out the x86 specific drivers to a seperate package `lcdproc-drivers-x86`, make that depend on x86 architecture, and disable all x86-specific code in the generic `lcdproc` packages?