openwrt/packages #18978: kmod-ath10k: QCA6174 WiFi adapter not functional

developer422, 2022-07-19:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: Felix Fietkau <nbd@nbd.name>
Environment:  **bcm27xx/bcm2711** OpenWrt 22.03.0-rc5

I get the following errors in my **dmesg** output:

```
[    7.347064] ath10k_pci 0000:03:00.0: failed to alloc CE src ring 0: -12
[    7.360318] ath10k_pci 0000:03:00.0: failed to allocate copy engine pipe 0: -12
[    7.367628] ath10k_pci 0000:03:00.0: failed to allocate copy engine pipes: -12
[    7.374848] ath10k_pci 0000:03:00.0: failed to setup resource: -12
[    7.381143] ath10k_pci: probe of 0000:03:00.0 failed with error -12

```
when I try to setup a Qualcomm Atheros QCA6174 802.11ac Wireless Network Adapter on a Raspberry Pi CM4 board.

My starting point is building OpenWrt 22.03.0-rc5 with the following packages enabled: **kmod-ath10k** and **ath10k-firmware-qca6174**

Acting on a tip, I made the two changes in this commit: [linux-firmware: Update to version 20220509 ](https://github.com/openwrt/openwrt/commit/25b29b53f01d3053bd7c8ef68b0bd7dcdb195913) to my Makefile.
However, I still get the same error messages in my system log and the QCA6174 WiFi adapter not functional.

felixbecker2, 2022-07-20:
> *Maintainer: Felix Fietkau nbd(at)nbd.name*

*\*ping\* @nbd168.*

BKPepe, 2022-07-23:
Please take a look at my response here: https://github.com/openwrt/packages/issues/19026#issuecomment-1193191088

~ Closing. kmod-ath10k is not located in the packages repository.

closed 2022-07-23 by BKPepe