openwrt/packages #18980: `mutt`: Fails to build with missing dependencies for `libcrypto.so.1.1`, `iblmdb.so`, `libsasl2.so.3` and `libssl.so.1.1` in 22.03.

felixbecker2, 2022-07-19:
### Maintainer:

@philenotfound

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-14, ca. 12:00 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`config.txt`](https://github.com/openwrt/packages/files/9140489/config.mutt-builderror.txt).

* `grep -i mutt .config`:  
  ```
  CONFIG_PACKAGE_mutt=m
  CONFIG_MUTT_POP=y
  CONFIG_MUTT_IMAP=y
  CONFIG_MUTT_SMTP=y
  CONFIG_MUTT_HCACHE=y
  CONFIG_MUTT_SASL=y
  # CONFIG_MUTT_GNUTLS is not set
  CONFIG_MUTT_OPENSSL=y
  ```
* `grep -i lmdb .config`:  
  ```
  CONFIG_PACKAGE_lmdb=m
  # CONFIG_PACKAGE_lmdb-test is not set
  # CONFIG_PACKAGE_lmdb-utils is not set
  ```

### Description:

A `make -j6` fails in above configuration on `package/feeds/packages/mutt`. After a `make package/feeds/packages/mutt/clean`, an explicit run of  
`make -j1 V=sc package/feeds/packages/mutt/compile`  
fails with
```
[...]
Package mutt is missing dependencies for the following libraries:
liblmdb.so
[...]
```

<details><summary>More context of the 'make' output (*click* to open):</summary>

```
[...]
make  install-exec-hook
make[7]: Entering directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5'
if test -f /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt.dotlock && test -f /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt_dotlock ; then	\
	rm -f /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt.dotlock ;		\
	ln -sf /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt_dotlock /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt.dotlock ; \
fi
if test -f /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt_dotlock && test x != x ; then \
	chgrp  /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt_dotlock && \
	chmod 755 /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt_dotlock || \
	{ echo "Can't fix mutt_dotlock's permissions!  This is required to lock mailboxes in the mail spool directory." >&2 ; exit 1 ; } \
fi
make[7]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5'
/usr/bin/mkdir -p /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/etc
/[...]/openwrt-22.03/staging_dir/host/bin/install -c -m 644 ./mime.types /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/etc/mime.types.dist
if [ ! -f /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/etc/mime.types ]; then \
	/[...]/openwrt-22.03/staging_dir/host/bin/install -c -m 644 ./mime.types /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/etc; \
fi
make[6]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5'
make[5]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5'
make[4]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5'
make[3]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5'
touch /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.built
rm -rf /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.pkgdir/mutt.installed /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.pkgdir/mutt
mkdir -p /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.pkgdir/mutt
install -d -m0755 /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.pkgdir/mutt/usr/bin
install -m0755 /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.pkgdir/mutt/usr/bin/
touch /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.pkgdir/mutt.installed
mkdir -p /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/stamp
SHELL= flock /[...]/openwrt-22.03/tmp/.root-copy.flock -c 'cp -fpR /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/.pkgdir/mutt/. /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/'
touch /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/stamp/.mutt_installed
mkdir -p /[...]/openwrt-22.03/bin/targets/sunxi/cortexa7/packages /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-arm_cortex-a7_neon-vfpv4/mutt/CONTROL /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/pkginfo
install -d -m0755 /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-arm_cortex-a7_neon-vfpv4/mutt/usr/bin
install -m0755 /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-install/usr/bin/mutt /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-arm_cortex-a7_neon-vfpv4/mutt/usr/bin/
find /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/mutt-2.1.5/ipkg-arm_cortex-a7_neon-vfpv4/mutt -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package mutt is missing dependencies for the following libraries:
liblmdb.so
make[2]: *** [Makefile:76: /[...]/openwrt-22.03/bin/packages/arm_cortex-a7_neon-vfpv4/packages/mutt_2.1.5-1_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/mail/mutt'
time: package/feeds/packages/mutt/compile#17.95#5.09#23.15
    ERROR: package/feeds/packages/mutt failed to build.
[...]
```
</details>

Full output of the abovementioned `make` run: [&rarr;&nbsp;`mutt.compile.log`](https://github.com/openwrt/packages/files/9140487/mutt.compile.log).

Error persists even if I manually issue a `make package/feeds/packages/lmdb/compile` before.

If I disable "*Header caching support*" (changing `CONFIG_MUTT_HCACHE=m` to `# CONFIG_MUTT_HCACHE is not set` via `make menuconfig`) does not solve this build error. If I disable all the optional mutt settings, so that `grep -i mutt .config` looks like  
```
CONFIG_PACKAGE_mutt=m
# CONFIG_MUTT_POP is not set
# CONFIG_MUTT_IMAP is not set
# CONFIG_MUTT_SMTP is not set
# CONFIG_MUTT_HCACHE is not set
# CONFIG_MUTT_SASL is not set
# CONFIG_MUTT_GNUTLS is not set
# CONFIG_MUTT_OPENSSL is not set
```
I even get more errors:  
```
[...]
Package mutt is missing dependencies for the following libraries:
libcrypto.so.1.1
liblmdb.so
libsasl2.so.3
libssl.so.1.1
[...]
```

closed 2022-07-20 by neheb

dreirund, 2022-07-21:
> *@neheb closed this as completed in #18986 16 hours ago*

I have to add that there still is one problem:

When `CONFIG_MUTT_HCACHE=y` ("*Header caching support*"), then build still fails for me with

```
Package mutt is missing dependencies for the following libraries:
liblmdb.so
```

When `# CONFIG_MUTT_HCACHE is not set` build succeeds.

neheb, 2022-07-22:
strange. CONFIG_MUTT_HCACHE is is PKG_CONFIG_DEPENDS

dhewg, 2022-07-22:
The configure script seems to prefer lmdb instead of gdmb, see also https://wiki.gentoo.org/wiki/Mutt#Header_cache_backends

So I guess switch from gdmb to lmdb via --enable-lmdb --disable-gdmb?