openwrt/packages #18982: transmission: Update package into transmission git

trippleflux, 2022-07-19:
Maintainer: Daniel Golle <daniel@makrotopia.org>
Environment: OpenWRT latest snapshot x86_64

Description:
I am trying to update transmission package into using the transmission git, because of build system changes into using cmake, I have encounter the following issue :
```
transmission-3.00.tar.xz: Download from https://github.com/transmission/transmission.git failed
transmission-3.00.tar.xz: Requires sha256sum for verification
Checking out files from the git repository...
Cloning into 'transmission-3.00'...
remote: Enumerating objects: 115422, done.
remote: Counting objects: 100% (468/468), done.
remote: Compressing objects: 100% (229/229), done.
remote: Total 115422 (delta 293), reused 364 (delta 237), pack-reused 114954
Receiving objects: 100% (115422/115422), 105.17 MiB | 1.51 MiB/s, done.
Resolving deltas: 100% (89968/89968), done.
Note: switching to '94eeebc5b309f38b648a179612bcee4e1c05c73f'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 94eeebc5b Update Scripts.md (#3475)
Submodule 'third-party/arc4' (https://github.com/transmission/arc4) registered for path 'third-party/arc4'
Submodule 'third-party/dht' (https://github.com/transmission/dht) registered for path 'third-party/dht'
Submodule 'third-party/fast_float' (https://github.com/transmission/fast_float) registered for path 'third-party/fast_float'
Submodule 'third-party/fmt' (https://github.com/transmission/fmt.git) registered for path 'third-party/fmt'
Submodule 'third-party/googletest' (https://github.com/google/googletest.git) registered for path 'third-party/googletest'
Submodule 'third-party/libb64' (https://github.com/transmission/libb64) registered for path 'third-party/libb64'
Submodule 'third-party/libdeflate' (https://github.com/transmission/libdeflate) registered for path 'third-party/libdeflate'
Submodule 'third-party/libevent' (https://github.com/transmission/libevent) registered for path 'third-party/libevent'
Submodule 'third-party/libnatpmp' (https://github.com/transmission/libnatpmp) registered for path 'third-party/libnatpmp'
Submodule 'third-party/libpsl' (https://github.com/transmission/libpsl.git) registered for path 'third-party/libpsl'
Submodule 'third-party/libutp' (https://github.com/transmission/libutp) registered for path 'third-party/libutp'
Submodule 'third-party/miniupnpc' (https://github.com/transmission/miniupnpc) registered for path 'third-party/miniupnpc'
Submodule 'third-party/utfcpp' (https://github.com/transmission/utfcpp) registered for path 'third-party/utfcpp'
Submodule 'third-party/wide-integer' (https://github.com/transmission/wide-integer) registered for path 'third-party/wide-integer'
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/arc4'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/dht'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/fast_float'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/fmt'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/googletest'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/libb64'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/libdeflate'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/libevent'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/libnatpmp'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/libpsl'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/libutp'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/miniupnpc'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/utfcpp'...
Cloning into '/home/username/works/openwrt/tmp/dl/transmission-3.00/third-party/wide-integer'...
Submodule path 'third-party/arc4': checked out '6c8a5dd7e5e3fafcf0afef74cdf71f9d20cb9a54'
Submodule path 'third-party/dht': checked out '015585510e402a057ec17142711ba2b568b5fd62'
Submodule path 'third-party/fast_float': checked out 'f2082bf747eabc0873f2fdceb05f9451931b96dc'
Submodule path 'third-party/fmt': checked out 'c4ee726532178e556d923372f29163bd206d7732'
Submodule path 'third-party/googletest': checked out 'af29db7ec28d6df1c7f0f745186884091e602e07'
Submodule path 'third-party/libb64': checked out '8456a5b5c99ef4544a2ed8e8f20a2fb673dd4192'
Submodule path 'third-party/libdeflate': checked out '45029bc1bc42ff55a0a8d4223ba9d9e8ec0da0d3'
Submodule path 'third-party/libevent': checked out 'e91a51eb5daba37418cc8426bd1f786f303ddbb1'
[... 169 lines cut ...]
	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/transmission-cli \
			$(PKG_INSTALL_DIR)/usr/bin/transmission-create \
			$(PKG_INSTALL_DIR)/usr/bin/transmission-edit \
			$(PKG_INSTALL_DIR)/usr/bin/transmission-show \
			$(1)/usr/bin/
endef

define Package/transmission-remote/install
	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/transmission-remote $(1)/usr/bin/
endef

define Package/transmission-web/install
	$(INSTALL_DIR) $(1)/usr/share/transmission
	$(CP) $(PKG_INSTALL_DIR)/usr/share/transmission/web $(1)/usr/share/transmission/
endef

$(eval $(call BuildPackage,transmission-daemon))
$(eval $(call BuildPackage,transmission-cli))
$(eval $(call BuildPackage,transmission-remote))
$(eval $(call BuildPackage,transmission-web))
```

In transmission CmakeList.txt file, there are no occurrence regarding download dht as error above. Still confuse on how to proceed.

trippleflux, 2022-07-19:
transmission author mentioned that I should separate build dir from source dir, how to call "cmake .." in a dir?, because the following doesn't work : 
```
define Build/Prepare
	mkdir -p $(PKG_BUILD_DIR)/build
	cd $(PKG_BUILD_DIR)/build
	$(Build/Configure/Default, ..)
endef
```

trippleflux, 2022-07-20:
Managed to get it working, currently only compile-tested on x86_64, haven't run it yet as I am still compiling my x86_64 OpenWRT image
[transmission-git.zip](https://github.com/openwrt/packages/files/9146019/transmission-git.zip)

Replace the feeds/packages/net/transmission with archive above.

**_Feel free to make a pull request based on work._**

Credits & Thanks goes for : 
@mikedld in transmission github

neheb, 2022-07-20:
you're looking for CMAKE_BINARY_SUBDIR.

Anyway, I hate this cmake build system. Everything seems to get linked statically, which blows up size under OpenWrt.

trippleflux, 2022-07-20:
Yup, managed to figuring it out after trials and errors and searching the openwrt packages github, thanks anyway.

The complete and working compile-tested is inside the [transmission-git.zip](https://github.com/openwrt/packages/files/9146019/transmission-git.zip) , incase you've missed.

Just realize and forgot mostly we run OpenWRT on embedded devices with limited storage, hmm, if I am not mistaken the transmission-daemon based on my package above is at 67 MB o_0; .

Well if we going for shared libs ways, then perhaps we need to separate the 3rd party libs into 5 (libb64, libdht, libutp, libdeflate, libnatpmp, etc?) different packages?.

neheb, 2022-07-20:
I doubt it. The autotools based system made properly linked size binaries. Something is wrong with the cmake build.

trippleflux, 2022-07-20:
I need to commenting in these line in order to make it successfully linking libraries: 
```
#TARGET_CFLAGS += -ffunction-sections -fdata-sections -flto
#TARGET_LDFLAGS += -Wl,--gc-sections -Wl,--as-needed
```

Perhaps somehow it messed up with the executables stripping process ?

mikedld, 2022-07-20:
> Anyway, I hate this cmake build system. Everything seems to get linked statically, which blows up size under OpenWrt.

Only if the appropriate package wasn't found in the system. There's a way to force it to only look for system libraries and not fallback to the bundled ones by setting these options to `ON` instead of default `AUTO`: https://github.com/transmission/transmission/blob/f8b3514c0e13329be8d6104f3764d01288a69246/CMakeLists.txt#L41-L48

Usually most of those packages are in fact provided by the system, and only a small set of quite small libraries (b64, dht) are linked in statically.

mikedld, 2022-07-20:
> Just realize and forgot mostly we run OpenWRT on embedded devices with limited storage, hmm, if I am not mistaken the transmission-daemon based on my package above is at 67 MB o_0; .

> I doubt it. The autotools based system made properly linked size binaries. Something is wrong with the cmake build.

You're setting `CMAKE_BUILD_TYPE` to `RelWithDebInfo` which means that the resulting binaries (unless stripped) will contain debug information embedded in them, and that makes up most of the their size. There is no way the actual executable code is that large.

By contrast, old autotools-based system defaulted to release build without any debugging information (https://github.com/transmission/transmission/blob/3.00/configure.ac#L23-L32). If you don't use (distribute) debugging information, this could be achieved with CMake by using `Release` or `MinSizeRel` build type.

trippleflux, 2022-07-20:
Yes, I overlook that build type, now transmission-daemon goes down into 2.7 MB. Based on [here](https://stackoverflow.com/questions/48754619/what-are-cmake-build-type-debug-release-relwithdebinfo-and-minsizerel), I have a guess Release probably better, Speed VS Optimized Size.

neheb, 2022-07-20:
https://github.com/openwrt/openwrt/blob/master/include/cmake.mk#L95

Where does RelWithDebInfo come from?

mikedld, 2022-07-20:
It's there in your own original post, in `CMAKE_OPTIONS`.

trippleflux, 2022-07-21:
Probably transmission.init needs a rewrites or fixes, because I have problem running transmission daemon with it. It always crashes!.

Need to use luci-app-transmission and then start transmission-daemon manually using sudo : 
```
sudo -u transmission /usr/bin/transmission-daemon --config-dir /tmp/transmission
```

neheb, 2022-07-21:
@trippleflux that crashing might be related to wolfssl actually. I'll probably which up a PR moving it back to mbedtls one of these days.

trippleflux, 2022-07-22:
Hmm I am using OpenSSL for curl backend, have you tested my transmission git package?, does in your box transmission-daemon is also crashes when using luci-startup or init.d ?.

neheb, 2022-07-22:
I don't use transmission anymore unfortunately.

trippleflux, 2022-07-22:
I guess we need other OpenWRT devs/contributors which using it, I am really sure something wrong with the transmission init script for the transmission git version.

sudo workaround is working but will be nice if using init.d.

trippleflux, 2022-07-22:
Found the fixes, need to remove seccomp from the init script!.

Attached below the revised package : 
[transmission.zip](https://github.com/openwrt/packages/files/9172252/transmission.zip)

neheb, 2024-06-09:
There have been many changes to transmission including a port to C++. Post the issue again if still problematic.

closed 2024-06-09 by neheb