openwrt/packages #18988: Rootless docker on OpenWrt 21.02.3

gheist, 2022-07-20:
Maintainer: @G-M0N3Y-2503 
Environment: aarch64 Cortex-A53 21.03.2

Description:
Trying to run rootless docker on the latest stable release. The feature is intended to allow non-privileged users to run Docker containers.
When trying to run rootless docker the following error is observed:

toor@OpenWrt:~$  docker run -it --rm arm64v8/busybox
docker: Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "proc" to rootfs at "/proc": mount proc:/proc (via /proc/self/fd/7), flags: 0xe: operation not permitted: unknown.

Dockerd is run as follows:
rootlesskit '--net=lxc-user-nic' '--mtu=1500' '--slirp4netns-sandbox=auto' '--slirp4netns-seccomp=auto' --disable-host-loopback '--port-driver=builtin' '--copy-up=/etc' '--copy-up=/run' '--propagation=rslave' '--lxc-user-nic-binary=/usr/lib/lxc/lxc-user-nic' --lxc-user-nic-bridge br-lan --debug --cgroupns /home/toor/bin/dockerd-rootless.sh --debug '--iptables=false'

Strace of dockerd shows that dockerd is trying to mount the container image opened via procfs (/proc/self/fd/7) and fails, despite running as "fake root" 

7885  mount("/home/toor/.local/share/docker/vfs/dir/28e1713fe7e5bb5cbdee5268a23bb19a41c2a7a699e408a147987e62a309e119", "/home/toor/.local/share/docker
7885  <... mount resumed>)              = 0                                                                                                           
7885  newfstatat(AT_FDCWD, "/home/toor/.local/share/docker/vfs/dir/28e1713fe7e5bb5cbdee5268a23bb19a41c2a7a699e408a147987e62a309e119/proc",  <unfinishe
7885  <... newfstatat resumed>{st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_SYMLINK_NOFOLLOW) = 0                                                     
7885  newfstatat(AT_FDCWD, "/home/toor/.local/share/docker/vfs/dir/28e1713fe7e5bb5cbdee5268a23bb19a41c2a7a699e408a147987e62a309e119/proc",  <unfinishe
7885  <... newfstatat resumed>{st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_SYMLINK_NOFOLLOW) = 0                                                     
7885  newfstatat(AT_FDCWD, "/home/toor/.local/share/docker/vfs/dir/28e1713fe7e5bb5cbdee5268a23bb19a41c2a7a699e408a147987e62a309e119/proc",  <unfinishe
7885  <... newfstatat resumed>{st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0                                                                       
7885  newfstatat(AT_FDCWD, "/home/toor/.local/share/docker/vfs/dir/28e1713fe7e5bb5cbdee5268a23bb19a41c2a7a699e408a147987e62a309e119/proc",  <unfinishe
7885  <... newfstatat resumed>{st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_SYMLINK_NOFOLLOW) = 0                                                     
7885  futex(0xcda550, FUTEX_WAIT_PRIVATE, 0, NULL <unfinished ...>                                                                                    
7885  <... futex resumed>)              = 0                                                                                                           
**7885  openat(AT_FDCWD, "/home/toor/.local/share/docker/vfs/dir/28e1713fe7e5bb5cbdee5268a23bb19a41c2a7a699e408a147987e62a309e119/proc", O_RDONLY|O_CLOE
7885  <... openat resumed>)             = 7**                                                                                                           
7885  epoll_ctl(8, EPOLL_CTL_ADD, 7, {EPOLLIN|EPOLLOUT|EPOLLRDHUP|EPOLLET, {u32=1746532184, u64=547207378776}} <unfinished ...>                       
7885  <... epoll_ctl resumed>)          = -1 EBADF (Bad file descriptor)          
7885  readlinkat(AT_FDCWD, "/proc/self/fd/7",  <unfinished ...>                                                                                       
7885  <... readlinkat resumed>"/home/toor/.local/share/docker/v"..., 128) = 108                                                                       
**7885  mount("proc", "/proc/self/fd/7", "proc", MS_NOSUID|MS_NODEV|MS_NOEXEC, NULL <unfinished ...>                                                    
7885  <... mount resumed>)              = -1 EPERM (Operation not permitted)**                                                                          
                                                      
The issue is likely related to/identical to https://github.com/openwrt/packages/issues/15096

Similar issue has been observed when running docker within lxc, but it appears to be related to apparmor configuration by lxc, and got resolved by the following lxc commit
https://github.com/lxc/lxd/commit/546e2a60809a108a1f505b99c6edbda52b12c739
In our case apparmor (or SELinux) are disabled in the kernel, it's kernel refusing mount even though the user has root uid

**Environment**

Raspberry Pi 3 with latest stable OpenWrt release
Kernel 5.4.188 running cgroups v2

root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.188 #0 SMP Sat Apr 16 12:59:34 2022 aarch64 GNU/Linux
root@OpenWrt:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.3'
DISTRIB_REVISION='r16554-1d4dea6d4f'
DISTRIB_TARGET='bcm27xx/bcm2710'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 21.02.3 r16554-1d4dea6d4f'
DISTRIB_TAINTS=''

Rootless docker is installed as follows:
curl -fsSL https://get.docker.com/rootless | SKIP_IPTABLES=1 sh

Docker uses rootlesskit https://github.com/rootless-containers/rootlesskit for unprivileged execution using user/network/mount namespace capabilities
We are using the setuid lxc-user-nic option for network namespacing and cgroup namespace (--cgroupns flag) for cgroups isolation

**Rootless dockerd output**
toor@OpenWrt:~$ dockerd-rootless.sh --debug --iptables=false  
+ '[' -w /home/toor/.docker/run ]
+ '[' -d /home/toor ]
+ rootlesskit=
+ command -v docker-rootlesskit
[... 418 lines cut ...]
INFO[2022-07-20T08:59:31.643772286Z] shim disconnected                             id=90527e27a9f1fa917228278335b461ad6da1a6ad1caa6d03f80b3b2b5d662c74
WARN[2022-07-20T08:59:31.644223326Z] cleaning up after shim disconnected           id=90527e27a9f1fa917228278335b461ad6da1a6ad1caa6d03f80b3b2b5d662c74 namespace=moby
INFO[2022-07-20T08:59:31.644330252Z] cleaning up dead shim                        
WARN[2022-07-20T08:59:31.675619512Z] cleanup warnings time="2022-07-20T08:59:31Z" level=info msg="starting signal loop" namespace=moby pid=7550 runtime=io.containerd.runc.v2
time="2022-07-20T08:59:31Z" level=warning msg="failed to read init pid file" error="open /home/toor/.docker/run/docker/containerd/daemon/io.containerd.runtime.v2.task/moby/90527e27a9f1fa917228278335b461ad6da1a6ad1caa6d03f80b3b2b5d662c74/init.pid: no such file or directory" runtime=io.containerd.runc.v2 
ERRO[2022-07-20T08:59:31.678005385Z] copy shim log                                 error="read /proc/self/fd/13: file already closed"
ERRO[2022-07-20T08:59:31.680066989Z] stream copy error: reading from a closed fifo 
DEBU[2022-07-20T08:59:31.681207035Z] attach: stdout: end                          
DEBU[2022-07-20T08:59:31.681238024Z] attach: stderr: end                          
DEBU[2022-07-20T08:59:31.681380523Z] attach: stdin: end                           
DEBU[2022-07-20T08:59:31.681685053Z] attach done                                  
DEBU[2022-07-20T08:59:31.706578670Z] event published                               ns=moby topic=/containers/delete type=containerd.events.ContainerDelete
DEBU[2022-07-20T08:59:31.720434273Z] Revoking external connectivity on endpoint serene_sanderson (41a7ecc3abfb2c501c912310683df1c8c2508ea7825039a41c0c7446b4bb0d4d) 
DEBU[2022-07-20T08:59:31.811405247Z] Releasing addresses for endpoint serene_sanderson's interface on network bridge 
DEBU[2022-07-20T08:59:31.811680922Z] ReleaseAddress(LocalDefault/172.17.0.0/16, 172.17.0.2) 
DEBU[2022-07-20T08:59:31.811912900Z] Released address PoolID:LocalDefault/172.17.0.0/16, Address:172.17.0.2 Sequence:App: ipam/default/data, ID: LocalDefault/172.17.0.0/16, DBIndex: 0x0, Bits: 65536, Unselected: 65532, Sequence: (0xe0000000, 1)->(0x0, 2046)->(0x1, 1)->end Curr:3 
DEBU[2022-07-20T08:59:31.817425319Z] garbage collected                             d=5.062837ms
ERRO[2022-07-20T08:59:31.829803430Z] 90527e27a9f1fa917228278335b461ad6da1a6ad1caa6d03f80b3b2b5d662c74 cleanup: failed to delete container from containerd: no such container 
ERRO[2022-07-20T08:59:31.876182922Z] Handler for POST /v1.41/containers
**/90527e27a9f1fa917228278335b461ad6da1a6ad1caa6d03f80b3b2b5d662c74/start returned error: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "proc" to rootfs at "/proc": mount proc:/proc (via /proc/self/fd/7), flags: 0xe: operation not permitted: unknown** 
DEBU[2022-07-20T08:59:31.878696607Z] Closing buffered stdin pipe 

```
Format code blocks by wrapping them with pairs of ```
```

G-M0N3Y-2503, 2022-07-21:
Looking at [the last time I looked into rootlesskit](https://github.com/G-M0N3Y-2503/packages/commit/a189760de91b0f15380025912f9e4c49b776de2e) It looks like I seemed to think that it depended on `systemd`. I don't recall why I thought that, but if that is still the case, since OpenWrt uses `procd` to replace `systemd` it looks like it might be an uphill battle to support it.

Also, not sure if it is relevant to this issue, but it looks https://get.docker.com/rootless downloads a binary, so maybe the binary needs to be compiled with the OpenWrt Buildsystem too.

gheist, 2022-07-22:
I didn't see any direct or claimed dependency
On systemd-based systems (checked on Ubuntu 18.04 LTS), systemd invokes rootlesskit and runs it as a service, otherwise rootlesskit can be run manually. 
I was able to get dockerd started, mount the container and set up container networking, however runc execution of the container image fails because the kernel in OpenWrt rejecting procfs mount in the runc init. It happens with both vfs and fuse-overlayfs storage drivers. 

7885 mount("proc", "/proc/self/fd/7", "proc", MS_NOSUID|MS_NODEV|MS_NOEXEC, NULL <unfinished ...>
7885 <... mount resumed>) = -1 EPERM (Operation not permitted)

Rootlesskit tries and is denied sysfs mount on start, which maybe the reason that the kernel denies that procfs mount later: 
**WARN[0000] The host root filesystem is mounted as "". Setting child propagation to "rslave" is not supported. 
WARN[0000] failed to mount sysfs, falling back to read-only mount: operation not permitted
WARN[0000] failed to mount sysfs: operation not permitted**
_

It is not fully clear to me why 5.4.188 kernel under OpenWrt denies sysfs and procfs mounts 
Maybe the sysfs mount failure in rootlesskit causes procfs mount failure in containerd downstream

Per https://lists.linuxfoundation.org/pipermail/containers/2018-April/038840.html (not sure if this reflects the current kernel mount handling logic with user ns): 
_Since Linux v4.2 with commit 1b852bceb0d1 ("mnt: Refactor the logic for mounting sysfs and proc in a user namespace"), new mounts of proc or sysfs in non init userns are only allowed when there is at least one fully-visible proc or sysfs mount.nor why  is denied

I'll try to set up rootlesskit as a non-privileged procd service, see if that makes a difference
Not sure compiling rootkit as OpenWrt build process would make a difference, but ultimately it would have to be an OpenWrt package for sure
 
One key difference between Ubuntu (5.4.0-122) and OpenWrt (5.4.188) is that rootfs in Ubuntu is mounted as shared  
Ubuntu:
30 1 8:7 / / rw,relatime shared:1 - ext4 /dev/sda7 rw,errors=remount-ro
OpenWrt: 
13 1 179:2 / / rw,noatime - ext4 /dev/root rw

Can this be done in OpenWrt and what are possible implications?