openwrt/packages #18992: https-dns-proxy: Does not work with latest master (r20146-bc43ad88ed)

khanjui, 2022-07-20:
Maintainer: @stangri
Environment: ramips, Xiaomi Mi Router 4A (MIR4A) Gigabit Edition, OpenWrt r20146-bc43ad88ed

Description:

`Wed Jul 20 22:57:18 2022 user.notice https-dns-proxy: Starting service ✓
Wed Jul 20 22:57:18 2022 daemon.err uhttpd[2400]: sh: curl: not found
Wed Jul 20 22:57:21 2022 daemon.err uhttpd[2400]: sh: curl: not found
Wed Jul 20 22:57:21 2022 daemon.notice ttyd[3627]: rops_handle_POLLIN_netlink: DELADDR
Wed Jul 20 22:57:21 2022 daemon.err modprobe: failed to find a module named act_ipt
Wed Jul 20 22:57:22 2022 daemon.err modprobe: failed to find a module named act_ipt
Wed Jul 20 22:57:32 2022 daemon.info https-dns-proxy[30162]: [F]      -25.1658347052 (null):636317 950F: CURLOPT_SSLVERSION error 4: Error
Wed Jul 20 22:57:37 2022 daemon.info https-dns-proxy[30855]: [F]      -25.1658347057 (null):858817 E520: CURLOPT_SSLVERSION error 4: Error
Wed Jul 20 22:57:43 2022 daemon.info https-dns-proxy[30868]: [F]      -25.1658347063 (null):781736 85E6: CURLOPT_SSLVERSION error 4: Error
Wed Jul 20 22:57:49 2022 daemon.info https-dns-proxy[30869]: [F]      -25.1658347069 (null):923718 E520: CURLOPT_SSLVERSION error 4: Error
Wed Jul 20 22:57:58 2022 daemon.info https-dns-proxy[30870]: [F]      -25.1658347078 (null):316064 BD15: CURLOPT_SSLVERSION error 4: Error
Wed Jul 20 22:58:04 2022 daemon.info https-dns-proxy[30871]: [F]      -25.1658347084 (null):665923 F9AA: CURLOPT_SSLVERSION error 4: Error
Wed Jul 20 22:58:04 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
Wed Jul 20 22:58:30 2022 daemon.err uhttpd[2400]: sh: curl: not found
Wed Jul 20 22:58:35 2022 daemon.err uhttpd[2400]: sh: curl: not found
Wed Jul 20 22:58:39 2022 daemon.info dnsmasq[30123]: exiting on receipt of SIGTERM
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: started, version 2.86 cachesize 150
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: DNS service limited to local subnets
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: UBus support enabled: connected to system bus
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq-dhcp[31292]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for test
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for onion
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for localhost
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for local
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for invalid
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for bind
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for use-application-dns.net
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for lan
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using nameserver 1.1.1.1#53
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using nameserver 1.0.0.1#53
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using nameserver 1.1.1.1#53
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using nameserver 1.0.0.1#53
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for test
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for onion
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for localhost
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for local
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for invalid
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for bind
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for use-application-dns.net
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: using only locally-known addresses for lan
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: read /etc/hosts - 4 addresses
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq[31292]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jul 20 22:58:43 2022 daemon.info dnsmasq-dhcp[31292]: read /etc/ethers - 0 addresses
Wed Jul 20 22:58:43 2022 user.notice https-dns-proxy: Stopping service ✓
Wed Jul 20 22:58:43 2022 daemon.err uhttpd[2400]: sh: curl: not found
Wed Jul 20 22:58:46 2022 daemon.notice ttyd[3627]: rops_handle_POLLIN_netlink: DELADDR
Wed Jul 20 22:58:46 2022 daemon.err modprobe: failed to find a module named act_ipt
Wed Jul 20 22:58:46 2022 daemon.err modprobe: failed to find a module named act_ipt`

stangri, 2022-07-20:
You may need to update to a revision of snapshot which includes curl built for the latest wolfssl.

khanjui, 2022-07-20:
Just checked out the commits, I guess that's the case. Thank you.

closed 2022-07-20 by khanjui