openwrt/packages #18997: `php8:` sodium extension is missing

Livy-Inverse, 2022-07-21:
Maintainer: @mhei
Environment: arm_cortex-a7_neon-vfpv4, MobiPromo CM520-79F, 22.03.0-rc5

Description:
My PHP installation requires Composer package manager. While trying to install Composer locally using the command:
```
php-cli composer.phar install --no-dev --prefer-source;
```
It complains about a lot of missing extensions. Most of them can be downloaded from separate packages, except for `sodium`. [According to PHP homepage](https://www.php.net/manual/en/sodium.installation.php), that extension is no longer available since PHP 7.2.0, and should be bundled with PHP by default. I am using 8.1.6-1 and it is not available. Please add support for that extension.

flyn-org, 2022-07-26:
Some time ago I produced a php8-pecl-sodium package for master. The application that I use that optionally relies on sodium is NextCloud, and I find that that application for some reason does not detect the sodium package despite it being installed. So it might be possible to backport php8-pecl-sodium, but there also might be something wrong in that package.

mhei, 2022-08-09:
The php8-pecl-sodium package is also available in 22.03 branch. @Livy-Inverse do you have a chance to test it? To be honest, I don't checked yet in detail, whether there is a difference between the pecl package and the extension shipped with PHP itself. However, I guess that the latter one might be more maintained than the package (last update 2020-12-06) - though since its only a wrapper, more updates might not be necessary...
@flyn-org : What do you think: would it make sense to drop the pecl package in favour of adding a package which bundles the extension?

flyn-org, 2022-08-10:
I have been poking around, and it is not clear to me which module is preferred, the PECL version or the version in PHP proper. To be honest, I have not yet found a clear description of the relationship between the two. The last update to the PECL version seems to have occurred in December of 2020. There are much newer mentions in the PHP proper Git log, so my instinct is to move away from the PECL module.

I am happy to drop the php8-pecl-sodium if the base PHP build provides a sodium module.

mhei, 2022-10-04:
Since we have `php8-pecl-sodium` in `22.03` available, I don't see what here is strictly required to do other than testing by @Livy-Inverse. In master we meanwhile moved away from the pecl package and replaced it with the vanilla packaged extension. I'm not yet sure, whether we should cherry-pick this to the `22.03` branch. Opinions?

esaaprillia, 2023-01-14:
@mhei 
sorry I commented in this thread 
are you helping me?
i want to make php7 in master but i have problem when i build it
```
checking for icu-uc >= 50.1 icu-io icu-i18n... yes                                                 
checking whether we are using the GNU C++ compiler... yes                                          
checking whether mipsel-openwrt-linux-musl-g++ accepts -g... yes                                   
checking how to run the C++ preprocessor... mipsel-openwrt-linux-musl-g++ -E                       
checking whether mipsel-openwrt-linux-musl-g++ supports C++11 features with -std=c++11... yes      
checking whether to enable JavaScript Object Serialization support... yes, shared                  
checking for LDAP support... yes, shared                                                           
checking whether to build with LDAP Cyrus SASL support... yes                                      
checking for 3 arg ldap_set_rebind_proc... no                                                      
checking for ldap_parse_result... no                                                               
checking for ldap_parse_reference... no                                                            
checking for ldap_start_tls_s... no                                                                
checking for ldap_control_find... no                                                               
checking for ldap_parse_extended_result... no                                                      
checking for ldap_extended_operation... no                                                         
checking for ldap_extended_operation_s... no                                                       
checking for ldap_passwd... no                                                                     
checking for ldap_whoami_s... no                                                                   
checking for ldap_refresh_s... no                                                                  
checking for libsasl2... yes                                                                       
checking for ldap_bind_s... no                                                                     
configure: error: LDAP build check failed. Please check config.log for more information.
```

mhei, 2023-01-14:
@esaaprilla: why didn't you create a new thread? looks like the ldap library stuff is missing or could not be detected. please just use php8 with snapshot, or is there a special reason for this old php version?

esaaprillia, 2023-01-14:
> @esaaprilla: why didn't you create a new thread? looks like the ldap library stuff is missing or could not be detected. please just use php8 with snapshot, or is there a special reason for this old php version?

the project I'm working on is still using php7

closed 2023-01-14 by Livy-Inverse

BKPepe, 2023-11-03:
> I'm not yet sure, whether we should cherry-pick this to the 22.03 branch. Opinions?

Sorry, for reviving old thread. Go for it! :)