openwrt/packages #19003: dns over https proxy stopped working in master.

tapper82, 2022-07-21:
Fresh build of master. 
TappersRouter
Model Netgear Nighthawk X4S R7800
Architecture ARMv7 Processor rev 0 (v7l)
Target Platform ipq806x/generic
Firmware Version OpenWrt SNAPSHOT r20146-bc43ad88ed / LuCI Master git-22.189.49133-e193259
Kernel Version 5.15.53

I built dns over https proxy in to my build.
This is with fresh configs.
When I click the start button on the dns over https proxy page it kills DNS.

This is with the proxy off.

/etc/config/https-dns-proxy

config main 'config'
	option force_dns '1'
	list force_dns_port '53'
	list force_dns_port '853'
	option update_dnsmasq_config '0'

config https-dns-proxy
	option listen_addr '127.0.0.1'
	option listen_port '5054'
	option user 'nobody'
	option group 'nogroup'
	option bootstrap_dns '1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001'
	option resolver_url 'https://cloudflare-dns.com/dns-query'

/etc/config/dhcp

config dnsmasq
	option domainneeded '1'
	option boguspriv '1'
	option filterwin2k '0'
	option localise_queries '1'
	option rebind_protection '1'
	option rebind_localhost '1'
	option local '/lan/'
	option domain 'lan'
	option expandhosts '1'
	option nonegcache '0'
	option authoritative '1'
	option readethers '1'
	option leasefile '/tmp/dhcp.leases'
	option resolvfile '/tmp/resolv.conf.d/resolv.conf.auto'
	option nonwildcard '1'
	option localservice '1'
	option ednspacket_max '1232'
	list server '/use-application-dns.net/'
	option confdir '/tmp/dnsmasq.d'

config dhcp 'lan'
	option interface 'lan'
	option start '100'
	option limit '150'
	option leasetime '12h'
	option dhcpv4 'server'
	option dhcpv6 'server'
	option ra 'server'
	option ra_slaac '1'
	list ra_flags 'managed-config'
	list ra_flags 'other-config'

config dhcp 'wan'
	option interface 'wan'
	option ignore '1'

config odhcpd 'odhcpd'
[... 398 lines cut ...]

config rule
	option proto 'tcp'
	option dest_port '22'
	option name 'Block SSH from lan2'
	option family 'ipv4'
	option src 'lan2'
	option dest 'wan'
	option target 'REJECT'

/etc/config/https-dns-proxy

config main 'config'
	option force_dns '1'
	list force_dns_port '53'
	list force_dns_port '853'
	option update_dnsmasq_config '0'

config https-dns-proxy
	option listen_addr '127.0.0.1'
	option listen_port '5054'
	option user 'nobody'
	option group 'nogroup'
	option bootstrap_dns '1.1.1.1,1.0.0.1,2606:4700:4700::1111,2606:4700:4700::1001'
	option resolver_url 'https://cloudflare-dns.com/dns-query'

tapper82, 2022-07-21:
Here is what I get in my logs.

Thu Jul 21 20:40:40 2022 daemon.info https-dns-proxy[22352]: [F]    92232.1658432440 (null):631144 79B1: CURLOPT_SSLVERSION error 4: Error
Thu Jul 21 20:40:46 2022 daemon.info https-dns-proxy[22441]: [F]    92232.1658432446 (null):637200 79B1: CURLOPT_SSLVERSION error 4: Error
Thu Jul 21 20:40:51 2022 daemon.info https-dns-proxy[22442]: [F]    92232.1658432451 (null):841949 79B1: CURLOPT_SSLVERSION error 4: Error
Thu Jul 21 20:40:56 2022 daemon.info https-dns-proxy[22443]: [F]    92232.1658432456 (null):941003 3E66: CURLOPT_SSLVERSION error 4: Error
Thu Jul 21 20:41:02 2022 daemon.info https-dns-proxy[22444]: [F]    92232.1658432462 (null):21621 3E66: CURLOPT_SSLVERSION error 4: Error
Thu Jul 21 20:41:07 2022 daemon.info https-dns-proxy[22449]: [F]    92232.1658432467 (null):70491 3E66: CURLOPT_SSLVERSION error 4: Error
Thu Jul 21 20:41:07 2022 daemon.info procd: Instance https-dns-proxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Thu Jul 21 20:41:54 2022 daemon.warn dnsmasq[1]: Maximum number of concurrent DNS queries reached (max: 150)
Thu Jul 21 20:41:57 2022 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Thu Jul 21 20:42:00 2022 user.notice https-dns-proxy: Stopping service ✓
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: started, version 2.86 cachesize 150
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: DNS service limited to local subnets
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP no-DHCPv6 no-Lua TFTP no-conntrack no-ipset no-auth no-cryptohash no-DNSSEC no-ID loop-detect inotify dumpfile
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.1.100 -- 192.168.1.249, lease time 12h
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for test
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for local
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for adz.co.zw
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for trust.zone
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using 70553 more local addresses
Thu Jul 21 20:42:00 2022 daemon.err uhttpd[721]: sh: curl: not found
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using nameserver 194.168.4.100#53
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using nameserver 194.168.8.100#53
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for test
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for local
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for adz.co.zw
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using only locally-known addresses for trust.zone
Thu Jul 21 20:42:00 2022 daemon.info dnsmasq[1]: using 70553 more local addresses
Thu Jul 21 20:42:01 2022 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Thu Jul 21 20:42:01 2022 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Jul 21 20:42:01 2022 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 3 addresses
Thu Jul 21 20:42:01 2022 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses

tapper82, 2022-07-21:
@stangri

stangri, 2022-07-21:
Check if you have curl/libcurl installed.

tapper82, 2022-07-21:
List of installed packages.

adblock - 4.1.3-9
ath10k-board-qca9984 - 20220708-1
ath10k-firmware-qca9984-ct - 2020-11-08-1
attr - 2.5.1-1
avahi-dbus-daemon - 0.8-7
base-files - 1491-r20151-a0b7fef0ff
block-mount - 2022-07-17-ebf7e902-1
blockd - 2022-07-17-ebf7e902-1
busybox - 1.35.0-3
ca-bundle - 20211016-1
cgi-io - 2021-09-08-98cef9dd-20
coreutils - 9.1-1
coreutils-sort - 9.1-1
curl - 7.83.1-2
dbus - 1.13.18-12
dnsmasq - 2.86-14
dropbear - 2022.82-2
e2fsprogs - 1.46.5-1
firewall4 - 2022-06-17-15c38317-1
fstools - 2022-07-17-ebf7e902-1
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2022-07-17-46a33b8b-1
hostapd-common - 2022-06-02-4383528e-11
https-dns-proxy - 2021-11-22-3
irqbalance - 1.9.0-6
iw - 5.19-1
iwinfo - 2022-04-26-dc6847eb-1
jansson5 - 2.14-2
jshn - 2022-05-15-d2223ef9-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.15.53-1-b74d1c5fc794ff68fd474ac2015d8349
kmod-ata-ahci - 5.15.53-1
kmod-ata-ahci-platform - 5.15.53-1
kmod-ata-core - 5.15.53-1
kmod-ath - 5.15.53+5.15.33-1-1
kmod-ath10k-ct - 5.15.53+2022-05-13-f808496f-1
kmod-cfg80211 - 5.15.53+5.15.33-1-1
kmod-crypto-acompress - 5.15.53-1
kmod-crypto-aead - 5.15.53-1
kmod-crypto-ccm - 5.15.53-1
kmod-crypto-cmac - 5.15.53-1
kmod-crypto-crc32c - 5.15.53-1
kmod-crypto-ctr - 5.15.53-1
kmod-crypto-gcm - 5.15.53-1
kmod-crypto-gf128 - 5.15.53-1
kmod-crypto-ghash - 5.15.53-1
kmod-crypto-hash - 5.15.53-1
kmod-crypto-hmac - 5.15.53-1
kmod-crypto-manager - 5.15.53-1
kmod-crypto-null - 5.15.53-1
kmod-crypto-rng - 5.15.53-1
kmod-crypto-seqiv - 5.15.53-1
kmod-crypto-sha256 - 5.15.53-1
kmod-fs-autofs4 - 5.15.53-1
kmod-fs-exfat - 5.15.53-1
kmod-gpio-button-hotplug - 5.15.53-3
kmod-hwmon-core - 5.15.53-1
kmod-leds-gpio - 5.15.53-1
kmod-lib-crc-ccitt - 5.15.53-1
kmod-lib-crc32c - 5.15.53-1
kmod-lib-lzo - 5.15.53-1
kmod-mac80211 - 5.15.53+5.15.33-1-1
kmod-nf-conntrack - 5.15.53-1
kmod-nf-conntrack6 - 5.15.53-1
kmod-nf-flow - 5.15.53-1
kmod-nf-log - 5.15.53-1
kmod-nf-log6 - 5.15.53-1
kmod-nf-nat - 5.15.53-1
[... 121 lines cut ...]
samba4-libs - 4.14.12-5
samba4-server - 4.14.12-5
swconfig - 12
tcpdump - 4.99.1-1
terminfo - 6.3-1
ubi-utils - 2.1.4-1
uboot-envtools - 2022.01-38
ubox - 2022-07-17-46a33b8b-1
ubus - 2022-06-01-2bebf93c-1
ubusd - 2022-06-01-2bebf93c-1
uci - 2021-10-22-f84f49f0-6
uclient-fetch - 2021-05-14-6a6011df-1
ucode - 2022-06-01-b211ca0e-1
ucode-mod-fs - 2022-06-01-b211ca0e-1
ucode-mod-ubus - 2022-06-01-b211ca0e-1
ucode-mod-uci - 2022-06-01-b211ca0e-1
uhttpd - 2022-06-01-d59d732a-1
uhttpd-mod-ubus - 2022-06-01-d59d732a-1
urandom-seed - 3
urngd - 2020-01-21-c7f7b6b6-1
usign - 2020-05-23-f1f65026-1
wireless-regdb - 2022.06.06-1
wpad-basic-wolfssl - 2022-06-02-4383528e-11
wsdd2 - 2021-10-22-9831daf2-3
zlib - 1.2.12-2

tapper82, 2022-07-21:
@stangri I am sorry to have mest you about, but it started working for me. I don't know what I did or how I fixt it. Thanks for looking.

tapper82, 2022-07-21:
If it messes upagain I will reopen this with as mutch info as I can get for you.

closed 2022-07-21 by tapper82