openwrt/packages #19005: curl 7.83.1 can not work with https

openips, 2022-07-21:
i build x64 with last master tree ,and curl 7.83.1 with commit b0f213a726a019fa864420fc9d50523ab8ec8901
then i find  curl 7.83.1 cannot work with https
log below 
```
curl 7.83.1 (x86_64-openwrt-linux-gnu) libcurl/7.83.1
Release-Date: 2022-05-11
Protocols: file ftp http mqtt
Features: alt-svc IPv6 Largefile
root@OpenBSD:~# curl https://123.down/123
curl: (1) Protocol "https" not supported or disabled in libcurl
```
check and fix  thks

openips, 2022-07-21:
@stangri

Livy-Inverse, 2022-07-22:
I am using snapshot r20146-bc43ad88ed (2022-07-19). This bug is present on my Newifi D2 (mipsel_24kc), and make my `git clone` command over https not work.

But on my MobiPromo CM520-79F (arm_cortex-a7_neon-vfpv4) using the same snapshot version, everything works well.

ksverdlov, 2022-07-22:
I have the same issue with xiaomi ax6s (mediatek/mt7622) and snapshot "openwrt-37e86780cc34-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-sysupgrade".

With OpenWrt 22.03.0-rc5 r20151-a0b7fef0ff I can not install curl:

```
root@OpenWrt:~# opkg install curl
Installing curl (7.83.1-2) to root...
Downloading https://downloads.openwrt.org/releases/22.03.0-rc5/packages/aarch64_cortex-a53/packages/curl_7.83.1-2_aarch64_cortex-a53.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libcurl4
 * pkg_hash_fetch_best_installation_candidate: Packages for libcurl4 found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for curl:
 * 	libwolfssl5.4.0.ee39414e
 * opkg_install_cmd: Cannot install package curl.
```

openips, 2022-07-23:
use openssl lib can fix this issues.

closed 2022-07-23 by openips

paulhybryant, 2022-10-08:
can you elaborate a bit more on how to fix it (what you mean by "use openssl lib")?

awsong, 2022-10-09:
My OpenWRT version is 21.02, just upgraded to curl/libcurl 7.83.1 and have the exact same problem. The previous curl version 7.82.0 supports https just fine.

awsong, 2022-10-09:
check #19547 and #19548 .

I just downgraded according to #19547 for a temp solution.