openwrt/packages #19009: python-dns: ERROR: Could not find a version that satisfies the requirement setuptools_scm[toml]>=3.4.3

trippleflux, 2022-07-22:
Maintainer: @BKPepe 
Environment: OpenWRT x86_64 snapshot

Description:
Getting the following when compiling snapshot : 
```
make[3]: Entering directory '/home/username/works/openwrt/feeds/packages/lang/python/python-dns'
rm -f /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/.built
touch /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/.built_check
install -d -m0755 /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/ipkg-install//usr/lib/python3.10/site-packages
cd "/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/" && CC="x86_64-openwrt-linux-gnu-gcc" CCSHARED="x86_64-openwrt-linux-gnu-gcc -DPIC -fpic" CXX="x86_64-openwrt-linux-gnu-g++" LD="x86_64-openwrt-linux-gnu-gcc" LDSHARED="x86_64-openwrt-linux-gnu-gcc -shared" CFLAGS="-mtune=skylake -march=skylake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=skylake -march=skylake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0=dnspython-2.2.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/usr/include -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/include -I/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/include/python3.10" LDFLAGS="-L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.1.0_glibc/lib -znow -zrelro -lpython3.10" _PYTHON_HOST_PLATFORM="linux-x86_64" __PYVENV_LAUNCHER__="/usr/bin/python3.10" PYTHONPATH="/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/python3.10:/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom//usr/lib/python3.10/site-packages:/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/ipkg-install//usr/lib/python3.10/site-packages" PYTHONDONTWRITEBYTECODE=1 PYTHONOPTIMIZE="" _python_sysroot="/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom" _python_prefix="/usr" _python_exec_prefix="/usr"     /home/username/works/openwrt/staging_dir/hostpkg/bin/python3.10  setup.py   install --prefix="/usr" --root="/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/ipkg-install" --single-version-externally-managed
WARNING: The wheel package is not available.
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
ERROR: Could not find a version that satisfies the requirement setuptools_scm[toml]>=3.4.3 (from versions: none)
ERROR: No matching distribution found for setuptools_scm[toml]>=3.4.3
Traceback (most recent call last):
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/installer.py", line 75, in fetch_build_egg
    subprocess.check_call(cmd)
  File "/home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/username/works/openwrt/staging_dir/hostpkg/bin/python3.10', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/home/username/works/openwrt/tmp/tmpo9s2o634', '--quiet', 'setuptools_scm[toml]>=3.4.3']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/setup.py", line 36, in <module>
    setup(**setup_kwargs)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/__init__.py", line 152, in setup
    _install_setup_requires(attrs)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/__init__.py", line 147, in _install_setup_requires
    dist.fetch_build_eggs(dist.setup_requires)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/dist.py", line 806, in fetch_build_eggs
    resolved_dists = pkg_resources.working_set.resolve(
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/pkg_resources/__init__.py", line 766, in resolve
    dist = best[req.key] = env.best_match(
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/pkg_resources/__init__.py", line 1051, in best_match
    return self.obtain(req, installer)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/pkg_resources/__init__.py", line 1063, in obtain
    return installer(requirement)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/dist.py", line 877, in fetch_build_egg
    return fetch_build_egg(self, req)
  File "/home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages/setuptools/installer.py", line 77, in fetch_build_egg
    raise DistutilsError(str(e)) from e
distutils.errors.DistutilsError: Command '['/home/username/works/openwrt/staging_dir/hostpkg/bin/python3.10', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/home/username/works/openwrt/tmp/tmpo9s2o634', '--quiet', 'setuptools_scm[toml]>=3.4.3']' returned non-zero exit status 1.
make[3]: *** [Makefile:44: /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/pypi/dnspython-2.2.0/.built] Error 1
make[3]: Leaving directory '/home/username/works/openwrt/feeds/packages/lang/python/python-dns'
time: package/feeds/packages/python-dns/compile#1.61#0.10#9.20
    ERROR: package/feeds/packages/python-dns failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/python-dns/compile] Error 1
make[2]: Leaving directory '/home/username/works/openwrt'
make[1]: *** [package/Makefile:110: /home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:230: world] Error 2
```

In config python3-ssl is already selected : 
```
CONFIG_PACKAGE_python3-openssl=y
```

trippleflux, 2022-07-22:
Tried to upgrade pip but failed also : 
```
root@username:/home/username/works/openwrt/staging_dir/hostpkg/bin# ./python3 -m pip install --upgrade pip
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Requirement already satisfied: pip in /home/username/works/openwrt/staging_dir/hostpkg/lib/python3.10/site-packages (22.0.4)
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pip/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pip/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pip/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pip/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/pip/
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
WARNING: You are using pip version 22.0.4; however, version 22.1.2 is available.
You should consider upgrading via the '/home/username/works/openwrt/staging_dir/hostpkg/bin/python3 -m pip install --upgrade pip' command.
```

BKPepe, 2022-07-22:
The error, which you reported to me:

> ERROR: Could not find a version that satisfies the requirement setuptools_scm[toml]>=3.4.3

is caused by this:
```
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/setuptools-scm/
ERROR: Could not find a version that satisfies the requirement setuptools_scm[toml]>=3.4.3 (from versions: none)
```

This was introduced by updating LibreSSL - https://github.com/openwrt/openwrt/pull/10305  in the main repository and there is a pending pull request, which fixes it for Python https://github.com/openwrt/packages/pull/18981 and that's working in my case. 

I am sorry, but with these hiccups I am not able to do anything as this issue is elsewhere as I said, because of that, I am closing this bug as Invalid.