openwrt/packages #19010: watchcat:  modus “ping reboot” is not working correctly

cmonty14, 2022-07-22:
Maintainer: @\nickberry17
Environment: mips, RB760iGS (hEX S), OpenWrt 22.03)

Description:
I have configured [Watchcat](https://openwrt.org/docs/guide-user/advanced/watchcat?s%5B%5D=watchcat) for (single) modus "ping reboot":
```
config watchcat
	option mode 'ping_reboot'
	option period '5m'
	option pinghosts '1.1.1.1'
	option pingperiod '1m'
	option forcedelay '2m'
	option pingsize 'standard'
	option interface 'wan'
```
For unknown reason Watchcat cannot ping 1.1.1.1 and consequently reboots the router.
However 1.1.1.1 is pingable.
[image]

Can you please advise how to fix this issue?

THX

kofec, 2022-07-22:
I wrote here: https://github.com/openwrt/packages/pull/18593#issuecomment-1192325025
I change line 42 from:
```
        case "$family" in
        any)
                family=""
```
to:
```
        case "$family" in
        any)
                family="-4"
```

kofec, 2022-07-22:
If you have other issue and you want to debug:
`ps` - to check what  options you have for watchcat
`/etc/init.d/watchcat stop` stop all watchcats
change first line from:
`#!/bin/sh`
to
`#!/bin/sh -x`
and run your script from command line

kofec, 2022-07-22:
@jtkohl if you plan to correct above issue I also suggest some other changes (from telekom systems :-) ) like:
```
 watchcat_run_script() {
@@ -108,6 +111,9 @@ watchcat_monitor_network() {
        address_family="$8"
        script="$9"

+        failure_period_default="$failure_period"
+        ping_frequency_interval_default="$ping_frequency_interval"
+
        time_now="$(cat /proc/uptime)"
        time_now="${time_now%%.*}"

@@ -149,6 +155,12 @@ watchcat_monitor_network() {

                        if [ "$ping_result" -eq 0 ]; then
                                time_lastcheck_withinternet="$time_now"
+                                if [ "$failure_period" -gt "$failure_period_default" ]; then
+                                        failure_period="$failure_period_default"
+                                fi
+                                if [ "$ping_frequency_interval" -gt "$ping_frequency_interval_default" ]; then
+                                        ping_frequency_interval="$ping_frequency_interval_default"
+                                fi
                        else
                                if [ "$script" != "" ]; then
                                        logger -p daemon.info -t "watchcat[$$]" "Could not reach $host via \"$iface\" for \"$((time_now - time_lastcheck_withinternet))\" seconds. Running script after reaching \"$failure_period\" seconds"
@@ -176,6 +188,14 @@ watchcat_monitor_network() {
                        /etc/init.d/watchcat start
                        # Restart timer cycle.
                        time_lastcheck_withinternet="$time_now"
+                        failure_period="$((failure_period + failure_period))"
+                        if [ "$failure_period" -gt "86400" ]; then
+                                failure_period="86400"
+                        fi
+                        ping_frequency_interval="$((ping_frequency_interval + ping_frequency_interval))"
+                        if [ "$ping_frequency_interval" -gt "3600" ]; then
+                                ping_frequency_interval="3600"
+                        fi
                }

        done

```

jtkohl, 2022-07-22:
I am mostly offline for a week (summer vacation), feel free to fix this for me if you have time before August.  Sorry for the bug :_(

Another quick fix is on these lines in `watchcat.sh`, get rid of the quotes around `$ping_family`:
```
 					ping "$ping_family" -I "$iface" -s "$ping_size" -c 1 "$host" &> /dev/null
```
and
```
 					ping "$ping_family" -s "$ping_size" -c 1 "$host" &> /dev/null
```

without the quotes, the shell will collapse all the whitespace and not give ping a blank argument.

cmonty14, 2022-07-24:
I have applied all recommended corrections, and the issue is fixed (in my opinion).

Thanks for your support.

closed 2022-07-24 by cmonty14