openwrt/packages #19011: cryptsetup: Needs `PKG_FIXUP:=autoreconf` in `Makefile`, otherwise fails with `/usr/lib/libssh.so: file not recognized: file format not recognized` when cross compiling; in 22.03.

dreirund, 2022-07-22:
### Maintainer:

@dangowrt

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-20, ca. 08:15 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [&rarr;&nbsp;Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`.config`](https://github.com/openwrt/packages/files/9166301/config.cryptsetup.txt).
* `grep -i cryptsetup .config`:  
  ```
  CONFIG_PACKAGE_cryptsetup=m
  CONFIG_PACKAGE_cryptsetup-ssh=m
  ```

### Description:

`make -j6` fails in above configuration on `package/feeds/packages/cryptsetup`.  
An explicit run of  
`make -j1 V=sc package/feeds/packages/cryptsetup/compile`  
fails while relinking with  
`/usr/lib/libssh.so: file not recognized: file format not recognized`.

This can be mitigated by adding the following line to `package/feeds/packages/cryptsetup/Makefile`:  
`PKG_FIXUP:=autoreconf`  
([&rarr;&nbsp;Patch](https://github.com/openwrt/packages/files/9166305/cryptsetup.autoreconf-fixup.Makefile.patch.txt)).

Please consider adding this change.

<details><summary>Here more detailed 'make' output (*click* to open):</summary>

```
[...]
libtool: warning: relinking 'libcryptsetup-token-ssh.la'
libtool: install: (cd /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/cryptsetup-2.4.3; /bin/sh "/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/cryptsetup-2.4.3/libtool"  --silent --tag CC --mode=relink ccache_cc -Wall -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/cryptsetup-2.4.3=cryptsetup-2.4.3 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/include -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/include -no-undefined -avoid-version -Wl,--version-script=./tokens/libcryptsetup-token.sym -L/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/lib -L/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/lib -DPIC -fpic -specs=/[...]/openwrt-22.03/include/hardened-ld-pie.specs -znow -zrelro -L/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libiconv-stub/lib -L/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/lib -Wl,-rpath-link=/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib/libintl-stub/lib -Wl,--gc-sections -o libcryptsetup-token-ssh.la -rpath /usr/lib/cryptsetup tokens/ssh/libcryptsetup-token-ssh.lo tokens/ssh/ssh-utils.lo libcryptsetup.la -L/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib -lssh -L/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/lib -ljson-c -inst-prefix-dir /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/cryptsetup-2.4.3/ipkg-install)
/usr/lib/libssh.so: file not recognized: file format not recognized
collect2: error: ld returned 1 exit status
libtool:   error: error: relink 'libcryptsetup-token-ssh.la' with the above command before installing it
make[5]: *** [Makefile:1405: install-tokenLTLIBRARIES] Error 1
[...]
```

Full compile log: [&rarr;&nbsp;`cryptsetup.compile.log`](https://github.com/openwrt/packages/files/9166306/cryptsetup.compile.log).
</details>

dangowrt, 2022-07-22:
Can you open a pull request adding the line to the Makefile?

dreirund, 2022-07-22:
*@dangowrt wrote:*  
> *Can you open a pull request adding the line to the Makefile?*  

No, I don't know how to do that, and I don't want to learn more about git and github right now and manage different branches and forks, no capacities.

But people with write access to the repository should easily be able to incorporate the changes. I have provided a [patch file](https://github.com/openwrt/packages/files/9166305/cryptsetup.autoreconf-fixup.Makefile.patch.txt), feel free to use it in a way that suits your workflow and toolings.

Regards!

closed 2022-07-23 by dangowrt

dreirund, 2022-10-11:
***@***.*** wrote:*  
> *Can you open a pull request adding the line to the Makefile?*

No, I don't know how to do that, and I don't want to learn more about git right now, no capacities.

But people with write access to the repository should easily be able to incorporate the changes.

Regards!

commit ad0ac519 (2022-07-23): cryptsetup: fix library paths by calling autoreconf

Use PKG_FIXUP:=autoreconf when building cryptsetup to prevent
accidental linkage against host libraries.

Fixes: #19011
Reported-by: @dreirund
commit 5c987e57 (2022-08-04): cryptsetup: fix library paths by calling autoreconf

Use PKG_FIXUP:=autoreconf when building cryptsetup to prevent
accidental linkage against host libraries.

Fixes: #19011
Reported-by: @dreirund

(cherry picked from commit ad0ac5198decbc15c57801509a3005c1608ecbce)
commit e7bd06cc (2022-10-13): cryptsetup: fix library paths by calling autoreconf

Use PKG_FIXUP:=autoreconf when building cryptsetup to prevent
accidental linkage against host libraries.

Fixes: #19011
Reported-by: @dreirund

(cherry picked from commit ad0ac5198decbc15c57801509a3005c1608ecbce)