openwrt/packages #19012: transmission-daemon: Couldn't save temporary file ".../transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)

siwind, 2022-07-22:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: arm_cortex-a7_neon-vfpv4,  Openwrt master

Description:

After update to latest master version,(kernel version: 5.10.131) 
my transmission failed to add torrent file:

```
daemon.err transmission-daemon[7929]: Couldn't save temporary file "/mnt/udisk/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)
daemon.err transmission-daemon[7929]: Unable to create session lock file (2): No such file or directory (session-id.c:96)
```

I use another user/group to run transmission:
and directory permission is right:

```
# ps | grep tran
 8164 root      2408 SN   {transmission} /sbin/ujail -t 5 -n transmission -S /etc/seccomp/transmission-daemon.json -U wangwang -G wangwang -l -e CURL_CA_BUNDLE -r /etc
 8166 wangwang  6156 SN   /usr/bin/transmission-daemon -f -g /mnt/udisk/transmission
 8214 root      1144 S    grep tran

#
root@OpenWrt:/mnt/udisk/transmission# ll
drw-rw-rw-    5 wangwang wangwang      4096 Feb 21 19:43 ./
drwxr-xr-x   17 wangwang wangwang      4096 Mar 30  2021 ../
drwxr-xr-x    2 wangwang wangwang      4096 Dec 26  2020 blocklists/
-rw-------    1 wangwang wangwang      6828 Feb 21 19:43 dht.dat
drwxr-xr-x    2 wangwang wangwang      4096 Jul 22 18:47 resume/
-rw-rw-rw-    1 wangwang wangwang      1506 Jul 22 18:46 settings.json
-rw-------    1 wangwang wangwang       163 Feb 21 19:43 stats.json
drwxr-xr-x    2 wangwang wangwang      4096 Jul 22 18:47 torrents/

```

siwind, 2022-07-22:
Transmission version:

```
root@OpenWrt:/mnt/udisk/transmission# opkg list-installed | grep transmission
luci-app-transmission - git-22.029.54222-e3fbfe9
luci-i18n-transmission-zh-cn - git-22.181.29827-f2e6f28
transmission-daemon - 3.00-19
transmission-web - 3.00-19
```

trippleflux, 2022-07-22:
I am also experiencing the same issue but on x86_64 OpenWRT snapshot with transmission from [github](https://github.com/openwrt/packages/issues/18982).

jow-, 2022-07-22:
@dangowrt - likely caused by ujail, config likely needs to move into its own subdir with rw mount.

trippleflux, 2022-07-22:
Hmm, this issue seems exist since [2021](https://github.com/openwrt/packages/issues/17674). The workaround in [here ](https://github.com/openwrt/packages/issues/17674#issuecomment-1030264509)is doesn't work anymore.

dangowrt, 2022-07-22:
The first error
> ```
> daemon.err transmission-daemon[7929]: Couldn't save temporary file "/mnt/udisk/transmission/settings.json.tmp.XXXXXX": Permission denied (variant.c:1235)
> ```
is harmless, it only prevents configuration changes from being stored by transmission -- which is anyway not relevant as in OpenWrt config is generated from UCI at every start and hence changes made through transmission-remote are not persistent.

The second error message is new to me:
> ```
> daemon.err transmission-daemon[7929]: Unable to create session lock file (2): No such file or directory (session-id.c:96)
> ```

Can you share your `/etc/config/transmission` file, so I can try to reproduce the issue (it does not occur in my setup/configuration, using neither using Linux 5.10 nor when using Linux 5.15)

trippleflux, 2022-07-22:
@dangowrt 
For my case my errors are :
```
Sat Jul 23 06:41:03 2022 user.notice transmission: Starting with 3906340000 virt mem
Sat Jul 23 06:41:04 2022 daemon.err transmission-daemon[4073]: [2022-07-22 23:41:04.543] ERR variant.cc:1136 Couldn't save '/data/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 06:41:04 2022 daemon.err transmission-daemon[4073]: [2022-07-22 23:41:04.543] ERR variant.cc:1136 Couldn't save '/data/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 06:41:04 2022 daemon.err transmission-daemon[4073]: [2022-07-22 23:41:04.543] ERR variant.cc:1136 Couldn't save '/data/dht.dat': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR file-posix.cc:271 Couldn't create '/data/resume': Permission denied (13) (transmission-3.00/libtransmission/file-posix.cc:271)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR file-posix.cc:271 Couldn't create '/data/torrents': Permission denied (13) (transmission-3.00/libtransmission/file-posix.cc:271)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR file-posix.cc:271 Couldn't create '/data/blocklists': Permission denied (13) (transmission-3.00/libtransmission/file-posix.cc:271)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR utils.cc:218 Couldn't read '/data/stats.json': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR utils.cc:218 Couldn't read '/data/stats.benc': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR utils.cc:218 Couldn't read '/data/dht.dat': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR variant.cc:1136 Couldn't save '/data/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 06:41:05 2022 daemon.err transmission-daemon[6365]: [2022-07-22 23:41:05.654] ERR variant.cc:1136 Couldn't save '/data/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
```

/etc/config/transmission : 
```
config transmission
	option config_overwrite '1'
	option user 'transmission'
	option group 'transmission'
	option mem_percentage '50'
	option nice '10'
	option alt_speed_enabled 'false'
	option alt_speed_time_enabled 'false'
	option bind_address_ipv6 '::'
	option blocklist_enabled 'false'
	option cache_size_mb '2'
	option dht_enabled 'true'
	option download_queue_enabled 'true'
	option download_queue_size '4'
	option encryption '1'
	option idle_seeding_limit_enabled 'false'
	option lazy_bitfield_enabled 'true'
	option peer_limit_global '240'
	option peer_limit_per_torrent '60'
	option peer_port_random_on_start 'false'
	option peer_socket_tos 'default'
	option pex_enabled 'true'
	option preallocation '1'
	option queue_stalled_enabled 'true'
	option queue_stalled_minutes '30'
	option ratio_limit_enabled 'false'
	option rename_partial_files 'true'
	option rpc_enabled 'true'
	option rpc_host_whitelist_enabled 'false'
	option rpc_port '9091'
	option rpc_url '/transmission/'
	option scrape_paused_torrents_enabled 'true'
	option script_torrent_done_enabled 'false'
	option seed_queue_enabled 'false'
	option speed_limit_down_enabled 'false'
	option speed_limit_up_enabled 'false'
	option start_added_torrents 'true'
	option trash_original_torrent_files 'false'
	option umask '18'
	option upload_slots_per_torrent '14'
	option utp_enabled 'true'
	option scrape_paused_torrents 'true'
	option watch_dir_enabled 'false'
	option enabled '1'
	option incomplete_dir_enabled 'true'
	option lpd_enabled 'true'
	option rpc_bind_address '10.10.8.1'
	option rpc_whitelist_enabled 'true'
	option rpc_whitelist '127.0.0.1,192.168.1.*,10.10.8.142'
	option rpc_authentication_required 'true'
	option rpc_username 'username'
	option rpc_password 'password'
	option message_level '3'
	option bind_address_ipv4 '10.10.8.1'
	option peer_port '51414'
	option port_forwarding_enabled 'false'
	option config_dir '/data'
	option download_dir '/data/done'
	option incomplete_dir '/data/incomplete'
```

But I am running transmission git, you can test OpenWRT package in here : 
[transmission.zip](https://github.com/openwrt/packages/files/9172713/transmission.zip)

siwind, 2022-07-23:
> Can you share your `/etc/config/transmission` file, so I can try to reproduce the issue (it does not occur in my setup/configuration, using neither using Linux 5.10 nor when using Linux 5.15)

My configuation  is here:
https://github.com/openwrt/packages/issues/19012#issuecomment-1192446094

dangowrt, 2022-07-23:
@trippleflux Please change
```
	option config_dir '/data'
```
into something which is not a parent folder of other folders user by transmission. That should fix it.

trippleflux, 2022-07-23:
@dangowrt 
Thanks, hmm but can you give me an example for your solution?, transmission user has $HOME set as /var/run/transmission, do you mean I need to use it inside /var/run/transmission ?

dangowrt, 2022-07-23:
@trippleflux Example based on your configuration, changed so that it would work:
```
	option config_dir '/data/config'
	option download_dir '/data/done'
	option incomplete_dir '/data/incomplete'
```
The problem is simply that setting `config_dir` to `/data` won't work if `download_dir`, `incomplete_dir` or any other directories used by transmissions are subdirectories of any of directory configured.

trippleflux, 2022-07-23:
Hmm still failed :-( : 
```
Sat Jul 23 16:02:48 2022 user.notice transmission: Starting with 3906340000 virt mem
Sat Jul 23 16:02:48 2022 daemon.err transmission-daemon[29039]: [2022-07-23 09:02:48.739] ERR variant.cc:1136 Couldn't save '/data/config/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 16:02:48 2022 daemon.err transmission-daemon[29039]: [2022-07-23 09:02:48.739] ERR variant.cc:1136 Couldn't save '/data/config/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 16:02:49 2022 daemon.err transmission-daemon[30894]: [2022-07-23 09:02:49.849] ERR utils.cc:218 Couldn't read '/data/config/stats.benc': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 16:02:49 2022 daemon.err transmission-daemon[30894]: [2022-07-23 09:02:49.849] ERR utils.cc:218 Couldn't read '/data/config/dht.dat': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 16:02:49 2022 daemon.err transmission-daemon[30894]: [2022-07-23 09:02:49.849] ERR variant.cc:1136 Couldn't save '/data/config/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 16:02:49 2022 daemon.err transmission-daemon[30894]: [2022-07-23 09:02:49.849] ERR variant.cc:1136 Couldn't save '/data/config/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
```

```
        option config_dir '/data/config'
        option download_dir '/data/done'
        option incomplete_dir '/data/incomplete'
```

Regarding ujail, does "CONFIG_CGROUPFS_MOUNT_KERNEL_CGROUPS" is required for ujail dependency?, because I wasn't enabled it, I have finished my OpenWRT built with that turned on but still waiting for night for flashing into the router as currently some peoples using the internet.

Just incase that config is related with log : 
```
Sat Jul 23 07:23:03 2022 user.err ujail: jail: failed to load dependencies
```

dangowrt, 2022-07-23:
@trippleflux Please create the directory `/data/config` (it can be empty) and make sure the UID transmission runs as can access it.

trippleflux, 2022-07-23:
transmission-daemon seems created on its own but cannot read/write into it somehow : 
```
root@router:/data# ls -al
total 20
drwxr-xr-x  5 transmission transmission 4096 Jul 23 16:02 .
drwxr-xr-x 22 root         root         4096 Jul 23 15:53 ..
drwxr-xr-x  5 transmission transmission 4096 Jul 23 16:01 config
drwxr-xr-x  2 transmission transmission 4096 Jul 23 06:01 done
drwxr-xr-x  2 transmission transmission 4096 Jul 23 06:01 incomplete
```
```
root@router:/data/config# ls -al
total 24
drwxr-xr-x 5 transmission transmission 4096 Jul 23 16:01 .
drwxr-xr-x 5 transmission transmission 4096 Jul 23 16:02 ..
drwxr-xr-x 2 transmission transmission 4096 Jul 23 16:01 blocklists
drwxr-xr-x 2 transmission transmission 4096 Jul 23 16:01 resume
-rwxr-xr-x 1 transmission transmission 1501 Jul 23 16:36 settings.json
-rwxr-xr-x 1 transmission transmission    0 Jul 23 16:01 stats.json
drwxr-xr-x 2 transmission transmission 4096 Jul 23 16:01 torrents
```

```
Sat Jul 23 16:39:00 2022 user.notice transmission: Starting with 3906340000 virt mem
Sat Jul 23 16:39:00 2022 daemon.err transmission-daemon[787]: [2022-07-23 09:39:00.995] ERR variant.cc:1136 Couldn't save '/data/config/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 16:39:00 2022 daemon.err transmission-daemon[787]: [2022-07-23 09:39:00.995] ERR variant.cc:1136 Couldn't save '/data/config/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 16:39:00 2022 daemon.err transmission-daemon[787]: [2022-07-23 09:39:00.995] ERR variant.cc:1136 Couldn't save '/data/config/dht.dat': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 16:39:02 2022 daemon.err transmission-daemon[2696]: [2022-07-23 09:39:02.108] ERR utils.cc:218 Couldn't read '/data/config/stats.benc': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 16:39:02 2022 daemon.err transmission-daemon[2696]: [2022-07-23 09:39:02.108] ERR utils.cc:218 Couldn't read '/data/config/dht.dat': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 16:39:02 2022 daemon.err transmission-daemon[2696]: [2022-07-23 09:39:02.108] ERR variant.cc:1136 Couldn't save '/data/config/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 16:39:02 2022 daemon.err transmission-daemon[2696]: [2022-07-23 09:39:02.108] ERR variant.cc:1136 Couldn't save '/data/config/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
```

Hmm really confuse, regarding procd jail does it has dependency with ujail?

siwind, 2022-07-23:
> @siwind Same for
> 
> ```
>         option watch_dir '/mnt/udisk/bt'
> ```
> 
> Please change it into something which is not a parent folder of other folders used by transmission.

I add these following lines to  init script,  then  got "config file existing error" message, transimssion  running in failed state:

add these lines to init script:
```
        procd_add_jail_mount_rw "/mnt/udisk/transmission"
        procd_add_jail_mount_rw "/mnt/udisk/bt"
        procd_add_jail_mount_rw "/mnt/udisk/bt/000-incomplete"
        procd_add_jail_mount_rw "/mnt/udisk/bt/down"

```

then got these logs:
```
user.notice transmission: Starting with 253926000 virt mem
daemon.err transmission-daemon[18563]: transmission-daemon Error loading config file -- exiting. (daemon.c:825)
daemon.err transmission-daemon[18590]: transmission-daemon Error loading config file -- exiting. (daemon.c:825)
daemon.err transmission-daemon[18593]: transmission-daemon Error loading config file -- exiting. (daemon.c:825)
daemon.err transmission-daemon[18595]: transmission-daemon Error loading config file -- exiting. (daemon.c:825)
daemon.err transmission-daemon[18597]: transmission-daemon Error loading config file -- exiting. (daemon.c:825)
daemon.err transmission-daemon[18599]: transmission-daemon Error loading config file -- exiting. (daemon.c:825)
daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
```

siwind, 2022-07-23:
After comment this line from init script,  got "jail: failed to load dependencies"  error.

```
	#procd_set_param command "$PROG"
	procd_set_param command -f -g "$config_dir"

```

log error :
```
user.notice transmission: Starting with 253926000 virt mem
user.err : jail: -f is not an absolute path
daemon.err -f[7786]: jail: -f is not an absolute path
user.err : jail: failed to load dependencies		
```

dangowrt, 2022-07-23:
> Hmm really confuse, regarding procd jail does it has dependency with ujail?

Yes, procd jails depend on `procd-ujail` package (which is installed by default). Removing `procd-ujail` package results in process being run without jail.

trippleflux, 2022-07-23:
@dangowrt 
Finished flashing new OpenWRT build,I am still getting errors even when "CONFIG_CGROUPFS_MOUNT_KERNEL_CGROUPS" is enabled.
```
Sat Jul 23 21:34:28 2022 user.notice transmission: Starting with 3906340000 virt mem
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR utils.cc:218 Couldn't read '/data/config/stats.benc': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR utils.cc:218 Couldn't read '/data/config/dht.dat': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR variant.cc:1136 Couldn't save '/data/config/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR variant.cc:1136 Couldn't save '/data/config/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR watchdir-inotify.cc:180 Couldn't watch '/data/watch': No such file or directory (2) (transmission-3.00/libtransmission/watchdir-inotify.cc:180)
```

I am also even enabled "CONFIG_PACKAGE_uxc=y" but still getting above.

dangowrt, 2022-07-23:
I assume that `/data` is a block volume mounted there, right? If so, are you using `blockd` and `/etc/config/fstab` (*not* `/etc/fstab`)?

`uxc` is a container runtime (ie. like `crun` or `runc`) based on `procd-ujail`. You don't need it for transmission and it's not used there.

In any case:
```
ls - l /data
```
to make sure the directories mentioned do exist and have correct permissions.

trippleflux, 2022-07-23:
Should I disable blockd?, I thought that's default in OpenWRT and not quite understand blockd.

ls -l /data : 
```
root@router:/# ls -l /data
total 12
drwxr-xr-x 5 transmission transmission 4096 Jul 23 21:29 config
drwxr-xr-x 2 transmission transmission 4096 Jul 23 21:29 done
drwxr-xr-x 2 transmission transmission 4096 Jul 23 21:29 incomplete
```

df -h : 
```
root@router:/# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G  942M  116G   1% /
tmpfs           3.8G  5.5M  3.8G   1% /tmp
/dev/nvme0n1p1 1022M  6.6M 1016M   1% /boot
tmpfs           512K     0  512K   0% /dev
cgroup          3.8G     0  3.8G   0% /sys/fs/cgroup
```

mount : 
```
root@router:/# mount
/dev/nvme0n1p2 on / type ext4 (rw,noatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/nvme0n1p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/nvme0n1p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
cgroup on /sys/fs/cgroup type tmpfs (rw,relatime,mode=755)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu type cgroup (rw,relatime,cpu)
cgroup on /sys/fs/cgroup/cpuacct type cgroup (rw,relatime,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,relatime,memory)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,relatime,net_cls)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,relatime,pids)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,relatime,rdma)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
mountd(pid5201) on /tmp/run/blockd type autofs (rw,relatime,fd=7,pgrp=5201,timeout=30,minproto=5,maxproto=5,indirect)
```

/etc/fstab is empty..

dangowrt, 2022-07-25:
Ah ok, blockd is not relevant then -- I thought `/data` would a an additional storage volume or disk mounted via block-mount or blockd. If `/data` is part of the rootfs all that doesn't matter.

As you are building from source, can you share your `.config` file so I can try to reproduce your build?
It'd also be interesting to know if the issue shows also when using official binaries and the ImageBuilder.

trippleflux, 2022-07-25:
Here we go : 
[tranmission_25-07-2022.zip](https://github.com/openwrt/packages/files/9177083/tranmission_25-07-2022.zip)

I have also included my latest OpenWRT transmission git related packages mod, since in tranmission git the web things changed into /usr/share/transmission/public_html.

Thanks a lot for helping out.

siwind, 2022-07-25:
>         option watch_dir '/mnt/udisk/bt'
> ```
> 
> Please change it into something which is not a parent folder of other folders used by transmission.

In my openwrt router,  I found the problem is caused by this line of init script file:
```
[ "$watch_dir_enabled" = "1" ] && procd_add_jail_mount_rw "$watch_dir"

```

But the value of 'watch_dir_enabled'  in settings.json is  TRUE,  so i change  the line to following:
```
[ "$watch_dir_enabled" = "0" ]  ||  procd_add_jail_mount_rw "$watch_dir"
```

then transmission run ok, but another error log occur:
```
web will verify tracker certs using envvar CURL_CA_BUNDLE: /etc/ssl/certs/ca-certificates.crt (web.c:455)
daemon.err transmission-daemon[31781]: [2022-07-25 15:12:13.018] web NB: this only works if you built against libcurl with openssl or gnutls, NOT nss (web.c:457)
daemon.err transmission-daemon[31781]: [2022-07-25 15:12:13.019] web NB: invalid certs will show up as 'Could not connect to tracker' like many other errors (web.c:458)
daemon.err transmission-daemon[31781]: [2022-07-25 15:12:13.019] Unable to create session lock file (2): No such file or directory (session-id.c:96)
```

i have no idea now.

closed 2022-07-28 by siwind

AapoTahkola, 2022-08-17:
Removing procd-ujail package does not yield a perfectly working transmission on latest openwrt releases.