openwrt/packages #19013: bandwidthd: Fails in `./configure` with `error: Bandwidthd requires but cannot libpng` despite `PKG_FIXUP:=autoreconf` in 22.03.

dreirund, 2022-07-22:
### Maintainer:

@padre-lacroix

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-20, ca. 08:15 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [&rarr;&nbsp;Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`.config`](https://github.com/openwrt/packages/files/9167491/config.bandwidthd.txt),
* `grep -i bandwidthd .config`:  
  ```
  CONFIG_PACKAGE_bandwidthd=m
  # CONFIG_PACKAGE_bandwidthd-pgsql is not set
  CONFIG_PACKAGE_bandwidthd-php=m
  CONFIG_PACKAGE_bandwidthd-sqlite=m
  ```

### Description:

`make -j6` fails in above configuration on `package/feeds/packages/bandwidthd`.  
An explicit run of  
`make -j1 V=sc package/feeds/packages/bandwidthd/compile`  
fails in `./configure` with  
`configure: error: Bandwidthd requires but cannot libpng`:  
```
[...]
configure: WARNING: unrecognized options: --disable-nls
configure: loading site script /[...]/openwrt-22.03/include/site/arm
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-openwrt-linux-gnu
checking for bison... bison -y
checking for arm-openwrt-linux-gcc... ccache_cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether ccache_cc accepts -g... yes
checking for ccache_cc option to accept ISO C89... none needed
checking for a BSD-compatible install... /[...]/openwrt-22.03/staging_dir/host/bin/install -c
checking for flex... flex
checking lex output file root... lex.yy
checking lex library... none needed
checking whether yytext is a pointer... no
checking how to run the C preprocessor... ccache_cc -E
checking for X... libraries /usr/lib, headers /usr/include
checking for /sw/lib... (cached) no
checking for /sw/include... (cached) no
checking for /usr/pkg/lib... (cached) no
checking for /usr/pkg/include... (cached) no
checking for connect in -lsocket... no
checking for gethostbyname in -lnsl... no
checking for inet_aton in -lresolv... yes
checking for pow in -lm... yes
checking for libiconv_open in -liconv... no
checking for png_read_info in -lpng... no
configure: error: Bandwidthd requires but cannot libpng
make[2]: *** [Makefile:226: /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/bandwidthd-no-db/bandwidthd-2.0.1-35/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/utils/bandwidthd'
time: package/feeds/packages/bandwidthd/no-db/compile#8.59#1.28#17.35
    ERROR: package/feeds/packages/bandwidthd failed to build (build variant: no-db).
[...]
```  
*(Note by the way also the `WARNING: unrecognized options: --disable-nls`, which does not seem to be related here, but a hint to update the build script(s).)*

* Full `make` output: [&rarr;&nbsp;`bandwidthd.compile.log`](https://github.com/openwrt/packages/files/9167487/bandwidthd.compile.log),
* `config.log`: [&rarr;&nbsp;`build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/bandwidthd-no-db/bandwidthd-2.0.1-35/config.log`](https://github.com/openwrt/packages/files/9167475/bandwidthd.config.log).

In the `config.log`, I find  
`/usr/lib/libpng.so: file not recognized: file format not recognized`

But `package/feeds/packages/bandwidthd/Makefile` already contains `PKG_FIXUP:=autoreconf`.

padre-lacroix, 2022-07-23:
I will try to look into that over the weekend.
Do you know if that happens when compiling for other targets or just the arm_cortex?

neheb, 2022-08-04:
sounds like libpng is not built.

padre-lacroix, 2022-08-04:
Sorry for not coming back earlier on this.

I agree that it looks like libpng is not built.

After my original post 12 days ago, I did look at the nightly built for the 2.0.1-35 branch for the specific target and it had built properly between the time of the original post from @dreirund  and the time of my post 12 days ago and it was built properly.  Therefore, there was no issue there.

I had asked question to @dreirund  in my post 12 days ago, but now answer were received.

I think that this issue can be closed without further actions.

dreirund, 2022-08-04:
*@padre-lacroix wrote:*

> Do you know if that happens when compiling for other targets or just the arm_cortex?

I don't know since that is the only target I build for.

> After my original post 12 days ago, I did look at the nightly built for the 2.0.1-35 branch for the specific target and it had built properly between the time of the original post from @dreirund and the time of my post 12 days ago and it was built properly.

Did you try my complete `.config`? Maybe something interfering there.

> I had asked question to @dreirund in my post 12 days ago, but now answer were received.

Sorry, I answered timely via email reply, but somehow it did not make it here.

I can retry a full new build some time later, note that this takes a day to complete so I start on that endevour only after I finished some other stuff.

Regards!

dreirund, 2022-08-04:
*@padre-lacroix wrote:*  
> I did look at the nightly built for the 2.0.1-35 branch for the specific target and it had built properly between the time of the original post from @dreirund and the time of my post 12 days ago and it was built properly. Therefore, there was no issue there.
> 
> I think that this issue can be closed without further actions.

.. I still have this error, on a `openwrt-22.03` branch source tree with latest  
```
make clean
./scripts/feeds clean
git pull
./scripts/feeds update -a && 
  ./scripts/feeds install -a && 
  make oldconfig && 
  make menuconfig && 
  make
```  
on 2022-08-02, 12:23 GMT, with the following `bandwidthd` configs (after changing the config, I did run a  
`make -j1 V=sc package/feeds/packages/bandwidthd/{clean,compile}`  
each time):

* ```
  CONFIG_PACKAGE_bandwidthd=m
  # CONFIG_PACKAGE_bandwidthd-pgsql is not set
  # CONFIG_PACKAGE_bandwidthd-php is not set
  # CONFIG_PACKAGE_bandwidthd-sqlite is not set
  ```  
  (Full `.config` of that configuration: [&rarr;&hairsp;Here](http://ix.io/46Fq))

* ```
  # CONFIG_PACKAGE_bandwidthd is not set
  CONFIG_PACKAGE_bandwidthd-pgsql=m
  # CONFIG_PACKAGE_bandwidthd-php is not set
  # CONFIG_PACKAGE_bandwidthd-sqlite is not set
  ```

* ```
  # CONFIG_PACKAGE_bandwidthd is not set
  # CONFIG_PACKAGE_bandwidthd-pgsql is not set
  CONFIG_PACKAGE_bandwidthd-php=m
  # CONFIG_PACKAGE_bandwidthd-sqlite is not set
  ```

* ```
  # CONFIG_PACKAGE_bandwidthd is not set
  # CONFIG_PACKAGE_bandwidthd-pgsql is not set
  # CONFIG_PACKAGE_bandwidthd-php is not set
  CONFIG_PACKAGE_bandwidthd-sqlite=m
  ```

It is `CONFIG_PACKAGE_libpng=y` in all cases.

Note that the error message says "`configure: error: Bandwidthd requires but cannot libpng`", without the verb, i.e. the error message produces an incomplete sentence.

I have not tested the `master` branch, but since `openwrt-22.03` is going to become a release, problems in that branch I think should be fixed also in that branch, not only in `master`.

It looks to me that it needs further investigation. Maybe other configuration options I have chosen are triggering this, I have no idea, I don't know OpenWrt internals and I don't know how the linker works and I am not a programmer.

padre-lacroix, 2022-08-07:
OK,

I checked your config file and libpng is set to be compiled.  So bandwidth should be able to find it!

I am not sure what is hapening here.  But, it looks like that when the compiler is trying to find libpng, it cannot find it.  I know that this is the messages that is given.  This implies that libpng has not been compiled probably or that the source code is not there (I am not sure which one).

Can you check on the compilation log what happened when libpng was compiled, and check also that indeed it was tried to be compiled.

I have mentioned the following in your message a few days ago: _but since openwrt-22.03 is going to become a release, problems in that branch I think should be fixed also in that branch_.

I checked for the target in question (https://downloads.openwrt.org/releases/packages-22.03/arm_cortex-a7_neon-vfpv4/packages/) for the release 22.03 and the package is available and was last compiled this morning and is therefore available for download in installation in a router.

[image]

Therefore, the problem is probably more in your compiling environment, a setup that you have, or something else.  I would like tot ry your .config file, but to do so, I would need to build a complete building environment (install a virtual machine with Linux, install the build environment, etc.): I have not done anything in OpenWrt for a while and the nevironment that I have for this is obsolete and needs to be set up from scratch, which takes some time.

Sorry that I cannot at this moment help more.

dreirund, 2022-08-08:
> *Can you check on the compilation log what happened when libpng was
> compiled,*

Where do I find that?

Or do you mean the terminal output of  
`make -j1 V=sc package/feeds/packages/libpng/compile`?

The latter one, after a `make package/feeds/packages/libpng/clean`, you can find [&rarr;&thinsp;here](http://ix.io/470l).

-- 
Planung ist der Ersatz des Zufalls durch den Irrtum.

padre-lacroix, 2022-10-11:
It compiles fine with a fresh build environment for a Archer C7 v2 (target-mips_24kc-musl) with 22.03.

But bandwidthd currently has another problem with 22.03: the child process that draw the bandwidth data at 192.168.1.1/bandwidthd crashed with a SIGSEGV.  This is discussed in issue #19510 , and I am trying to fix it.  This is a very hard bug to trace!

But, bandwidthd-sqlite (and probably bandwidthd-pgsql) can collect data and can show the bandwidthd data on 192.168.1.1/phphtdocs, but suffer from the same problem to show the data at 192.168.1.1/bandwidthd.