openwrt/packages #19015: uacme: Fails to compile with `uacme-ualpn` enabled: `Package uacme-ualpn is missing dependencies for the following libraries: libev.so.4`; in 22.03.

dreirund, 2022-07-22:
### Maintainer:

@lucize

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-20, ca. 08:15 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [&rarr;&nbsp;Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`.config`](https://github.com/openwrt/packages/files/9167950/config.txt),
* `grep -i uacme .config`:  
  ```
  CONFIG_PACKAGE_uacme=m
  CONFIG_PACKAGE_uacme-ualpn=m
  ```

### Description:

`make -j6` fails in above configuration on `package/feeds/packages/uacme`.  
An explicit run of  
`make -j1 V=sc package/feeds/packages/uacme/compile`  
fails in `./configure` with  
```
Package uacme-ualpn is missing dependencies for the following libraries:
libev.so.4
```

More context:

```
[...]
find /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/uacme-upstream-1.7.1/ipkg-arm_cortex-a7_neon-vfpv4/uacme-ualpn -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package uacme-ualpn is missing dependencies for the following libraries:
libev.so.4
make[2]: *** [Makefile:145: /[...]/openwrt-22.03/bin/packages/arm_cortex-a7_neon-vfpv4/packages/uacme-ualpn_1.7.1-3_arm_cortex-a7_neon-vfpv4.ipk] Error 1
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/net/uacme'
time: package/feeds/packages/uacme/compile#0.69#0.20#0.84
    ERROR: package/feeds/packages/uacme failed to build.
[...]
```

Full `make` output: [&rarr;&nbsp;`uacme.compile.log`](https://github.com/openwrt/packages/files/9167951/compile.log).

Adding to the `package/feeds/packages/uacme/Makefile`'s `PKG_CONFIG_DEPENDS:=` variable the entry `CONFIG_PACKAGE_libev` does not solve the problem.

Disabling `uacme-ualpn` via `make menuconfig` makes `uacme` compile.

lucize, 2022-07-28:
https://github.com/openwrt/packages/commit/1642b68d4583c45e87b9628a38ae039e23617e0d maybe it needs backporting

dreirund, 2022-08-03:
> [1642b68](https://github.com/openwrt/packages/commit/1642b68d4583c45e87b9628a38ae039e23617e0d) maybe it needs backporting

I can confirm that this (manual backporting by editing the `Makefile`) fixes the issue for me.

closed 2022-08-03 by dreirund