openwrt/packages #19016: libsoup: Fails to build with fortify source due to `-Werror=format-nonliteral`; [PATCH] `PKG_FORTIFY_SOURCE:=0` makes it build; in 22.03.

dreirund, 2022-07-22:
### Maintainer:

@flyn-org

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-20, ca. 08:15 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [&rarr;&nbsp;Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`.config`](http://ix.io/4589),
* `grep -i soup .config`:  
  ```
  CONFIG_PACKAGE_libsoup=m
  CONFIG_PACKAGE_gst1-mod-soup=m
  ```
* `grep -i fortify config.txt`:  
  ```
  # CONFIG_PKG_FORTIFY_SOURCE_NONE is not set
  CONFIG_PKG_FORTIFY_SOURCE_1=y
  # CONFIG_PKG_FORTIFY_SOURCE_2 is not set
  ```

### Description:

`make -j6` fails in above configuration on `package/feeds/packages/libsoup`.  
An explicit run of  
`make -j1 V=sc package/feeds/packages/libsoup/compile`  
fails in `./configure` with  
`FAILED: libsoup/libsoup-2.4.so.1.11.2.p/soup-auth-ntlm.c.o`  
and  
`staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify/stdio.h:101:9: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]`  

This error could be mitigated by adding  
`PKG_FORTIFY_SOURCE:=0`  
to  
`package/feeds/packages/libsoup/Makefile`.  

[&rarr;&nbsp;Patch](http://ix.io/458b):  
```
--- Makefile.org	2022-07-22 15:51:16.858835397 +0200
+++ Makefile	2022-07-22 15:52:58.328832223 +0200
@@ -20,2 +20,3 @@
 
+PKG_FORTIFY_SOURCE:=0
 PKG_BUILD_DEPENDS:=glib2/host
```

Feel free to use that patch or your own way to incorporate that fix.  
Or, which I suspect would be the more clean way, not regard the warning as error? (`-Wno-error=format-nonliteral` somewhere?)

More context of the output of the above failing `make` run:

```
[...]
ninja: Entering directory `/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libsoup-2.74.2/openwrt-build'
[... 25 lines cut ...]
../libsoup/soup-auth-ntlm.c:770:17: warning: 'g_memdup' is deprecated: Use 'g_memdup2' instead [-Wdeprecated-declarations]
  770 |                 *target_info = g_memdup (chall + target.offset, target.length);
      |                 ^
In file included from /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/glib-2.0/glib.h:82,
                 from ../libsoup/soup-auth-ntlm.c:17:
/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/glib-2.0/glib/gstrfuncs.h:257:23: note: declared here
  257 | gpointer              g_memdup         (gconstpointer mem,
      |                       ^~~~~~~~
../libsoup/soup-auth-ntlm.c: At top level:
../libsoup/soup-auth-ntlm.c:1340:39: warning: argument 2 of type 'unsigned char[8]' with mismatched bound [-Warray-parameter=]
 1340 | des (guint32 ks[16][2], unsigned char block[8])
      |                         ~~~~~~~~~~~~~~^~~~~~~~
../libsoup/soup-auth-ntlm.c:604:44: note: previously declared as 'unsigned char *'
  604 | static void des                   (DES_KS, unsigned char *);
      |                                            ^~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
make[2]: *** [Makefile:79: /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/libsoup-2.74.2/.built] Error 1
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/libs/libsoup'
time: package/feeds/packages/libsoup/compile#2.13#0.30#5.09
    ERROR: package/feeds/packages/libsoup failed to build.
[...]
```

Full `make` output: [&rarr;&nbsp;`libsoup.compile.log`](http://ix.io/458a).

dreirund, 2022-08-23:
Should be fixed by #19245.

flyn-org, 2022-08-23:
Is it musl code (stdio.h) that is triggering these compiler errors? That sounds like a mismatch between GCC and musl, not a problem with libsoup. I would like to avoid disabling compiler protections.

dreirund, 2022-08-24:
*@flyn-org wrote:*  
> *Is it musl code (stdio.h) that is triggering these compiler errors? That sounds like a mismatch between GCC and musl, not a problem with libsoup.*

I have no idea, I don't know almost anything about linking and compilers and such stuff.

I have seen those errors also on a few other packages.