openwrt/packages #19019: Curl dependent on nonexistant libwolfssl5.4

jeisom, 2022-07-22:
Maintainer: stangri@github-user (find it by checking history of the package Makefile)
Environment: [22.03.0-rc5 mvebu cortexa9 linksys_wrt1900ac-v1

Description:
I installed OpenWrt 22.03.0-rc5 on wrt1900ac-v1 and I can't install curl as it depends on libwolfssl5.4 and libwolfssl5.3 is the newest available.

cmonty14, 2022-07-23:
Confirm issue with curl_7.83.1 on *mipsel_24kc*.

BKPepe, 2022-07-23:
Reference: https://github.com/openwrt/openwrt/issues/10302

Fixed by https://github.com/openwrt/packages/pull/19008

closed 2022-07-23 by BKPepe

cmonty14, 2022-07-24:
> Reference: [openwrt/openwrt#10302](https://github.com/openwrt/openwrt/issues/10302)
> 
> Fixed by #19008

You closed this issue, but it's not solved, yet.
I understand there's a pull request, but as long as this pull request is not completed, the issue is not resolved.

BKPepe, 2022-07-24:
The pull request was already merged when I wrote my comment and it was not reverted. You need to wait until buildbots will compile the packages for your device, and they will be available to install. It could take up to 24/48 hours.

cmonty14, 2022-07-24:
> The pull request was already merged when I wrote my comment and it was not reverted. You need to wait until buildbots will compile the packages for your device, and they will be available to install. It could take up to 24/48 hours.

Thanks for confirmation.

adamcstephens, 2022-07-26:
Will the #19008 change be applied to the 22.03 branch as well? rc5 is currently broken as is.

karafen, 2022-07-27:
> Will the #19008 change be applied to the 22.03 branch as well? rc5 is currently broken as is.
A similar question, every day I wait for an update, but so far it has not been fixed!
Corrected only in snapshot.

stangri, 2022-07-27:
> Will the #19008 change be applied to the 22.03 branch as well? rc5 is currently broken as is.

I cherry-picked @neheb's fix for 22.03 it should be available when buildbots get to it.