openwrt/packages #19020: shadow: Update to v4.11.1 to prevent CVE

trippleflux, 2022-07-22:
Environment: OpenWRT x86_64 snapshot

Description:

There is a CVE-2013-4235 going on version 4.11 and below, currently on OpenWRT package is version 4,8.1.
I have tried to change it my self but stuck at linking error

shadow Makefile : 
```
#
# Copyright (C) 2008-2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=shadow
PKG_VERSION:=4.11.1
PKG_RELEASE:=1

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=https://github.com/shadow-maint/shadow/releases/download/v$(PKG_VERSION)/
PKG_HASH:=41f093ce58b2ae5f389a1c5553e0c18bc73e6fe27f66273891991198a7707c95

PKG_MAINTAINER:=
PKG_LICENSE:=BSD-3-Clause
PKG_LICENSE_FILES:=COPYING
PKG_CPE_ID:=cpe:/a:debian:shadow

PKG_INSTALL:=1
PKG_BUILD_PARALLEL:=1
PKG_BUILD_DEPENDS:=libxcrypt

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/nls.mk

SHADOW_APPLETS := \
	chage chfn chgpasswd chpasswd chsh expiry faillog gpasswd \
	groupadd groupdel groupmems groupmod groups grpck grpconv grpunconv \
	lastlog login logoutd newgidmap newgrp newuidmap newusers nologin \
	passwd pwck pwconv pwunconv su \
	useradd userdel usermod vipw

CONFIGURE_ARGS += \
	--without-audit \
	--without-libpam \
	--without-selinux \
	--without-acl \
	--without-attr \
	--without-tcb \
	--without-nscd \
	--with-bcrypt

CONFIGURE_VARS += \
	ac_cv_func_ruserok=no

define Package/shadow/Default
  SECTION:=utils
  CATEGORY:=Utilities
  TITLE:=The PLD Linux shadow utilities
  URL:=https://github.com/shadow-maint/shadow
endef

define Package/shadow
  $(call Package/shadow/Default)
  TITLE+= (all)
  DEPENDS:=$(foreach u,$(SHADOW_APPLETS),+shadow-$(u))
  HIDDEN:=1
endef

define Package/shadow/description
  Full versions of standard shadow utilities. Normally, you would not
  use this package, since the functionality in BusyBox is more than
  sufficient and much smaller.
endef

define Package/shadow/install
	true
endef

define Package/shadow-utils
  $(call Package/shadow/Default)
endef

define Package/shadow-utils/config
    menu "Select shadow utilities"
      depends on PACKAGE_shadow-utils

    config shadow-all
      bool "Include all PLD shadow utilities"
      select PACKAGE_shadow
      default y

    comment "Utilities"

  $(foreach u,$(SHADOW_APPLETS), \
    config PACKAGE_shadow-$(u)
      prompt "Full PLD version of $(u) utility"

   )

  endmenu

endef

Package/shadow-utils/description = $(Package/shadow/description)

define Package/shadow-common
  $(call Package/shadow/Default)
  TITLE:=Shared definitions for the PLD Linux shadow utilities
  DEPENDS:=$(ICONV_DEPENDS) $(INTL_DEPENDS)
  HIDDEN:=1
endef

define GenPlugin
  define Package/shadow-$(1)
    $(call Package/shadow/Default)
    TITLE:=Utility $(1) from the PLD Linux shadow utilities
    DEPENDS:=+shadow-common
    HIDDEN:=1
  endef

  define Package/shadow-$(1)/description
    Full version of standard $(1) utility. Normally, you would not use this
    package, since the functionality in BusyBox is more than sufficient.
  endef
endef

$(foreach u,$(SHADOW_APPLETS),$(eval $(call GenPlugin,$(u))))

define Package/shadow-common/conffiles
/etc/login.defs
endef

#hack to get libxcrypt working
TARGET_LDFLAGS:=-L$(STAGING_DIR)/usr/lib/libxcrypt $(TARGET_LDFLAGS)

define Package/shadow-common/install
	$(INSTALL_DIR) $(1)/etc
	$(INSTALL_CONF) $(PKG_INSTALL_DIR)/etc/login.defs $(1)/etc/
	$(SED) 's,SU_NAME,#SU_NAME,g' $(1)/etc/login.defs
	$(SED) 's,#ENCRYPT_METHOD DES,ENCRYPT_METHOD BCRYPT,g' $(1)/etc/login.defs
endef

define Package/shadow-utils/install
	true
endef

define BuildPlugin
  define Package/shadow-$(1)/install
	[ -x "$(PKG_INSTALL_DIR)/usr/sbin/$(1)" ] && { \
		$(INSTALL_DIR) $$(1)/usr/sbin; \
		$(CP) $(PKG_INSTALL_DIR)/usr/sbin/$(1) $$(1)/usr/sbin/; \
	} || { \
		$(INSTALL_DIR) $$(1)/usr/bin; \
		$(CP) $(PKG_INSTALL_DIR)/usr/bin/$(1) $$(1)/usr/bin/; \
	}
  endef

  $$(eval $$(call BuildPackage,shadow-$(1)))
endef

$(foreach u,$(SHADOW_APPLETS),$(eval $(call BuildPlugin,$(u))))

$(eval $(call BuildPackage,shadow-utils))
$(eval $(call BuildPackage,shadow-common))
$(eval $(call BuildPackage,shadow))
```

004-fix-su-controoling-term.patch : 
```
diff -Naur a/src/su.c b/src/su.c
--- a/src/su.c	2022-07-22 23:11:33.224952300 +0700
+++ b/src/su.c	2022-07-23 00:51:50.922436700 +0700
@@ -1136,8 +1136,12 @@
 
 		if (fd >= 0) {
 			err = ioctl (fd, TIOCNOTTY, (char *) 0);
+			if (-1 == err && ENOTTY == errno) {
+				/* There are no controlling terminal already */
+				err = 0;
+			}
 			(void) close (fd);
-		} else if (ENXIO == errno) {
+		} else if (ENXIO == errno || EACCES == errno) {
 			/* There are no controlling terminal already */
 			err = 0;
 		}
```

BKPepe, 2022-07-23:
Check this out about the referenced CVE: https://github.com/shadow-maint/shadow/issues/317

trippleflux, 2022-07-23:
@BKPepe 
Thanks, closing this then.

closed 2022-07-23 by trippleflux