openwrt/packages #19034: samba4-server: Bus error on bcm2711 with 21.02.3

ffainelli, 2022-07-24:
smbd cannot execute at all and throws a bus error upon the first mmap() call:

`root@OpenWrt:/etc/config# strace smbd
execve("/usr/sbin/smbd", ["smbd"], 0x7fd4260e30 /* 13 vars */) = 0
set_tid_address(0x7fa811f0cc)           = 2785
openat(AT_FDCWD, "/usr/lib/libsamba-util.so.0", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=518320, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\330\316\0\0\0\0\0\0"..., 960) = 960
mmap(NULL, 528384, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fa800d000
mmap(0x7fa808b000, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x6e000) = 0x7fa808b000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libsmbd-base-samba4.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/lib/samba/libsmbd-base-samba4.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl(3, F_SETFD, FD_CLOEXEC)           = 0
fstat(3, {st_mode=S_IFREG|0755, st_size=2650112, ...}) = 0
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\220)\6\0\0\0\0\0"..., 960) = 960
mmap(NULL, 3256320, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0x7fa7cf2000
mmap(0x7fa7ffd000, 65536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2fb000) = 0x7fa7ffd000
--- SIGBUS {si_signo=SIGBUS, si_code=BUS_ADRERR, si_addr=0x7fa800ac28} ---
+++ killed by SIGBUS +++
Bus error
`

@neheb have you seen something like that before?

neheb, 2022-07-24:
nope. sounds like a format string error where a time_t is attempted to be printed with %d.

ffainelli, 2022-07-24:
Looks like a reinstall of samba4-libs fixed it, not sure what happened here, I did resize my root filesystem after installing OpenWrt on my Pi 4.

closed 2022-07-24 by ffainelli