openwrt/packages #19037: squeezelite: Fails to build with `ffmpeg.c:125:31: error: 'avcodec_decode_audio4' undeclared (first use in this function); did you mean 'avcodec_decode_subtitle2'?` in 22.03.

dreirund, 2022-07-24:
### Maintainer:

@thess

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-07-20, ca. 08:15 UTC  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [&rarr;&nbsp;Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`.config.compile-error`](http://ix.io/45ob),
* `grep -i squeezelite .config`:  
  ```
  CONFIG_PACKAGE_squeezelite-full=m
  CONFIG_SQUEEZELITE_WMA=y
  CONFIG_SQUEEZELITE_RESAMPLE=y
  CONFIG_SQUEEZELITE_DSD=y
  # CONFIG_PACKAGE_squeezelite-mini is not set
  ```

### Description:

`make -j6` fails in above configuration on `package/feeds/packages/squeezelite`.  
An explicit run of  
`make -j1 V=sc package/feeds/packages/squeezelite/compile`  
fails in `./configure` with  
```
ffmpeg.c: In function 'ff_decode':
[...]
ffmpeg.c:125:31: error: 'avcodec_decode_audio4' undeclared (first use in this function); did you mean 'avcodec_decode_subtitle2'?
```

<details><summary>More context (*click* to open):</summary>

```
ccache_cc -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squeezelite-full/squeezelite-1.9.6-1210=squeezelite-1.9.6-1210 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include  -fcommon  -DFFMPEG -DDSD -DRESAMPLE -DLINKALL ffmpeg.c -c -o ffmpeg.o
ffmpeg.c: In function 'ff_decode':
ffmpeg.c:350:23: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  350 |                 codec = AVCODEC(ff, find_decoder, ff->codecP->codec_id);
      |                       ^
ffmpeg.c:366:17: warning: 'av_init_packet' is deprecated [-Wdeprecated-declarations]
  366 |                 AV(ff, init_packet, ff->avpkt);
      |                 ^~
In file included from /[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libavformat/avformat.h:316,
                 from ffmpeg.c:26:
/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libavcodec/packet.h:506:6: note: declared here
  506 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
ffmpeg.c:125:31: error: 'avcodec_decode_audio4' undeclared (first use in this function); did you mean 'avcodec_decode_subtitle2'?
  125 | #define AVCODEC(h, fn, ...)  (avcodec_ ## fn)(__VA_ARGS__)
      |                               ^~~~~~~~
ffmpeg.c:406:23: note: in expansion of macro 'AVCODEC'
  406 |                 len = AVCODEC(ff, decode_audio4, ff->codecC, ff->frame, &got_frame, &pkt_c);
      |                       ^~~~~~~
ffmpeg.c:125:31: note: each undeclared identifier is reported only once for each function it appears in
  125 | #define AVCODEC(h, fn, ...)  (avcodec_ ## fn)(__VA_ARGS__)
      |                               ^~~~~~~~
ffmpeg.c:406:23: note: in expansion of macro 'AVCODEC'
  406 |                 len = AVCODEC(ff, decode_audio4, ff->codecC, ff->frame, &got_frame, &pkt_c);
      |                       ^~~~~~~
ffmpeg.c: In function 'ff_open_wma':
ffmpeg.c:594:26: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  594 |         ff->input_format = AV(ff, find_input_format, "asf");
      |                          ^
ffmpeg.c: In function 'ff_open_alac':
ffmpeg.c:610:26: warning: assignment discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
  610 |         ff->input_format = AV(ff, find_input_format, "mp4");
      |                          ^
ffmpeg.c: In function 'register_ff':
ffmpeg.c:123:31: error: 'av_register_all' undeclared (first use in this function)
  123 | #define AV(h, fn, ...)       (av_ ## fn)(__VA_ARGS__)
      |                               ^~~
ffmpeg.c:762:17: note: in expansion of macro 'AV'
  762 |                 AV(ff, register_all);
      |                 ^~
make[3]: *** [Makefile:149: ffmpeg.o] Error 1
make[3]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squeezelite-full/squeezelite-1.9.6-1210'
make[2]: *** [Makefile:145: /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squeezelite-full/squeezelite-1.9.6-1210/.built] Error 2
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/sound/squeezelite'
time: package/feeds/packages/squeezelite/full/compile#0.68#0.16#0.88
    ERROR: package/feeds/packages/squeezelite failed to build (build variant: full).
```

[&rarr;&nbsp;Full `make` output](http://ix.io/45oh)
</details>

closed 2022-08-04 by neheb

dreirund, 2022-08-16:
> Closed #19037 as completed via #19046.

Did not make it to the 22.03 branch yet. Since this seems to be a bug,
it should be backported.

-- 
I'll catch you if you fall.
                      - floor

neheb, 2022-08-16:
Done.