openwrt/packages #19043: banIP: set pipefail + "| grep -q" / "| grep --quiet"

realizelol, 2022-07-24:
Maintainer: @dibdot
Environment: amd64, generic, OpenWrt 21.02.3, r16554-1d4dea6d4f

Description:
/usr/bin/banip.sh has `set -o pipefail` declared which isn't wrong but stop execution after piping to `grep -q` / `grep --quiet` with error 141.
This is already a known bug: https://github.com/koalaman/shellcheck/issues/665 / https://github.com/koalaman/shellcheck/issues/1109
Also piping to `head` or `grep -m` or `grep --max-counts` shouldn't be working. (not tested)

<br>

I am actually doing this, which isn't the right way:
```
if [[ "$(opkg list-installed banip | awk '{print$3}')" == "0.7.10-3" ]]; then
  sed -i '/^set -o pipefail/d' /usr/bin/banip.sh
fi
```

<br>

<br>

best regards
realizelol

realizelol, 2022-07-24:
Oh btw I am using GNU grep:
```
opkg list-installed grep
grep - 3.6-1
```
```
grep -V
grep (GNU grep) 3.6
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Mike Haertel and others; see
<https://git.sv.gnu.org/cgit/grep.git/tree/AUTHORS>.
```

dibdot, 2022-08-11:
What happend if you just remove the pipefail command?

realizelol, 2022-08-15:
If I remove the pipefail the script will work finely.

Here's another sed after debug
```
/bin/sed -i -e 's!\(grep\) -q "\(.*\)" ||!\1 "\2" >/dev/null ||!g' -e 's!\(grep\) -q "\(.*\)";!\1 "\2" >/dev/null;!g' /usr/bin/banip.sh
```

closed 2023-05-16 by realizelol