openwrt/packages #19062: cloudflared: Package not available to install on OpenWRT 21.02.3

jjpinom, 2022-07-27:
Maintainer: @\1715173329
Environment:  OpenWrt 21.02.3

Description:

Cloudflared is not included in the package list to install

1715173329, 2022-07-28:
That's correct. Only available on 22.03 or snapshots.

jjpinom, 2022-07-28:
thanks!.

is there any way to install it on 21.02.3?.

I wan to migrate to 22.03, but I depend heavily on iptables

jow-, 2022-07-28:
Given that it is a Go executable you can probably just copy it onto your system or install the package manually from the 22.03 repos.

jjpinom, 2022-07-28:
Thanks!.

I'm learning OpenWRT and Linux. Could you please guide how to do that??

jjpinom, 2022-07-28:
Hello!.

I was able to installa cloudlared on 22.03 and configure a new tunnel.

The problem now is that cloudflared daemon does not start in boot. 

When I tried to enable it using I got this error:

Failed to execute "/etc/init.d/cloudflared enable" action: Command failed

jjpinom, 2022-07-28:
I put a command on the startup script:

/etc/init.d/cloudflared start

It worked, but one should be able to activate init script in the startup/initscript section

closed 2022-07-28 by jjpinom