openwrt/packages #19064: vpn-policy-routing: VPN Policies create iptables on OpenWRT 22.03

jjpinom, 2022-07-27:
Maintainer: @\stangri
Environment: 22.03

Description:

vpn-policy-routing is creating iptables and not nftables it seems to be working, but the firewall status section alerts about legacy iptables entries

jjpinom, 2022-07-28:
Thanks for taking this. 

Just a comment. I use also dnsmasq ipset functionality. I don't know if ipset and nftables work together.

Best regards.

stangri, 2022-09-20:
I've updated the [vpn-policy-routing README](https://docs.openwrt.melmac.net/vpn-policy-routing/) and also the README for the new package -- [pbr](https://docs.openwrt.melmac.net/pbr/).
I'm closing this, as this is "wont't fix" for vpn-policy-routing since the switch to nft is going to be implemented in a different package.

closed 2022-09-20 by stangri