openwrt/packages #19065: watchcat: is does not work any more -- ping is not used properly.

mistepien, 2022-07-27:
My watchcat does not work any more. My watchcat version is 1-16.

>    config watchcat
>   option period '1h'
>    option pingsize 'standard'
>    option mode 'restart_iface'
>    option pinghosts '1.1.1.1'
>    option interface 'eth0'
>    option pingperiod '3m'

and in logs I have

> Wed Jul 27 20:51:06 2022 daemon.info watchcat[14595]: Could not reach 1.1.1.1 via "eth0" for "3420" seconds. Restarting "eth0" after reaching "3600" seconds"

"ping -I eth0 1.1.1.1" works. Before that I tried 8.8.8.8 and the problem was the same. I have internet but rebooting router/interface is pointless if connection is fine.

kofec, 2022-07-27:
https://github.com/openwrt/packages/pull/19059

jtkohl, 2022-08-02:
#19084

jtkohl, 2022-08-02:
> My watchcat does not work any more. My watchcat version is 1-16.
> 
> > config watchcat
> > option period '1h'
> > option pingsize 'standard'
> > option mode 'restart_iface'
> > option pinghosts '1.1.1.1'
> > option interface 'eth0'
> > option pingperiod '3m'
> 
> and in logs I have
> 
> > Wed Jul 27 20:51:06 2022 daemon.info watchcat[14595]: Could not reach 1.1.1.1 via "eth0" for "3420" seconds. Restarting "eth0" after reaching "3600" seconds"
> 
> "ping -I eth0 1.1.1.1" works. Before that I tried 8.8.8.8 and the problem was the same. I have internet but rebooting router/interface is pointless if connection is fine.
> 
> That is not only my problem: https://forum.openwrt.org/t/watchcat-modus-ping-reboot-is-not-working-correctly/132713

workaround until this is fixed:  specify the ping address family in the watchcat configuration

jtkohl, 2022-08-12:
> My watchcat does not work any more. My watchcat version is 1-16.

@mistepien Please try now with version 1-17

closed 2025-11-15 by GeorgeSapkin