openwrt/packages #19067: transmission-daemon: permission denied problems

trippleflux, 2022-07-28:
Maintainer: @dangowrt 
Environment: OpenWRT x86_64 snapshot

Description:

transmission-daemon failed to read/write directories that are configured in the config file : 
```
Sat Jul 23 21:34:28 2022 user.notice transmission: Starting with 3906340000 virt mem
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR utils.cc:218 Couldn't read '/data/config/stats.benc': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR utils.cc:218 Couldn't read '/data/config/dht.dat': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR variant.cc:1136 Couldn't save '/data/config/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR variant.cc:1136 Couldn't save '/data/config/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)
Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR watchdir-inotify.cc:180 Couldn't watch '/data/watch': No such file or directory (2) (transmission-3.00/libtransmission/watchdir-inotify.cc:180)
```

ls -l /data : 
```
root@router:/# ls -l /data
total 12
drwxr-xr-x 5 transmission transmission 4096 Jul 23 21:29 config
drwxr-xr-x 2 transmission transmission 4096 Jul 23 21:29 done
drwxr-xr-x 2 transmission transmission 4096 Jul 23 21:29 incomplete
```

df -h : 
```
root@router:/# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       117G  942M  116G   1% /
tmpfs           3.8G  5.5M  3.8G   1% /tmp
/dev/nvme0n1p1 1022M  6.6M 1016M   1% /boot
tmpfs           512K     0  512K   0% /dev
cgroup          3.8G     0  3.8G   0% /sys/fs/cgroup
```

mount : 
```
root@router:/# mount
/dev/nvme0n1p2 on / type ext4 (rw,noatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/nvme0n1p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/nvme0n1p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
cgroup on /sys/fs/cgroup type tmpfs (rw,relatime,mode=755)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,relatime,cpuset)
cgroup on /sys/fs/cgroup/cpu type cgroup (rw,relatime,cpu)
cgroup on /sys/fs/cgroup/cpuacct type cgroup (rw,relatime,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,relatime,blkio)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,relatime,memory)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,relatime,net_cls)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,relatime,pids)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,relatime,rdma)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
mountd(pid5201) on /tmp/run/blockd type autofs (rw,relatime,fd=7,pgrp=5201,timeout=30,minproto=5,maxproto=5,indirect)
```

/etc/fstab is empty..

OpenWRT package mod used : 
[transmission-git](https://github.com/openwrt/packages/files/9177083/tranmission_25-07-2022.zip)

In continuation of [this issue](https://github.com/openwrt/packages/issues/19012) regarding my specific case above.

dangowrt, 2022-07-28:
Apart from the errors showing in the log, does transmission actually work?
Because line by line:
> `Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR utils.cc:218 Couldn't read '/data/config/stats.benc': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)`

Stats have not been created (and won't be), so not surprising they cannot be read. Should not be a problem.

> `Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR utils.cc:218 Couldn't read '/data/config/dht.dat': No such file or directory (2) (transmission-3.00/libtransmission/utils.cc:218)`

Same for `dht.cat`.

> `Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR variant.cc:1136 Couldn't save '/data/config/settings.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)`

Configuration is rendered from UCI `/etc/config/transmission`, hence transmission itself cannot and should not save configuration.

> `Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR variant.cc:1136 Couldn't save '/data/config/bandwidth-groups.json': Permission denied (13) (transmission-3.00/libtransmission/variant.cc:1136)`

This has probably been added recently to transmission, I don't know this file and also don't see this error with the 3.00 release of transmission.

> `Sat Jul 23 21:34:29 2022 daemon.err transmission-daemon[23381]: [2022-07-23 14:34:29.647] ERR watchdir-inotify.cc:180 Couldn't watch '/data/watch': No such file or directory (2) (transmission-3.00/libtransmission/watchdir-inotify.cc:180)`

`/data/watch` indeed doesn't exist. Create it or remove the `watchdir` option.

trippleflux, 2022-07-28:
Getting the following when downloading : 
```
Permission denied (/data/transmission/done/Test[Test]/Test.mkv)
```

And the from log(Info) : 
```
Thu Jul 28 18:48:06 2022 user.notice transmission: Starting with 3905432000 virt mem
Thu Jul 28 18:48:07 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:48:07.971] ERR utils.cc:218 Couldn't read '/data/transmission/config/stats.benc': No such file or directory (2) (../libtransmission/utils.cc:218)
Thu Jul 28 18:48:07 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:48:07.971] ERR utils.cc:218 Couldn't read '/data/transmission/config/dht.dat': No such file or directory (2) (../libtransmission/utils.cc:218)
Thu Jul 28 18:48:07 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:48:07.971] ERR variant.cc:1136 Couldn't save '/data/transmission/config/settings.json': Permission denied (13) (../libtransmission/variant.cc:1136)
Thu Jul 28 18:48:07 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:48:07.971] ERR variant.cc:1136 Couldn't save '/data/transmission/config/bandwidth-groups.json': Permission denied (13) (../libtransmission/variant.cc:1136)
Thu Jul 28 18:49:20 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:20.971] ERR utils.cc:218 Couldn't read '/data/transmission/config/resume/232a4757fbed6827fba2082ebccf5a5d68892200.resume': No such file or directory (2) (../libtransmission/utils.cc:218)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR file-posix.cc:271 Couldn't create '/data/transmission/incomplete': Permission denied (13) (../libtransmission/file-posix.cc:271)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR open-files.cc:159 Couldn't create '/data/transmission/incomplete/Test[Test]': Permission denied (13) (../libtransmission/open-files.cc:159)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR Test[Test] Couldn't get '/data/transmission/incomplete/Test[Test]/Test.mkv.part': Permission denied (13) (../libtransmission/inout.cc:146)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR file-posix.cc:271 Couldn't create '/data/transmission/incomplete': Permission denied (13) (../libtransmission/file-posix.cc:271)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR open-files.cc:159 Couldn't create '/data/transmission/incomplete/Test[Test]': Permission denied (13) (../libtransmission/open-files.cc:159)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR Test[Test] Couldn't get '/data/transmission/incomplete/Test[Test]/Test.mkv.part': Permission denied (13) (../libtransmission/inout.cc:146)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR file-posix.cc:271 Couldn't create '/data/transmission/incomplete': Permission denied (13) (../libtransmission/file-posix.cc:271)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR open-files.cc:159 Couldn't create '/data/transmission/incomplete/Test[Test]': Permission denied (13) (../libtransmission/open-files.cc:159)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR Test[Test] Couldn't get '/data/transmission/incomplete/Test[Test]/Test.mkv.part': Permission denied (13) (../libtransmission/inout.cc:146)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR file-posix.cc:271 Couldn't create '/data/transmission/incomplete': Permission denied (13) (../libtransmission/file-posix.cc:271)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR open-files.cc:159 Couldn't create '/data/transmission/incomplete/Test[Test]': Permission denied (13) (../libtransmission/open-files.cc:159)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR Test[Test] Couldn't get '/data/transmission/incomplete/Test[Test]/Test.mkv.part': Permission denied (13) (../libtransmission/inout.cc:146)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR file-posix.cc:271 Couldn't create '/data/transmission/incomplete': Permission denied (13) (../libtransmission/file-posix.cc:271)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR open-files.cc:159 Couldn't create '/data/transmission/incomplete/Test[Test]': Permission denied (13) (../libtransmission/open-files.cc:159)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR Test[Test] Couldn't get '/data/transmission/incomplete/Test[Test]/Test.mkv.part': Permission denied (13) (../libtransmission/inout.cc:146)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR file-posix.cc:271 Couldn't create '/data/transmission/incomplete': Permission denied (13) (../libtransmission/file-posix.cc:271)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR open-files.cc:159 Couldn't create '/data/transmission/incomplete/Test[Test]': Permission denied (13) (../libtransmission/open-files.cc:159)
Thu Jul 28 18:49:34 2022 daemon.err transmission-daemon[2352]: [2022-07-28 11:49:34.973] ERR Test[Test] Couldn't get '/data/transmission/incomplete/Test[Test]/Test.mkv.part': Permission denied (13) (../libtr [... cut ...]
#[ "$incomplete_dir_enabled" = "1" ] && procd_add_jail_mount_rw "$incomplete_dir"
#[ "$watch_dir_enabled" = "1" ] && procd_add_jail_mount_rw "$watch_dir"
web_home="${web_home:-/usr/share/transmission/public_html}"
[ -d "$web_home" ] && procd_add_jail_mount "$web_home"
[ -f "$ca_bundle_file" ] && procd_add_jail_mount "$ca_bundle_file"
procd_close_instance
```

Managed to downloaded the test torrent, should I close this?, since it's not an official OpenWRT package?.

Thanks again @dangowrt for the helps and pointers.

trippleflux, 2022-08-01:
@dangowrt 
What is the correct way to create seccomp for the transmission-daemon?, should I eliminating all ujail related calls in the /etc/init.d/transmission and add ujail before the /usr/bin/transmission-daemon?.

Tried with /etc/init.d/tranmission trace but getting _**ujail: unable to load dependecies**_.

Tried with /sbin/utrace /usr/bin/transmission-daemon but only getting a few, how to make utrace sort of 'hooking' into the transmission-daemon process until it ended?.

Currently I have the following seccomp for the transmission-daemon : 
```
{
	"defaultAction": "SCMP_ACT_KILL_PROCESS",
	"syscalls": [
		{
			"names": [
				"accept",
				"accept4",
				"access",
				"arm_fadvise64_64",
				"bind",
				"brk",
				"clock_gettime",
				"clock_gettime64",
				"clone",
				"close",
				"connect",
				"dup2",
				"epoll_create1",
				"epoll_create",
				"epoll_ctl",
				"epoll_pwait",
				"epoll_wait",
				"exit",
				"exit_group",
				"faccessat",
				"fadvise64",
				"fadvise64_64",
				"fallocate",
				"fchmod",
				"fcntl",
				"fcntl64",
				"fstat",
				"fstat64",
				"fsync",
				"futex",
				"getdents64",
				"getegid32",
				"getenv",
				"geteuid32",
				"getgid32",
				"getpeername",
				"getpid",
				"getrandom",
				"getsockname",
				"getsockopt",
				"getuid",
				"getuid32",
				"inotify_add_watch",
				"inotify_init1",
				"inotify_rm_watch",
				"ioctl",
				"listen",
				"_llseek",
				"lseek",
				"madvise",
				"membarrier",
				"mkdir",
				"mkdirat",
				"mmap",
				"mmap2",
				"mprotect",
				"mremap",
				"munmap",
				"nanosleep",
				"newfstatat",
				"_newselect",
				"open",
				"openat",
				"pipe",
				"pipe2",
				"poll",
				"ppoll",
				"pread64",
				"prlimit64",
				"pselect6",
				"pwrite64",
				"quotactl",
				"read",
				"readlink",
				"readlinkat",
				"readv",
				"recvfrom",
				"recvmsg",
				"rename",
				"renameat",
				"rmdir",
				"rt_sigaction",
				"rt_sigprocmask",
				"rt_sigreturn",
				"select",
				"sendmsg",
				"sendto",
				"set_robust_list",
				"setsid",
				"setsockopt",
				"shutdown",
				"sigreturn",
				"socket",
				"socketpair",
				"stat",
				"stat64",
				"statfs",
				"statfs64",
				"statx",
				"time",
				"umask",
				"uname",
				"unlink",
				"unlinkat",
				"write",
				"writev"
			],
			"action": "SCMP_ACT_ALLOW"
		}
	]
}
```

With above seccomp, transmission-daemon still crashes.