openwrt/packages #19072: banip: multiple instances

KevinKapner, 2022-07-28:
Maintainer: @dibdot 
Environment: OpenWrt 21.02.3 on Fritzbox 4040

Description:

Hi everyone. Banip works great here, thank you very much and please keep up the good work. 
It seems not possible to create multiple instances for banip (via config file like it is e.g. for dnsmasq respectively and adblock). Would be great if it's not to time-consuming to implement. I've got several vlans and some wans (like different vpns), but I would like to block different things, for example more in guest network than for my private networks (and there maybe differentiate adult from kids). DNS-based blocking like adblock is (obviously) not equal, and so is blocking single IPs vs. blocking multiple ranges by ASNs. I hope I described the use case well enough. 

Tried it like this in /etc/config/banip just for fun: 
```
config banip 'global'
 list ban_ifaces 'wan'
 [truncated, including different things like inputchains, ban_asns etc.]

config banip 'guests'
 list ban_ifaces 'vpn'
 [truncated, including things like guest inputchains, some same and some other ban_asns, country codes etc.]
```

I searched in the support thread and for issues related to this but didn't find a thing. Maybe I'm just getting blind.

dibdot, 2022-08-11:
This kind of functionality is simply not implemented in banIP.

closed 2022-08-11 by dibdot

KevinKapner, 2022-08-12:
> This kind of functionality is simply not implemented in banIP.

Do you (or somebody else) have plans to implement this?