openwrt/packages #19078: Coova-chilli compile error using the latest OpenWrt Master (July 30 2022)

vochong, 2022-07-31:
Maintainer: Jaehoon You <teslamint@gmail.com>
Environment: ArchLinux,  Ubuntu

Description:

```
Coova-chilli compile works well in OpenWrt 22.03 branch but always fails in the latest OpenWrt Master.

$ make -j1 V=s package/coova-chilli/compile

make[2]: Entering directory '/home/python/Venv/NSS-MASTER/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/python/Venv/NSS-MASTER/openwrt/scripts/config'
make[1]: Entering directory '/home/python/Venv/NSS-MASTER/openwrt'
make[2]: Entering directory '/home/python/Venv/NSS-MASTER/openwrt/package/libs/toolchain'
echo "libc" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libgcc" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libatomic" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "libpthread" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
echo "librt" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/python/Venv/NSS-MASTER/openwrt/package/libs/toolchain'
time: package/libs/toolchain/compile#0.19#0.18#0.34
make[2]: Entering directory '/home/python/Venv/NSS-MASTER/openwrt/package/firmware/linux-firmware'
echo "ath10k-board-qca9984" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux-firmware.default.install
echo "ath10k-firmware-qca9984" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux-firmware.default.install
make[2]: Leaving directory '/home/python/Venv/NSS-MASTER/openwrt/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#1.35#0.37#1.65
make[2]: Entering directory '/home/python/Venv/NSS-MASTER/openwrt/package/kernel/gpio-button-hotplug'
echo "kmod-gpio-button-hotplug" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/gpio-button-hotplug.default.install
make[2]: Leaving directory '/home/python/Venv/NSS-MASTER/openwrt/package/kernel/gpio-button-hotplug'
time: package/kernel/gpio-button-hotplug/compile#0.19#0.24#0.40
make[2]: Entering directory '/home/python/Venv/NSS-MASTER/openwrt/package/kernel/linux'
echo "kernel" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-ata-core" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-ata-ahci" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-ata-ahci-platform" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-dax" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-dm" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-scsi-core" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-cdrom" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-acompress" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-aead" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-authenc" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-cbc" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-ccm" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-cmac" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-crc32" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-crc32c" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-ctr" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-deflate" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-des" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-ecb" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-echainiv" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-gcm" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-gf128" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-ghash" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-hash" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-hmac" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
WARNING: kmod-crypto-hw-padlock is not available in the kernel config - generating empty package
echo "kmod-crypto-kpp" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-lib-chacha20" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-lib-chacha20poly1305" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-lib-curve25519" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-lib-poly1305" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-manager" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-md5" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-misc" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-null" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
echo "kmod-crypto-rng" >> /home/python/Venv/NSS-MASTER/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/linux.default.install
[... 153 lines cut ...]
Making all in src
make[5]: Entering directory '/home/python/Venv/NSS-MASTER/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/coova-chilli-1.6/src'
Making all in linux
[... cut ...]
make[2]: *** [Makefile:150: /home/python/Venv/NSS-MASTER/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/coova-chilli-1.6/.built] Error 2
make[2]: Leaving directory '/home/python/Venv/NSS-MASTER/openwrt/feeds/packages/net/coova-chilli'
time: package/feeds/packages/coova-chilli/compile#0.34#0.29#0.58
ERROR: package/feeds/packages/coova-chilli failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/coova-chilli/compile] Error 1
make[1]: Leaving directory '/home/python/Venv/NSS-MASTER/openwrt'
make: *** [/home/python/Venv/NSS-MASTER/openwrt/include/toplevel.mk:230: package/coova-chilli/compile] Error 2
```

teslamint, 2022-08-20:
fixes in #19177

neheb, 2022-08-21:
merged.

closed 2022-08-21 by neheb