openwrt/packages #19086: tailscale: improve packaging

CAFxX, 2022-08-02:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @BKPepe (?)
Environment: 21.03

Description: 

The tailscale and tailscaled binaries are quite large, even when squashfs'd:

[image]

The tailscale [documentation](https://tailscale.com/kb/1207/small-tailscale/) mentions that, exactly to address this, tailscaled can be optionally built including the cli functionality, so that it can be symlinked busybox-style to support both the tailscaled and tailscale cli roles:

```
go build -o tailscale.combined -tags ts_include_cli ./cmd/tailscaled
ln -s tailscale.combined tailscale
ln -s tailscale.combined tailscaled
```

As currently tailscale depends on tailscaled, we could maybe move from the current two packages

- tailscaled 
- tailscale (depends on tailscaled)

to two packages and a virtual package

- tailscaled
- tailscaled-full (provides tailscale)

where tailscaled-full is the tailscaled command built using the ts_include_ci tag.

Quickly building the packages locally and compressing with upx (as a stand-in for squashfs) seems to indicate that this approach would save ~1.3MB in a firmware image that includes tailscale (e.g. such as those built via attended-sysupgrades).

```
        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
   9961472 ->   2308728   23.18%   linux/mips    tailscale
  26083328 ->   4790500   18.37%   linux/mips    tailscale.combined
  17235968 ->   3765896   21.85%   linux/mips    tailscaled
```

BKPepe, 2022-08-02:
> As currently tailscale depends on tailscaled, we could maybe move from the current two packages

Feel free to create a pull request to address this.

> Quickly building the packages locally and compressing with upx (as a stand-in for squashfs) seems to indicate that this approach would save ~1.3MB in a firmware image that includes tailscale (e.g. such as those built via attended-sysupgrades).

Please see: https://github.com/openwrt/packages/issues/17984#issuecomment-1059674191

CAFxX, 2022-08-02:
> Please see: https://github.com/openwrt/packages/issues/17984#issuecomment-1059674191

Yeah as I mentioned, I used upx just as a quick stand-in for squashfs (since internally both squashfs and upx use lzma). I am aware upx wouldn't help with squashfs. I was not advocating we should use upx. It was just to give an estimate of how much we would save after lzma compression.