openwrt/packages #19099: squid: Build fails with `SQUID_with-libxml2=y` with `Libxml2Parser.cc:94:5: error: 'xmlSetFeature' was not declared in this scope; did you mean 'xmlHasFeature'?` in 22.03.

dreirund, 2022-08-03:
### Maintainer:

@aTanW

### Environment:

* Building for Xunlog OrangePi Plus (sun8i):  
  `grep ^CONFIG_TARGET .config`:
  ```
  CONFIG_TARGET_sunxi=y
  CONFIG_TARGET_sunxi_cortexa7=y
  CONFIG_TARGET_sunxi_cortexa7_DEVICE_xunlong_orangepi-plus=y
  CONFIG_TARGET_BOARD="sunxi"
  CONFIG_TARGET_SUBTARGET="cortexa7"
  CONFIG_TARGET_PROFILE="DEVICE_xunlong_orangepi-plus"
  CONFIG_TARGET_ARCH_PACKAGES="arm_cortex-a7_neon-vfpv4"
  [...]
  ```
* OpenWrt version:
  - Git branch `openwrt-22.03` from `git://git.openwrt.org/openwrt/openwrt.git`,
  - initially checked out on 2022-07-05,
  - latest `git pull` and `./scripts/feeds update -a && ./scripts/feeds update -a` on 2022-08-02, ca. 12:23 GMT  
    *(I give up on posting latest commit date and hash, since `git log` gives me some old dates, and I don't know how I can make `git log` printing out the last operation. I don't know much about `git`.)*
* Building on GNU/Linux on x86_64-architecture (Distribution: [&rarr;&nbsp;Artix Linux](http://artixlinux.org)).
* OpenWrt configuration: [&rarr;&nbsp;`.config`](http://ix.io/46xL),
* `grep -i squid .config | grep -v i18n`:  
  ```
  CONFIG_PACKAGE_luci-app-squid=y
  CONFIG_PACKAGE_squid=y
  CONFIG_SQUID_enable-ipv6=y
  CONFIG_SQUID_enable-snmp=y
  CONFIG_SQUID_enable-icmp=y
  CONFIG_SQUID_enable-icap-client=y
  # CONFIG_SQUID_enable-dlmalloc is not set
  CONFIG_SQUID_enable-ssl-crtd=y
  # CONFIG_SQUID_auth-basic is not set
  CONFIG_SQUID_auth-digest=y
  CONFIG_SQUID_auth-negotiate=y
  CONFIG_SQUID_auth-ntlm=y
  CONFIG_SQUID_use-openssl=y
  # CONFIG_SQUID_use-gnutls is not set
  CONFIG_SQUID_with-libcap=y
  CONFIG_SQUID_with-nettle=y
  # CONFIG_SQUID_with-expat is not set
  CONFIG_SQUID_with-libxml2=y
  CONFIG_PACKAGE_squid-mod-cachemgr=y
  ```

### Description:

After a `make clean; ./scripts/feeds clean; git pull && ./scripts/feeds update -a && ./scripts/feeds install -a && make oldconfig && make menuconfig`, a `make -j6` fails in above configuration on `package/feeds/packages/squid`.
An explicit run of  
`make -j1 V=sc package/feeds/packages/squid/compile`  
fails with `Libxml2Parser.cc:94:5: error: 'xmlSetFeature' was not declared in this scope; did you mean 'xmlHasFeature'?`:  
```
[...]
OpenWrt-libtool: compile:  ccache_cxx -DHAVE_CONFIG_H -DDEFAULT_CONFIG_FILE=\"/etc/squid/squid.conf\" -DDEFAULT_SQUID_DATA_DIR=\"/usr/share/squid\" -DDEFAULT_SQUID_CONFIG_DIR=\"/etc/squid\" -I../.. -I../../include -I../../lib -I../../src -I../../include -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libxml2 -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Woverloaded-virtual -Werror -pipe -D_REENTRANT -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libxml2 -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16=squid-4.16 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error -MT Context.lo -MD -MP -MF .deps/Context.Tpo -c Context.cc  -fPIC -DPIC -o .libs/Context.o
depbase=`echo Libxml2Parser.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/sh ../../libtool  --tag=CXX   --mode=compile ccache_cxx -DHAVE_CONFIG_H -DDEFAULT_CONFIG_FILE=\"/etc/squid/squid.conf\" -DDEFAULT_SQUID_DATA_DIR=\"/usr/share/squid\" -DDEFAULT_SQUID_CONFIG_DIR=\"/etc/squid\"   -I../.. -I../../include -I../../lib -I../../src -I../../include    -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include  -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libxml2  -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Woverloaded-virtual -Werror -pipe -D_REENTRANT -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libxml2  -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include  -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16=squid-4.16 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error  -MT Libxml2Parser.lo -MD -MP -MF $depbase.Tpo -c -o Libxml2Parser.lo Libxml2Parser.cc &&\
mv -f $depbase.Tpo $depbase.Plo
OpenWrt-libtool: compile:  ccache_cxx -DHAVE_CONFIG_H -DDEFAULT_CONFIG_FILE=\"/etc/squid/squid.conf\" -DDEFAULT_SQUID_DATA_DIR=\"/usr/share/squid\" -DDEFAULT_SQUID_CONFIG_DIR=\"/etc/squid\" -I../.. -I../../include -I../../lib -I../../src -I../../include -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/usr/include -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include/fortify -I/[...]/openwrt-22.03/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.2.0_musl_eabi/include -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libxml2 -Wall -Wpointer-arith -Wwrite-strings -Wcomments -Wshadow -Woverloaded-virtual -Werror -pipe -D_REENTRANT -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/libxml2 -I/[...]/openwrt-22.03/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include -O3 -pipe -g0 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16=squid-4.16 -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error -MT Libxml2Parser.lo -MD -MP -MF .deps/Libxml2Parser.Tpo -c Libxml2Parser.cc  -fPIC -DPIC -o .libs/Libxml2Parser.o
Libxml2Parser.cc: In constructor 'ESILibxml2Parser::ESILibxml2Parser(ESIParserClient*)':
Libxml2Parser.cc:94:5: error: 'xmlSetFeature' was not declared in this scope; did you mean 'xmlHasFeature'?
   94 |     xmlSetFeature(parser, "substitute entities", 0);
      |     ^~~~~~~~~~~~~
      |     xmlHasFeature
make[6]: *** [Makefile:863: Libxml2Parser.lo] Error 1
make[6]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16/src/esi'
make[5]: *** [Makefile:6948: all-recursive] Error 1
make[5]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16/src'
make[4]: *** [Makefile:5964: all] Error 2
make[4]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16/src'
make[3]: *** [Makefile:584: all-recursive] Error 1
make[3]: Leaving directory '/[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16'
make[2]: *** [Makefile:157: /[...]/openwrt-22.03/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/squid-4.16/.built] Error 2
make[2]: Leaving directory '/[...]/openwrt-22.03/feeds/packages/net/squid'
time: package/feeds/packages/squid/compile#150.86#33.23#179.83
    ERROR: package/feeds/packages/squid failed to build.
[...]
```

[&rarr;&nbsp;Full `make` output](https://github.com/openwrt/packages/files/9251535/make.build-error-xml.log).

&zwnj;

Deselecting `CONFIG_SQUID_with-libxml2` does mitigate this build issue. Choosing `CONFIG_SQUID_with-expat=y` works fine.

The-BB, 2022-08-07:
need to compile libxml2 with "--with-legacy" (https://github.com/openwrt/packages/blob/master/libs/libxml2/Makefile#L85) 
see: https://gitlab.gnome.org/GNOME/libxml2/-/blob/master/include/libxml/parser.h#L993-1011

dreirund, 2022-08-08:
*@The-BB wrote*:  
> *need to compile libxml2 with "--with-legacy"*

Is this then an issue which should be adressed to `package/feeds/packages/libxml2` (@mhei), or an issue within the squid package, that selecting the option `SQUID_with-libxml2=y` changes the build options for `package/feeds/packages/libxml2`? (I don't know how the OpenWrt build system works, this questions are to maybe (re)direct the issue to the correct place.)

mhei, 2022-08-10:
I think it is an issue with squid itself. It should be updated to not use the legacy stuff of libxml2 anymore. While we easily could compile libxml2 with the mentioned option `--with-legacy`, this would increase the library size - I'd prefer to prevent this.

mhei, 2022-08-30:
@dreirund: could you test my PR whether there is a functionality regression or did you only tried to compile squid with selected options?

dreirund, 2022-08-30:
> @dreirund: could you test my PR whether there is a functionality
> regression or did you only tried to compile squid with selected
> options?

I don't use squid yet. I just compiled and reported my compilation
error findings.

Regards!

mhei, 2022-10-09:
Fixed with #19438.

commit aaedc165 (2022-09-25): squid: fix compilation with libxml (fixes #19099)

Add a patch which removes a call in Libxml2Parser.cc to 'xmlSetFeature'.
This function belongs to the 'depreciated' API part and is not
available in OpenWrt builds.

According to my understanding, this call can be removed safely since
it disables the feature "substitute entities" which is disabled by default.

(cherry picked from commit 3ec47dc85cc4b191be1b2fee3195680343f770e1)
commit d6c7e63c (2022-09-25): Merge pull request #19438 from mhei/22.03-squid-libxml2-backport

[22.03] squid: fix compilation with libxml (fixes #19099)
closed 2022-10-09 by mhei