openwrt/packages #19101: jsonschema is installed in the system with name unknown

BKPepe, 2022-08-03:
Maintainer: @jmarcet 
Environment: Turris 1.1, powerpc_8540, OpenWrt master

Description:

jsonschema is not detected in the system as it shows the name of the package as UNKNOWN. This can be verified by opkg files. 

```
root@turris:~# opkg files python3-jsonschema
Package python3-jsonschema (4.9.0-3.10-1) is installed on root and has the following files:
/usr/lib/python3.10/site-packages/UNKNOWN-4.9.0-py3.10.egg-info/PKG-INFO
/usr/lib/python3.10/site-packages/UNKNOWN-4.9.0-py3.10.egg-info/top_level.txt
/usr/lib/python3.10/site-packages/UNKNOWN-4.9.0-py3.10.egg-info/SOURCES.txt
/usr/lib/python3.10/site-packages/UNKNOWN-4.9.0-py3.10.egg-info/dependency_links.txt
```

BKPepe, 2022-08-18:
ping @jmarcet

jmarcet, 2022-09-06:
@BKPepe Ouch! Somehow I could import it in ipython, I'll double check.

jmarcet, 2022-09-08:
@BKPepe Please see #19267

closed 2024-09-25 by BKPepe