openwrt/packages #19102: fail2ban: postinst script: permission denied for /etc/init.d

BKPepe, 2022-08-03:
Maintainer: @erdoukki 
Environment: Turris 1.1, powerpc_8540, OpenWrt master

Description:

```
root@turris:~# opkg install fail2ban
Installing fail2ban (0.11.2-3.10-6) to root...
Downloading https://repo.turris.cz/hbd/turris1x/packages/packages/fail2ban_0.11.2-3.10-6_powerpc_8548.ipk
Configuring fail2ban.
Adding the following UCI config:
 firewall.cfg10af89='include'
firewall.cfg10af89.path='/etc/firewall.fail2ban'
firewall.cfg10af89.enabled='1'
firewall.cfg10af89.reload='1'
//usr/lib/opkg/info/fail2ban.postinst: line 304: /etc/init.d/: Permission denied
```

The last row should not be there. Why the permission was denied?

BKPepe, 2022-08-03:
Are you sure that your's package should add something to the UCI config by default without my approval? I don't like this behavior. Also, if you install fail2ban in LuCI, you would not know that it added something to firewall's config.

BKPepe, 2022-08-03:
Same happens for prerm script.
```
Removing package fail2ban from root...
//usr/lib/opkg/info/fail2ban.prerm: line 196: /etc/init.d/: Permission denied
//usr/lib/opkg/info/fail2ban.prerm: line 198: /etc/init.d/: Permission denied
```

peci1, 2022-08-23:
This seems to be a turris-specific issue: https://forum.turris.cz/t/postinst-script-permission-denied-for-etc-init-d/17473/6?u=peci1 .

jow-, 2022-08-23:
See https://forum.turris.cz/t/postinst-script-permission-denied-for-etc-init-d/17473/6

closed 2022-08-23 by jow-