openwrt/packages #19107: travelmate: VPN connection is always terminated, even if option is off in settings

schuettecarsten, 2022-08-04:
I am using travelmate on openwrt 21.02.3 without problems. I have configured several networks in trm, all have the VPN option off. I also have a wireguard VPN tunnel that should not be managed by travelmate.

All this work fine in 21.02.3

After updating to 22.03.0-rc6, travelmate always terminates the wireguard interface. I think it should not do this when the VPN option in trm is not used at all.

```
config travelmate 'global'
        option trm_captive '1'
        option trm_proactive '1'
        option trm_autoadd '0'
        option trm_mail '0'
        option trm_debug '0'
        option trm_enabled '1'
        option trm_radio 'radio1'
        option trm_netcheck '1'
        option trm_randomize '1'
        option trm_iface 'wwan'

config uplink
        option device 'radio1'
        option ssid '<REMOVED>' 
        option con_start_expiry '0'
        option con_end_expiry '0'
        option enabled '1'
        option con_start '2022.08.04-06:12:12'
```

dibdot, 2022-08-11:
> I also have a wireguard VPN tunnel that should not be managed by travelmate.

Could you please describe your usecase for manually managing the vpn connection  on a travel router ?

Thanks!

schuettecarsten, 2022-08-13:
I use the VPN to access only three IPs on my private network. Regardless if I use travel Wifi, an attached UMTS modem or a wired connection. OpenWRT manages the VPN connection for me, I do not need Travelmate to try this. That's why I do not use the VPN functionality in Travelmate and in that case, I expect that Travelmate does not care about the Wireguard VPN at all.

PR #19214 (2022-08-20): travelmate: update 2.1.0
Maintainer: me / @dibdot
Compile tested: not relevant, script only
Run tested: GL.iNet GL-MT1300, OpenWrt SNAPSHOT r20333-a9eb5cce7e

Description:
* vpn support can be disabled (enabled by default), fixes #19107 (see trm_vpn option)
* vpn support can be limited to certain interfaces (see trm_vpnifacelist list option)
* openvpn support works now per instance (same as wireguard today)
* add an auto-login script for tplink-omada hotspots provided by Sebastian Muszynski <basti@linkt.de>
* remove pipefail command, see #19043 for reference
commit 963137dc (2022-08-20): travelmate: update 2.1.0

* vpn support can be disabled (enabled by default), fixes #19107 (see trm_vpn option)
* vpn support can be limited to certain interfaces (see trm_vpnifacelist list option)
* openvpn support works now per instance (same as wireguard today)
* add an auto-login script for tplink-omada hotspots provided by Sebastian Muszynski <basti@linkt.de>
* remove pipefail command, see #19043 for reference
commit 1e53ad68 (2022-08-20): travelmate: update 2.1.0

* vpn support can be disabled (enabled by default), fixes #19107 (see trm_vpn option)
* vpn support can be limited to certain interfaces (see trm_vpnifacelist list option)
* openvpn support works now per instance (same as wireguard today)
* add an auto-login script for tplink-omada hotspots provided by Sebastian Muszynski <basti@linkt.de>
* remove pipefail command, see #19043 for reference

(cherry picked from commit 963137dcc5bbfca4810860a8d85688b612d1c955)
closed 2022-08-20 by dibdot