openwrt/packages #19111: `squashfs-tools-mksquashfs:` add support for zstd compression algorithm

Livy-Inverse, 2022-08-04:
Maintainer: @commodo
Environment: (put here arch, model, OpenWrt version)

Description: `mksquashfs` does not support Facebook Zstandard as a compression option. Using gzip and xz is really too slow for an embedded device. Please consider adding support for this algorithm.

```
root@OpenWrt:~# mksquashfs -help
[...]
Compressors available and compressor specific options:
        gzip (default)
          -Xcompression-level <compression-level>
                <compression-level> should be 1 .. 9 (default 9)
          -Xwindow-size <window-size>
                <window-size> should be 8 .. 15 (default 15)
          -Xstrategy strategy1,strategy2,...,strategyN
                Compress using strategy1,strategy2,...,strategyN in turn
                and choose the best compression.
                Available strategies: default, filtered, huffman_only,
                run_length_encoded and fixed
        xz
          -Xbcj filter1,filter2,...,filterN
                Compress using filter1,filter2,...,filterN in turn
                (in addition to no filter), and choose the best compression.
                Available filters: x86, arm, armthumb, powerpc, sparc, ia64
          -Xdict-size <dict-size>
                Use <dict-size> as the XZ dictionary size.  The dictionary size
                can be specified as a percentage of the block size, or as an
                absolute value.  The dictionary size must be less than or equal
                to the block size and 8192 bytes or larger.  It must also be
                storable in the xz header as either 2^n or as 2^n+2^(n+1).
                Example dict-sizes are 75%, 50%, 37.5%, 25%, or 32K, 16K, 8K
                etc.
```

commodo, 2022-08-05:
hey,

so we support ZSTD, we just don't enable it by default:

see :  https://github.com/openwrt/packages/blob/master/utils/squashfs-tools/Config.in

i don't see a reason not to do it; the list of default compression algorithms on OpenWrt is a bit old/out-dated;

Livy-Inverse, 2022-08-05:
Thanks a lot. It takes me 5 hours straight to make an "xz" SquashFS image out of 14GiB mssql-server database files on a Qualcomm IQP4019. Pointing a thermal camera to the router CPU drives me crazy (~70°C).

Making a `.tar.zstd` file at level 3 only takes ~30 mins, at a cost of a little bigger size. I expect similar result when `zstd` is supported. The benefit is to mount the image at anytime, and the reading speed from the mount point shall be day and night compared to that of `xz`.

I am not familiar with git and build process. When should I expect the changes to be available on OpenWrt repository?

commodo, 2022-08-05:
See https://github.com/openwrt/packages/pull/19117 & https://github.com/openwrt/packages/pull/19114

commodo, 2022-08-08:
Closing this as the PRs are merged.

closed 2022-08-08 by commodo