openwrt/packages #19115: strongswan: version 5.9.6 introduced new plugin (kdf), new subpackage strongswan-mod-kdf is needed

GitArUs, 2022-08-05:
Changelog: https://github.com/strongswan/strongswan/releases/tag/5.9.6

libstrongswan-kdf.so is compiled, but not packaged.

Problems with ikev2 were introduced most likely because of missing kdf plugin:

[IKE] KDF_PRF with PRF_UNDEFINED not supported
[IKE] key derivation failed

pprindeville, 2022-08-07:
@Thermi Do you have time to look into this?

Neustradamus, 2022-08-09:
@GitArUs: A PR for 5.9.7 version here:
- https://github.com/openwrt/packages/pull/19128

GitArUs, 2022-08-10:
PR #19128 looks good to me. Presence of kdf plugin indeed fixes issue with ikev2. Thanks.

Neustradamus, 2022-08-10:
@pprindeville has needed testers, can you reply on the PR?

Thermi, 2022-08-10:
We're not yet done with this because the botan, openssl. and wolfssl plugins for example also provides KDFs, so the kdf plugin is actually optional.

pprindeville, 2022-08-10:
Doh.  Saw @Thermi's comment too late for PR #19128.

Thermi, 2022-08-10:
don't mind, we'll do it better next time.

pprindeville, 2022-08-10:
Want to take a stab at it?

Thermi, 2022-08-10:
This should be a reasonable implementation of that: https://github.com/openwrt/packages/pull/19145

Thermi, 2022-08-15:
done

GitArUs, 2022-08-19:
Works for me. Thank You. I'm closing this.

closed 2022-08-19 by GitArUs