openwrt/packages #19127: unbound: config zone of type forward_zone not creating a forward-zone:

brianjmurrell, 2022-08-07:
Maintainer: @EricLuehrsen 
Environment: ath-79(put here arch, model, OpenWrt version) ath79/generic GL.iNet GL-AR750 21.02.3 r16554-1d4dea6d4f

Description:

When adding a `config_zone` for type `forward_zone` given the following config:
```
config unbound 'ub_main'
	option dns64 '0'
	option domain 'lan'
	option edns_size '1232'
	option extended_stats '0'
	option hide_binddata '1'
	option interface_auto '1'
	option localservice '1'
	option manual_conf '0'
	option num_threads '1'
	option protocol 'default'
	option rate_limit '0'
	option rebind_localhost '0'
	option recursion 'default'
	option resource 'default'
	option root_age '9'
	option ttl_min '120'
	option ttl_neg_max '1000'
	option unbound_control '1'
	option validator '0'
	option verbosity '1'
	list iface_wan 'wan'
	option enabled '1'
	option listen_port '53'
	option dhcp_link 'dnsmasq'
	option rebind_protection '0'
	list iface_trig 'lan'
	list iface_trig 'wan'

config zone 'auth_icann'
	option enabled '0'
	option fallback '1'
	option url_dir 'https://www.internic.net/domain/'
	option zone_type 'auth_zone'
	list server 'lax.xfr.dns.icann.org'
	list server 'iad.xfr.dns.icann.org'
	list zone_name '.'
	list zone_name 'arpa.'
	list zone_name 'in-addr.arpa.'
	list zone_name 'ip6.arpa.'

config zone 'fwd_isp'
	option enabled '0'
	option fallback '1'
	option resolv_conf '1'
	option zone_type 'forward_zone'
	list zone_name 'isp-bill.example.com.'
	list zone_name 'isp-mail.example.net.'

config zone 'fwd_google'
	option enabled '0'
	option fallback '1'
	option tls_index 'dns.google'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '8.8.4.4'
	list server '8.8.8.8'
	list server '2001:4860:4860::8844'
	list server '2001:4860:4860::8888'
	list zone_name '.'

config zone 'fwd_cloudflare'
	option enabled '0'
	option fallback '1'
	option tls_index 'cloudflare-dns.com'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '1.1.1.1'
	list server '1.0.0.1'
	list server '2606:4700:4700::1111'
	list server '2606:4700:4700::1001'
	list zone_name '.'

config zone 'fwd_example_com'
	option fallback '0'
	option enabled '1'
	option zone_type 'forward_zone'
	option zone_name 'example.com'
	option server '10.75.22.247'

config zone 'fwd_10_75_22'
	option fallback '1'
	option enabled '1'
	option zone_type 'forward_zone'
	option zone_name '22.75.10.in-addr.arpa'
	option server '10.75.22.247'
```

the `example.com` forward zone (or it's reverse) don't actually appear in the unbound configuration:

```
# /var/lib/unbound/unbound.conf generated by UCI 2022-08-07T08:07:23-04:00
# /var/lib/unbound/server.conf.tmp generated by UCI 2022-08-07T08:07:23-04:00
server:
  username: unbound
  chroot: /var/lib/unbound
  directory: /var/lib/unbound
  pidfile: /var/run/unbound.pid
  tls-cert-bundle: /etc/ssl/certs/ca-certificates.crt

  num-threads: 1
  msg-cache-slabs: 1
  rrset-cache-slabs: 1
  infra-cache-slabs: 1
  key-cache-slabs: 1
  ratelimit-slabs: 1
  ip-ratelimit-slabs: 1

  use-syslog: yes
  statistics-interval: 0
  statistics-cumulative: no
  verbosity: 1
  extended-statistics: no

  interface-automatic: yes
  edns-buffer-size: 1232
  port: 53
  outgoing-port-permit: 10240-65535
  module-config: "iterator"

  cache-min-ttl: 120
  cache-max-ttl: 72000
  cache-max-negative-ttl: 1000
  val-bogus-ttl: 300
  infra-host-ttl: 900

  hide-identity: yes
  hide-version: yes

  access-control: 10.75.24.254/24 allow
  access-control: 192.168.102.254/24 allow
  access-control: 127.0.0.0/8 allow
  access-control: ::1/128 allow
  access-control: fe80::/10 allow

# /var/lib/unbound/host.conf.tmp generated by UCI 2022-08-07T08:07:23-04:00
# Local zone is handled by dnsmasq

# /var/lib/unbound/dnsmasq_srv.conf.tmp generated by UCI 2022-08-07T08:07:23-04:00
  do-not-query-localhost: no

  domain-insecure: lan
  private-domain: lan
  local-zone: lan transparent

  domain-insecure: 24.75.10.in-addr.arpa
  private-domain: 24.75.10.in-addr.arpa
  local-zone: 24.75.10.in-addr.arpa transparent

  domain-insecure: 102.168.192.in-addr.arpa
  private-domain: 102.168.192.in-addr.arpa
  local-zone: 102.168.192.in-addr.arpa transparent

include: /var/lib/unbound/adb_list.*

include: /var/lib/unbound/unbound_srv.conf

# /var/lib/unbound/zone.conf.tmp generated by UCI 2022-08-07T08:07:22-04:00
 # Special zone  was not enabled or had UCI conflicts.

 # Special zone  was not enabled or had UCI conflicts.

 # Special zone  was not enabled or had UCI conflicts.

 # Special zone  was not enabled or had UCI conflicts.

# /var/lib/unbound/ctrl.conf.tmp generated by UCI 2022-08-07T08:07:23-04:00
remote-control:
  control-enable: yes
  control-use-cert: no
  control-interface: 127.0.0.1
  control-interface: ::1

# /var/lib/unbound/dnsmasq_ext.conf.tmp generated by UCI 2022-08-07T08:07:23-04:00
forward-zone:
  name: lan
  forward-first: no
  forward-addr: 127.0.0.1@5353

forward-zone:
  name: 24.75.10.in-addr.arpa
  forward-first: no
  forward-addr: 127.0.0.1@5353

forward-zone:
  name: 102.168.192.in-addr.arpa
  forward-first: no
  forward-addr: 127.0.0.1@5353

include: /var/lib/unbound/unbound_ext.conf
```

EricLuehrsen, 2022-08-08:
So you are trying to forward local rev. DNS to dnsmasq and forward local rev. DNS somewhere else?

brianjmurrell, 2022-08-08:
No.  Please notice the subtle difference in IP address ranges.  The local network is 10.75.24.0/24 which is handled by the local `dnsmasq` (for now, I intend to drop it in favour of `unbound` but just have not taken that step yet) and the forward that I am trying to install which is not working is 10.75.22.0/24.

But really, even if I were, that is not even the issue.  The issue is that the UCI config I have added for `config zone 'fwd_example_com'` and `config zone 'fwd_10_75_22'` is not even being unravelled and turned into unbound configuration at all.

brianjmurrell, 2022-08-09:
Upon the clarification of the two different subnets, do you have any idea why my config is not producing the forwards?

dave14305, 2022-08-11:
zone_name and server are list entries, but you show them as option entries. Try changing them?

> config zone 'fwd_example_com'
> 	option fallback '0'
> 	option enabled '1'
> 	option zone_type 'forward_zone'
> 	option zone_name 'example.com'
> 	option server '10.75.22.247'
>
> config zone 'fwd_10_75_22' [...]

brianjmurrell, 2022-08-26:
@dave14305 It was Luci that created those zone entries in `/etc/config/unbound` including the use of `option` for the `server` entries.  IIUC, a single `list` item can be expressed equally as an `option`.

In any case, just to prove the above I have changed the entries to:

```

config zone 'fwd_example_com'
	option fallback '0'
	option enabled '1'
	option zone_type 'forward_zone'
	option zone_name 'example.com'
	list server '10.75.22.247'

config zone 'fwd_10_75_22'
	option fallback '1'
	option enabled '1'
	option zone_type 'forward_zone'
	option zone_name '22.75.10.in-addr.arpa'
	list server '10.75.22.247'
```

and unsurprisingly, it still does not work.  But given the proposed "shot in the dark" solution I'm assuming my config looks otherwise correct?  There is nothing glaringly wrong with it, yes?  So maybe we are looking at some kind of bug here?

dave14305, 2022-08-26:
Switch the zone_name to a list entry also, just to fully eliminate it as a possibility.

brianjmurrell, 2022-12-22:
Indeed, switching `zone_name` to a `list` entry does resolve this problem.

But my understanding has always been that a `list` of one and an `option` are supposed to be the same thing and always interchangable.  `uci` certainly thinks they are as the following `uci batch` commands:

```
set unbound.fwd_example_com=zone
set unbound.fwd_example_com.fallback='0'
set unbound.fwd_example_com.enabled='1'
set unbound.fwd_example_com.zone_type='forward_zone'
set unbound.fwd_example_com.zone_name='example.com'
set unbound.fwd_example_com.server='10.75.22.247'
set unbound.fwd_10_75_22=zone
set unbound.fwd_10_75_22.fallback='1'
set unbound.fwd_10_75_22.enabled='1'
set unbound.fwd_10_75_22.zone_type='forward_zone'
set unbound.fwd_10_75_22.zone_name='22.75.10.in-addr.arpa'
set unbound.fwd_10_75_22.server='10.75.22.247'
```

are what are creating the `option`s rather than `list`s.