openwrt/packages #19139: acme: issue with certificate durations lower than 30d

GGG-KILLER, 2022-08-09:
# Issue template (remove lines from top till here)

Maintainer: @tohojo 
Environment: (put here arch, model, OpenWrt version)
- Arch: Qualcomm Atheros QCA956X ver 1 rev 0
- Model: TP-Link Archer C7 v5
- OpenWRT Version: OpenWrt 22.03.0-rc4 r19426-2b1941e47d / LuCI openwrt-22.03 branch git-22.140.66268-ef99568

Description:

When using a CA server that only emits certificates with duration lower than 30 days (e.g.: in my case I have a Step CA server that only emits 24h certs and I have everything set to check for renewals every hour), the acme scripts incorrectly skips the renewal for the certificates.
It seems to not respect the set "Days until renewal" setting **at all** (ideally this would be a setting that accepts hours as some day I'd like to make the certs last 6 hours or less).

<details>
<summary>Acme logs</summary>

```
Tue Aug  9 00:00:00 2022 cron.err crond[1768]: USER root pid 7479 cmd /etc/init.d/acme start
Tue Aug  9 00:00:01 2022 daemon.info acme: Running pre checks for openwrt.lan.
Tue Aug  9 00:00:01 2022 daemon.err run-acme[7502]: acme: Running pre checks for openwrt.lan.
Tue Aug  9 00:00:01 2022 daemon.debug acme: port80 listens: 24777/uhttpd 24777/uhttpd
Tue Aug  9 00:00:01 2022 daemon.err run-acme[7502]: acme: port80 listens: 24777/uhttpd 24777/uhttpd
Tue Aug  9 00:00:01 2022 daemon.debug acme: Found uhttpd listening on port 80; trying to disable.
Tue Aug  9 00:00:01 2022 daemon.err run-acme[7502]: acme: Found uhttpd listening on port 80; trying to disable.
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: BusyBox v1.35.0 (2022-06-10 07:20:21 UTC) multi-call binary.
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]:
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Usage: basename FILE [SUFFIX] | -a FILE... | -s SUFFIX FILE...
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]:
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]:
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: 	-a		All arguments are FILEs
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: 	-s SUFFIX	Remove SUFFIX (implies -a)
Tue Aug  9 00:00:02 2022 daemon.debug acme: Nothing listening on port 80.
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: acme: Nothing listening on port 80.
Tue Aug  9 00:00:02 2022 daemon.debug acme: added nft rule: handle 491
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: acme: added nft rule: handle 491
Tue Aug  9 00:00:02 2022 daemon.info acme: Running ACME for openwrt.lan
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: acme: Running ACME for openwrt.lan
Tue Aug  9 00:00:02 2022 daemon.info acme: Found previous cert config. Issuing renew.
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: acme: Found previous cert config. Issuing renew.
Tue Aug  9 00:00:02 2022 daemon.debug acme: Running acme.sh as '/usr/lib/acme/acme.sh --home /etc/acme --renew -d openwrt.lan --debug --ecc'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: acme: Running acme.sh as '/usr/lib/acme/acme.sh --home /etc/acme --renew -d openwrt.lan --debug --ecc'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Lets find script dir.
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: _SCRIPT_='/usr/lib/acme/acme.sh'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: _script='/usr/lib/acme/acme.sh'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: _script_home='/usr/lib/acme'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Using config home:/etc/acme
Tue Aug  9 00:00:02 2022 daemon.info run-acme[7502]: https://github.com/acmesh-official/acme.sh
Tue Aug  9 00:00:02 2022 daemon.info run-acme[7502]: v3.0.1
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Running cmd: renew
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Using config home:/etc/acme
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: default_acme_server
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: ACME_DIRECTORY='https://acme.zerossl.com/v2/DV90'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: DOMAIN_PATH='/etc/acme/openwrt.lan_ecc'
Tue Aug  9 00:00:02 2022 daemon.info run-acme[7502]: Renew: 'openwrt.lan'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Le_API='http://my-ca.server/acme/acme/directory'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Using config home:/etc/acme
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: ACME_DIRECTORY='http://my-ca.server/acme/acme/directory'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: _init api for server: http://my-ca.server/acme/acme/directory
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: Retrying GET
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: GET
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: url='http://my-ca.server/acme/acme/directory'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: timeout=
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: displayError='1'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: _WGET='wget -q --content-on-error '
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: ret='0'
Tue Aug  9 00:00:02 2022 daemon.err run-acme[7502]: _hcode='0'
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: ACME_KEY_CHANGE='https://my-ca.server/acme/acme/key-change'
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: ACME_NEW_AUTHZ
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: ACME_NEW_ORDER='https://my-ca.server/acme/acme/new-order'
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: ACME_NEW_ACCOUNT='https://my-ca.server/acme/acme/new-account'
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: ACME_REVOKE_CERT='https://my-ca.server/acme/acme/revoke-cert'
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: ACME_AGREEMENT
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: ACME_NEW_NONCE='https://my-ca.server/acme/acme/new-nonce'
Tue Aug  9 00:00:03 2022 daemon.info run-acme[7502]: Skip, Next renewal time is: Tue Sep  6 17:27:58 UTC 2022
Tue Aug  9 00:00:03 2022 daemon.info run-acme[7502]: Add '--force' to force to renew.
Tue Aug  9 00:00:03 2022 daemon.info acme: Running post checks (cleanup).
Tue Aug  9 00:00:03 2022 daemon.err run-acme[7502]: acme: Running post checks (cleanup).
```
</details>

Expired certificate:
[image]
Public key algo:
[image]

Neustradamus, 2022-08-15:
Maybe a comeback from some people, this is a list of everyone who have contributed on this package: @tohojo, @jannispinter, @httpstorm, @neheb, @scde, @lialosiu, @hgl, @Meano, @peter-stadler, @lemoer, @4wrxb, @feckert, @pesintta, @socram8888, @yangfl, @wt, @azuwis, @lucize, @tripolar, @yousong, @hnyman, @raorn, @Ansuel, @halmi, @dauriata, @nos1609, @stokito.

Note: There are other PRs about acme:
- https://github.com/openwrt/packages/search?o=desc&q=acme&s=updated&type=issues

tohojo, 2022-08-15:
Hmm, I replied to this via email, but guess the email bridge failed? Here's what I wrote:

It's renewing an existing certificate, so it won't honour --ecc if the old one was non-ecc (it'll just renew whatever's there already). From this line:

```Tue Aug  9 00:00:03 2022 daemon.info run-acme[7502]: Skip, Next renewal time is: Tue Sep  6 17:27:58 UTC 2022```

it looks like it thinks the certificate is valid until September? So if it's not showing up in the browser it's because whatever daemon is serving the certificate needs to be restarted?

GGG-KILLER, 2022-08-15:
> Hmm, I replied to this via email, but guess the email bridge failed? Here's what I wrote:
>
> It's renewing an existing certificate, so it won't honour --ecc if the old one was non-ecc (it'll just [...]

Well, the thing is that I have restarted even the router itself to no avail, it still gets the renewal time wrong.

And the certificate has never been non-ecc, it has always been configured as ecc

GGG-KILLER, 2022-08-15:
Update: I have deleted everything in `/etc/acme` and re-issued the certificate will post an update here in 24 hours if the cert is still valid or not.

httpstorm, 2022-08-16:
@GGG-KILLER 
To achieve what you want, someone has to update the LuCI app, and the acme script `/usr/lib/acme/run-acme` to support hours. I would expect a granularity of 1 day to be sufficient for most users. Also note that the cron job for acme `0 0 * * * /etc/init.d/acme start` is set to run every day at midnight, so you might need to change that too. See `man 5 crontabs`.

As a quick and dirty workaround, you may add a cron job, which runs as frequently as you like.
Delete the certificate and restart acme to issue a new one:
`rm -rf /etc/acme/yourcert ; /etc/init.d/acme restart`

In case you need to preserve the existing key and generate a new certificate for it, you may try to delete only the certificate and then restart acme. _I'm not sure if this will work._

GGG-KILLER, 2022-08-16:
> @GGG-KILLER To achieve what you want, someone has to update the LuCI app, and the acme script `/usr/lib/acme/run-acme` to support hours. I would expect a granularity of 1 day to be sufficient for [...]

Yeah, I have deleted everything in `/etc/acme` and the certificate has been issued correctly, I'm just waiting for the renewal to run to see if it'll properly renew the cert

closed 2022-08-16 by GGG-KILLER

httpstorm, 2022-10-12:
@tohojo @GGG-KILLER 
How do I run a custom command whenever the certificate is updated?

tohojo, 2022-10-13:
Georgi Valkov ***@***.***> writes:

> @tohojo @GGG-KILLER 
> How do I run a custom command whenever the certificate is updated?

I guess you'd need to drop in a custom hotplug script?

httpstorm, 2022-10-14:
@tohojo 
Do you think this will work? I created it two weeks ago, but ran out of certificate requests, so I could not test it.

``` bash
# /etc/hotplug.d/acme/10-httpstorm
case "$ACTION" in
	issued | renewed)
		# reload config
		/usr/sbin/httpstorm -dc
		;;

	*)
		;;
esac
```

Also in `/etc/config/acme` I have `option update_uhttpd '1'`, which should restart uhttpd used by LuCI, but looking at the acme scripts I could not find where this setting is used and how it works.

tohojo, 2022-10-14:
Georgi Valkov ***@***.***> writes:

> @tohojo 
> Do you think this will work? I created it two weeks ago, but ran out of certificate requests, so I could not test it.
>
> ``` bash
> # /etc/hotplug.d/acme/10-httpstorm
> case "$ACTION" in
> 	issued | renewed)
> 		# reload config
> 		/usr/sbin/httpstorm -dc
> 		;;
>
> 	*)
> 		;;
> esac
> ```

I guess?

> Also in `/etc/config/acme` I have `option update_uhttpd '1'`, which
> should restart uhttpd used by LuCI, but looking at the acme scripts I
> could not find where this setting is used and how it works.

That's deprecated; the hotplug script for uhttpd will always reload the
daemon, but there's no config changes applied after the acme refactor.
If you have a previously adjusted the uhttpd config to pickup the acme
cert, though, things should continue to work...

httpstorm, 2022-10-14:
@tohojo 
Something is not right. You mentioned a `hotplug.d` script for uhttpd, and I would expect it to be under `/etc/hotplug.d/acme/`, but that path did not exist. Any idea why? So I created that path two weeks ago, then added `10-httpstorm`. Finally I tried running my script manually:
`ACTION=renewed /etc/hotplug.d/acme/10-httpstorm`
The configuration for `httpstorm` is updated, though I am not sure if the script is called when an acme certificate is updated.

My current build is based on master, commit cf2c9498bed5e2f34170ada394be26348949f107.

tohojo, 2022-10-14:
Hmm, yeah, there should be one. Seems it wasn't included in the initial PR, probably because uhttpd lives in the openwrt source tree not in openwrt-packages. @hgl care to fix that?

hgl, 2022-10-14:
Will send a PR soon.