openwrt/packages #19148: xinetd: postinst script: permission denied for /etc/init.d/

code-surfer, 2022-08-11:
Maintainer: common packages maintainers
Environment: Turris MOX with a fresh Turris OS 5.4.0 which is based on OpenWrt 19.07.

Description:

I want to install `xinetd` (required for `muninlite`).
```
# opkg install xinetd
Installing xinetd (2.3.15-5) to root...
Downloading https://repo.turris.cz/hbs/mox/packages/packages/xinetd_2.3.15-5_aarch64_cortex-a53.ipk
Configuring xinetd.
//usr/lib/opkg/info/xinetd.postinst: line 261: /etc/init.d/: Permission denied
```
The postinst script is only a few lines long, so the line number is misleading.

I installed `strace` and reinstalled `xinetd`. Same error again, and in the trace I found
```
8930  rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
8930  close(10)                         = 0
8930  rt_sigaction(SIGQUIT, NULL, {sa_handler=SIG_IGN, sa_mask=[], sa_flags=0}, 8) = 0
8930  execve("/etc/init.d/", ["/etc/init.d/", "enabled"], 0x5588b2fbe8 /* 17 vars */) = -1 EACCES (Permission denied)
8930  writev(2, [{iov_base="//usr/lib/opkg/info/xinetd.posti"..., iov_len=37}, {iov_base=NULL, iov_len=0}], 2) = 37
8930  writev(2, [{iov_base="line 261: ", iov_len=10}, {iov_base=NULL, iov_len=0}], 2) = 10
8930  writev(2, [{iov_base="/etc/init.d/: Permission denied", iov_len=31}, {iov_base=NULL, iov_len=0}], 2) = 31
8930  writev(2, [{iov_base="", iov_len=0}, {iov_base="\n", iov_len=1}], 2) = 1
8930  exit_group(126)                   = ?
8930  +++ exited with 126 +++
```
So, something seems to try to exec the `/etc/init.d/` directory, which doesn't make any sense.

This looks similar to #19102 and also behaves the same during `opkg remove`:
```
# opkg remove xinetd
Removing package xinetd from root...
//usr/lib/opkg/info/xinetd.prerm: line 169: /etc/init.d/: Permission denied
//usr/lib/opkg/info/xinetd.prerm: line 171: /etc/init.d/: Permission denied

```

code-surfer, 2022-08-11:
The same is also happening when I try to install `collected`.

peci1, 2022-08-23:
And [softethervpn5-client](https://forum.turris.cz/t/installing-softethervpn5-client-fails-with-etc-init-d-permission-denied/17544).

peci1, 2022-08-23:
This seems to be a turris-specific issue: https://forum.turris.cz/t/postinst-script-permission-denied-for-etc-init-d/17473/6?u=peci1 .

jow-, 2022-08-23:
See https://forum.turris.cz/t/postinst-script-permission-denied-for-etc-init-d/17473/6

closed 2022-08-23 by jow-

BKPepe, 2022-08-31:
> Turris-specific issue

Partially. We applied this commit https://github.com/openwrt/openwrt/pull/10061, which causes it, but https://gitlab.nic.cz/turris/os/build/-/merge_requests/555 should solve it. I will add it to that PR.