openwrt/packages #19149: dnscrypt-proxy2: update to v2.1.2

elgatov, 2022-08-12:
Maintainer: @BKPepe 
Environment: `OpenWrt 21.02.0, r16279-5cc0535800`

Description:

`dnscrypt-proxy2` has been [recently updated](https://github.com/dnscrypt/dnscrypt-proxy/releases/tag/2.1.2) with an important fix:

> In previous versions, memory usage kept growing due to channels not being properly closed, causing goroutines to pile up. This was fixed, resulting in an important reduction of memory usage. Thanks to @lifenjoiner for investigating and fixing this!

Sometimes, when the device was running out of memory (around 80%), it would trigger oom-reaper and kill the `dnscrypt-proxy2` process, which in turn would start the cycle again.

BKPepe, 2022-08-12:
Someone reached me a few days ago at my email address to update it. Wasn't it you? If you would like to be faster, please create a pull request.

elgatov, 2022-08-12:
> Someone reached me a few days ago at my email address to update it. Wasn't it you? If you would like to be faster, please create a pull request.

Hi!

No it wasn't me, I did not know you were already noticed about this. I can close the issue if you want, then.

dynasticorpheus, 2022-08-16:
See #19180

closed 2022-08-21 by BKPepe