openwrt/packages #19155: strongswan 5.9.5-8 in 22.03.0-rc6 does not seem to work

lejeczek, 2022-08-12:
Hi.

OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"

with a few 'strongswan-minimal' and 'strongswan-swanctl' plus a few more pkgs I get "dysfunctional" Swan.
From 'logread':

Fri Aug 12 16:53:53 2022 daemon.info ipsec: 00[DMN] Starting IKE charon daemon (strongSwan 5.9.5, Linux 5.10.134, armv7l)
Fri Aug 12 16:53:53 2022 daemon.info ipsec: 00[LIB] failed to open /dev/net/tun: No such file or directory
Fri Aug 12 16:53:53 2022 daemon.info ipsec: 00[KNL] failed to create TUN device
Fri Aug 12 16:53:53 2022 daemon.info ipsec: 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Fri Aug 12 16:53:53 2022 daemon.info ipsec: 00[KNL] unable to create IPv4 routing table rule
Fri Aug 12 16:53:53 2022 daemon.info ipsec: 00[KNL] unable to create IPv6 routing table rule
...
Fri Aug 12 16:53:53 2022 daemon.info ipsec: 03[DMN] killing ourself, received critical signal
Fri Aug 12 16:53:53 2022 authpriv.info ipsec_starter[8366]: charon has died -- restart scheduled (5sec)
Fri Aug 12 16:53:53 2022 authpriv.info ipsec_starter[8366]: charon refused to be started
Fri Aug 12 16:53:58 2022 daemon.info ipsec: 00[DMN] Starting IKE charon daemon (strongSwan 5.9.5, Linux 5.10.134, armv7l)
Fri Aug 12 16:53:58 2022 daemon.info ipsec: 00[LIB] failed to open /dev/net/tun: No such file or directory
Fri Aug 12 16:53:58 2022 daemon.info ipsec: 00[KNL] failed to create TUN device
Fri Aug 12 16:53:58 2022 daemon.info ipsec: 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
....
Fri Aug 12 16:56:10 2022 daemon.info ipsec: 00[DMN] executing start script 'load-all' (/usr/sbin/swanctl --load-all --noprompt)
Fri Aug 12 16:56:15 2022 daemon.info ipsec: 00[DMN] executing start script 'load-all' (/usr/sbin/swanctl --load-all --noprompt)
Fri Aug 12 16:56:20 2022 daemon.info ipsec: 00[DMN] executing start script 'load-all' (/usr/sbin/swanctl --load-all --noprompt)
...

-> $ swanctl --stats
connecting to 'unix:///var/run/charon.vici' failed: Connection refused
Error: connecting to 'default' URI failed: Connection refused

many thanks, L.

pprindeville, 2022-08-12:
@Thermi any ideas?

Thermi, 2022-08-13:
check dmesg

closed 2025-11-15 by GeorgeSapkin